coolsnowwolf/lede #11161: kernel: add support for threaded network backlog processing 不能按照commit描述设置为enable

GreatMichaelLee, 2023-04-28:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：kernel: add support for threaded network backlog processing 不能按照commit描述设置为enable, 提示没有相关内核参数

### (2) 路由器型号和固件版本

A：x86 OpenWrt R23.4.1 / LuCI Master (git-23.118.10415-24f8cbd)，内核6.1.25

### (3) 详细日志

A：
[image]

root@OpenWrt:~# echo 1 > /proc/sys/net/core/backlog_threaded
-ash: can't create /proc/sys/net/core/backlog_threaded: nonexistent directory

GreatMichaelLee, 2023-04-29:
> 这次commit只提供了5.15内核的补丁

6.1.x是不支持还是官方补丁没出？

WYC-2020, 2023-05-04:
这玩意照葫芦画瓢适配一下就完事，其实我发现很多 都是只适配了部分内核的，其实有好几个内核都可以适配，只是没做，可能大雕没时间也有可能是没时间测，因为适配了还需要做测试，5.4 5.10 5.15 我都搞了，明天搞下6.1就完事

GreatMichaelLee, 2023-05-05:
> 这玩意照葫芦画瓢适配一下就完事，其实我发现很多 都是只适配了部分内核的，其实有好几个内核都可以适配，只是没做，可能大雕没时间也有可能是没时间测，因为适配了还需要做测试，5.4 5.10 5.15 我都搞了，明天搞下6.1就完事

老板，搞了没~

WYC-2020, 2023-05-05:
> > 这玩意照葫芦画瓢适配一下就完事，其实我发现很多 都是只适配了部分内核的，其实有好几个内核都可以适配，只是没做，可能大雕没时间也有可能是没时间测，因为适配了还需要做测试，5.4 5.10 5.15 我都搞了，明天搞下6.1就完事
> 
> 老板，搞了没~

早就搞了，昨晚都搞好了

GreatMichaelLee, 2023-05-05:
lede仓库没看到？今天只看到一个5.10的

WYC-2020, 2023-05-05:
> lede仓库没看到？今天只看到一个5.10的
> […](#)

那是因为我没有提，你知道的，我不喜欢提交，
[patch.zip](https://github.com/coolsnowwolf/lede/files/11406176/patch.zip)
拿去

WYC-2020, 2023-05-05:
主要这几天在修macos 下面编译，昨天才修完，晚上才搞的这个

GreatMichaelLee, 2023-05-05:
> 主要这几天在修macos 下面编译，昨天才修完，晚上才搞的这个

提交一下嘛....

> > lede仓库没看到？今天只看到一个5.10的
> > […](#)
> 
> 那是因为我没有提，你知道的，我不喜欢提交， [patch.zip](https://github.com/coolsnowwolf/lede/files/11406176/patch.zip) 拿去

提交一下嘛，提交最光荣:P

GreatMichaelLee, 2023-05-06:
刚在6.1上测了，有问题，kernel panic, 起不来，环境x86+n5105, VMware 8.0.1, kernel panic 时trace如下：

at May  6 21:42:46 2023 kern.warn kernel: [  119.674527] ------------[ cut here ]------------
Sat May  6 21:42:46 2023 kern.warn kernel: [  119.679655] WARNING: CPU: 1 PID: 7863 at napi_threaded_poll+0xcb/0xd0
Sat May  6 21:42:46 2023 kern.warn kernel: [  119.680696] Modules linked in: ksmbd pppoe ppp_async i915 amdgpu wireguard video snd_usb_audio snd_hda_intel snd_hda_codec_via snd_hda_codec_realtek sn
Sat May  6 21:42:46 2023 kern.warn kernel: [  119.680766]  r8125 poly1305_x86_64 pcnet32 nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject_bridge nft_reject nft_redir nft_quota nft_objref
Sat May  6 21:42:46 2023 kern.warn kernel: [  119.692164]  act_connmark sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred a
Sat May  6 21:42:46 2023 kern.warn kernel: [  119.703069]  rpcsec_gss_krb5 auth_rpcgss veth tun snd_rawmidi snd_seq_device snd_pcm_oss snd_mixer_oss snd_hwdep snd_compress snd_pcm snd_timer snd sou
Sat May  6 21:42:46 2023 kern.warn kernel: [  119.738259] CPU: 1 PID: 7863 Comm: napi/backlog-0 Tainted: G        W          6.1.26 #0
Sat May  6 21:42:46 2023 kern.warn kernel: [  119.739290] Hardware name: VMware, Inc. VMware7,1/440BX Desktop Reference Platform, BIOS VMW71.00V.20829224.B64.2211211837 11/21/2022
Sat May  6 21:42:46 2023 kern.warn kernel: [  119.741269] RIP: 0010:napi_threaded_poll+0xcb/0xd0
Sat May  6 21:42:46 2023 kern.warn kernel: [  119.742448] Code: 00 00 48 8d 75 e7 48 89 df e8 41 fe ff ff be 00 02 00 00 48 c7 c7 04 38 78 81 e8 10 26 94 ff 80 7d e7 00 75 c9 e9 52 ff ff ff <0f> 0b
Sat May  6 21:42:46 2023 kern.warn kernel: [  119.744394] RSP: 0018:ffffc90008867ee8 EFLAGS: 00010282
Sat May  6 21:42:46 2023 kern.warn kernel: [  119.748014] RAX: 0000000000000000 RBX: ffff88807dc240d0 RCX: 0000000000000002
Sat May  6 21:42:46 2023 kern.warn kernel: [  119.749079] RDX: 0000000000000000 RSI: 0000000000000001 RDI: ffff88807dca2fc0
Sat May  6 21:42:46 2023 kern.warn kernel: [  119.750106] RBP: ffffc90008867f08 R08: 0000000000000400 R09: 0000000000000000
Sat May  6 21:42:46 2023 kern.warn kernel: [  119.751025] R10: 0000000000000001 R11: 0000000000000000 R12: ffff88800388a880
Sat May  6 21:42:46 2023 kern.warn kernel: [  119.752142] R13: 0000000000000001 R14: ffff888012c6ac80 R15: ffff88807dc240d0
Sat May  6 21:42:46 2023 kern.warn kernel: [  119.753064] FS:  0000000000000000(0000) GS:ffff88807dc80000(0000) knlGS:0000000000000000
Sat May  6 21:42:46 2023 kern.warn kernel: [  119.754025] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Sat May  6 21:42:46 2023 kern.warn kernel: [  119.754968] CR2: 00007fff1c3d41e8 CR3: 000000000320a000 CR4: 0000000000350ee0
Sat May  6 21:42:46 2023 kern.warn kernel: [  119.758891] Call Trace:
Sat May  6 21:42:46 2023 kern.warn kernel: [  119.759658]  <TASK>
Sat May  6 21:42:46 2023 kern.warn kernel: [  119.760406]  kthread+0xe2/0x110
Sat May  6 21:42:46 2023 kern.warn kernel: [  119.761200]  ? __napi_poll+0x110/0x110
Sat May  6 21:42:46 2023 kern.warn kernel: [  119.762498]  ? kthread_complete_and_exit+0x20/0x20
Sat May  6 21:42:46 2023 kern.warn kernel: [  119.763325]  ret_from_fork+0x1f/0x30
Sat May  6 21:42:46 2023 kern.warn kernel: [  119.764064]  </TASK>
Sat May  6 21:42:46 2023 kern.warn kernel: [  119.764734] ---[ end trace 0000000000000000 ]---

WYC-2020, 2023-05-06:
> 刚在6.1上测了，有问题，kernel panic, 起不来，环境x86+n5105, VMware 8.0.1, kernel panic 时trace如下：
> 
> at May 6 21:42:46 2023 kern.warn kernel: [ 119.674527] ------------[ cut here ]------------ Sat May 6 21:42:46 2023 kern.warn kernel: [ 119.679655] WARNING: CPU: 1 PID: 7863 at napi_threaded_poll+0xcb/0xd0 Sat May 6 21:42:46 2023 kern.warn kernel: [ 119.680696] Modules linked in: ksmbd pppoe ppp_async i915 amdgpu wireguard video snd_usb_audio snd_hda_intel snd_hda_codec_via snd_hda_codec_realtek sn Sat May 6 21:42:46 2023 kern.warn kernel: [ 119.680766] r8125 poly1305_x86_64 pcnet32 nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject_bridge nft_reject nft_redir nft_quota nft_objref Sat May 6 21:42:46 2023 kern.warn kernel: [ 119.692164] act_connmark sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred a Sat May 6 21:42:46 2023 kern.warn kernel: [ 119.703069] rpcsec_gss_krb5 auth_rpcgss veth tun snd_rawmidi snd_seq_device snd_pcm_oss snd_mixer_oss snd_hwdep snd_compress snd_pcm snd_timer snd sou Sat May 6 21:42:46 2023 kern.warn kernel: [ 119.738259] CPU: 1 PID: 7863 Comm: napi/backlog-0 Tainted: G W 6.1.26 #0 Sat May 6 21:42:46 2023 kern.warn kernel: [ 119.739290] Hardware name: VMware, Inc. VMware7,1/440BX Desktop Reference Platform, BIOS VMW71.00V.20829224.B64.2211211837 11/21/2022 Sat May 6 21:42:46 2023 kern.warn kernel: [ 119.741269] RIP: 0010:napi_threaded_poll+0xcb/0xd0 Sat May 6 21:42:46 2023 kern.warn kernel: [ 119.742448] Code: 00 00 48 8d 75 e7 48 89 df e8 41 fe ff ff be 00 02 00 00 48 c7 c7 04 38 78 81 e8 10 26 94 ff 80 7d e7 00 75 c9 e9 52 ff ff ff <0f> 0b Sat May 6 21:42:46 2023 kern.warn kernel: [ 119.744394] RSP: 0018:ffffc90008867ee8 EFLAGS: 00010282 Sat May 6 21:42:46 2023 kern.warn kernel: [ 119.748014] RAX: 0000000000000000 RBX: ffff88807dc240d0 RCX: 0000000000000002 Sat May 6 21:42:46 2023 kern.warn kernel: [ 119.749079] RDX: 0000000000000000 RSI: 0000000000000001 RDI: ffff88807dca2fc0 Sat May 6 21:42:46 2023 kern.warn kernel: [ 119.750106] RBP: ffffc90008867f08 R08: 0000000000000400 R09: 0000000000000000 Sat May 6 21:42:46 2023 kern.warn kernel: [ 119.751025] R10: 0000000000000001 R11: 0000000000000000 R12: ffff88800388a880 Sat May 6 21:42:46 2023 kern.warn kernel: [ 119.752142] R13: 0000000000000001 R14: ffff888012c6ac80 R15: ffff88807dc240d0 Sat May 6 21:42:46 2023 kern.warn kernel: [ 119.753064] FS: 0000000000000000(0000) GS:ffff88807dc80000(0000) knlGS:0000000000000000 Sat May 6 21:42:46 2023 kern.warn kernel: [ 119.754025] CS: 0010 DS: 0000 ES: 0000 CR0: 0000000080050033 Sat May 6 21:42:46 2023 kern.warn kernel: [ 119.754968] CR2: 00007fff1c3d41e8 CR3: 000000000320a000 CR4: 0000000000350ee0 Sat May 6 21:42:46 2023 kern.warn kernel: [ 119.758891] Call Trace: Sat May 6 21:42:46 2023 kern.warn kernel: [ 119.759658] Sat May 6 21:42:46 2023 kern.warn kernel: [ 119.760406] kthread+0xe2/0x110 Sat May 6 21:42:46 2023 kern.warn kernel: [ 119.761200] ? __napi_poll+0x110/0x110 Sat May 6 21:42:46 2023 kern.warn kernel: [ 119.762498] ? kthread_complete_and_exit+0x20/0x20 Sat May 6 21:42:46 2023 kern.warn kernel: [ 119.763325] ret_from_fork+0x1f/0x30 Sat May 6 21:42:46 2023 kern.warn kernel: [ 119.764064] Sat May 6 21:42:46 2023 kern.warn kernel: [ 119.764734] ---[ end trace 0000000000000000 ]---

那就等看那个大雕适配了

closed 2023-05-11 by aiamadeus

GreatMichaelLee, 2023-05-13:
这个问题没有解决啊，在6.1上不work啊，现在把x86 主版本切到6.1了，麻烦在6.1上实现一下。