coolsnowwolf/lede #11164: 最新6.1.25源码无法驱动QCNFA765

Y-ZD-heheka, 2023-04-29:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：使用lede最新6.1.25源码无法驱动QCNFA765网卡

### (2) 路由器型号和固件版本

A：X86_64，kernel 6.1.25，PVE虚拟机直通QCNFA765

### (3) 详细日志

A：
[image]
[image]
[image]
无线中点扫描没有东西
[image]
选择的固件
[image]
[image]

Y-ZD-heheka, 2023-04-29:
[image]
[image]
换上QCNFA765后内存占用高了30MB

closed 2023-05-05 by coolsnowwolf

Y-ZD-heheka, 2023-05-06:
大佬还是驱动不起来啊

Kisoul, 2024-08-12:
> 大佬还是驱动不起来啊

解决了吗？怎么解决的？

Y-ZD-heheka, 2024-08-13:
> > 大佬还是驱动不起来啊
> 
> 解决了吗？怎么解决的？

我现在用immortal，把ath11k pci相关的驱动选上就能驱动