coolsnowwolf/lede #11165: 模块配置中<M>不能切换成<*>

tudouershi, 2023-04-29:
[image]
[image]
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ×] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：在Linux内核目录下模块配置中不能安装USB driver for GSM and CDMA modems模块，<M>不能切换成<*>

#
### (2) 路由器型号和固件版本

A.蒲公英X4C    openwrt R23.5

### (3) 详细日志

A：

dgyangxf2022, 2023-05-01:
自己手动修改一下配置文件，把=m改成=y就行了吧

coolsnowwolf, 2026-09-05:
该 issue 只有截图，没有具体内核配置符号、依赖链或完整编译日志。Kconfig 中 `<M>` 无法切换为 `<*>` 通常是依赖项未满足或该驱动被设计为模块，评论已给出直接修改 `.config` 的办法；当前没有足够证据形成仓库修复，关闭。

closed 2026-09-05 by coolsnowwolf