coolsnowwolf/lede #11172: R619ac(竞斗云2.0)编译ath10k-ct失败

Luupw3d, 2023-04-30:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：竞斗云2.0使用最新源码编译失败，具体模块为ath10k-ct.

### (2) 路由器型号和固件版本

A：竞斗云2.0

### (3) 详细日志

A：
`Applying ./patches/960-0011-ath10k-limit-pci-buffer-size.patch using plaintext: 
patching file ath10k-5.15/pci.c
touch /home/ubuntu/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/ath10k-ct-regular/ath10k-ct-2022-05-13-f808496f/.prepared_a9a7fb4853f797ff87cb4a0b03c20ca6_6664517399ebbbc92a37c5bb081b5c53
make[3]: *** No rule to make target '/home/ubuntu/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/usr/include/mac80211-backport/backport/autoconf.h', needed by '/home/ubuntu/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/ath10k-ct-regular/ath10k-ct-2022-05-13-f808496f/.configured_68b329da9893e34099c7d8ad5cb9c940'.  Stop.
make[3]: Leaving directory '/home/ubuntu/lede/package/kernel/ath10k-ct'
time: package/kernel/ath10k-ct/regular/compile#0.45#0.42#0.76
    ERROR: package/kernel/ath10k-ct failed to build (build variant: regular).
make[2]: *** [package/Makefile:116: package/kernel/ath10k-ct/compile] Error 1
make[2]: Leaving directory '/home/ubuntu/lede'
make[1]: *** [package/Makefile:110: /home/ubuntu/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/ubuntu/lede'
make: *** [/home/ubuntu/lede/include/toplevel.mk:231：world] 错误 2
`

AE86JAY, 2023-05-02:
好像是openwrt官方更改了ipq40xx的内核版本到5.15，且删除了5.4，默认编译5.15，但L大的还是5.4

WYC-2020, 2023-05-05:
这不是刚升级了么，你在编译试试呢

AE86JAY, 2023-05-05:
> 这不是刚升级了么，你在编译试试呢

我这几天编译了十几次，用本地编译成功的.config、feeds.conf.default、diy脚本文件放到云编译都会出楼主一样的错误，但不加插件就每次都成功。有些加了插件也能成功编译，但刷机后发现插件没有完全编译进去，config.buildinfo 里面也只有一部分。

hzjnet, 2023-05-05:
> 这不是刚升级了么，你在编译试试呢

这问题存在几个月了，翻翻关闭的多个类似的issus就知道了；当然也能解决，但是麻烦；同时集成nss加速后影响mesh应用

AE86JAY, 2023-05-06:
> 

今天又试了几次，只要加easymesh插件必出错。

Luupw3d, 2023-05-06:
> > 
> 
> 今天又试了几次，只要加easymesh插件必出错。

与我的测试结果相同，很可能是easymesh导致的。对了，从今天开始，源码同时导致无线无法正常驱动。

hzjnet, 2023-05-09:
> > > 
> > 
> > 
> > 今天又试了几次，只要加easymesh插件必出错。
> 
> 与我的测试结果相同，很可能是easymesh导致的。对了，从今天开始，源码同时导致无线无法正常驱动。

应该不是 我编译的无线正常

AE86JAY, 2023-05-14:
> > > > 
> > > 
> > > 
> > > 今天又试了几次，只要加easymesh插件必出错。
> > 
> > 
> > 与我的测试结果相同，很可能是easymesh导致的。对了，从今天开始，源码同时导致无线无法正常驱动。
> 
> 应该不是 我编译的无线正常

反正我的只加easymesh其它插件都是默认的都会出错

AE86JAY, 2023-05-18:
2023-05-18T18:07:27.0148556Z make[2]: Circular package/kernel/ath10k-ct/compile <- package/kernel/mac80211/compile dependency dropped.
2023-05-18T18:07:27.0257334Z make[3]: Entering directory '/workdir/openwrt/package/kernel/ath10k-ct'
2023-05-18T18:07:27.2265318Z touch /workdir/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/ath10k-ct-regular/ath10k-ct-2022-05-13-f808496f/.prepared_33bd6a5bdaf3afe4c738ba76f5e01fd0_6664517399ebbbc92a37c5bb081b5c53_check
2023-05-18T18:07:27.2290663Z . /workdir/openwrt/include/shell.sh; xzcat /workdir/openwrt/dl/ath10k-ct-2022-05-13-f808496f.tar.xz | tar -C /workdir/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/ath10k-ct-regular/ath10k-ct-2022-05-13-f808496f/.. -xf -
2023-05-18T18:07:27.3194722Z [ ! -d ./src/ ] || cp -fpR ./src/. /workdir/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/ath10k-ct-regular/ath10k-ct-2022-05-13-f808496f
2023-05-18T18:07:27.3253119Z 
2023-05-18T18:07:27.3253854Z Applying ./patches/100-api_update.patch using plaintext: 
2023-05-18T18:07:27.3269474Z patching file ath10k-5.15/mac.c
2023-05-18T18:07:27.3282384Z patching file ath10k-5.15/txrx.c
2023-05-18T18:07:27.3290097Z patching file ath10k-5.15/wmi.c
2023-05-18T18:07:27.3306400Z patching file ath10k-5.15/htt_rx.c
2023-05-18T18:07:27.3313663Z patching file ath10k-5.15/wmi-tlv.c
2023-05-18T18:07:27.3320788Z patching file ath10k-5.15/core.c
2023-05-18T18:07:27.3327818Z patching file ath10k-5.15/pci.c
2023-05-18T18:07:27.3333710Z patching file ath10k-5.15/sdio.c
2023-05-18T18:07:27.3338456Z patching file ath10k-5.15/snoc.c
2023-05-18T18:07:27.3343987Z 
2023-05-18T18:07:27.3344591Z Applying ./patches/120-ath10k-fetch-calibration-data-via-nvmem-subsystem.patch using plaintext: 
2023-05-18T18:07:27.3357278Z patching file ath10k-5.15/core.c
2023-05-18T18:07:27.3365267Z patching file ath10k-5.15/core.h
2023-05-18T18:07:27.3371334Z 
2023-05-18T18:07:27.3373262Z Applying ./patches/130-ath10k-read-qcom-coexist-support-as-a-u32.patch using plaintext: 
2023-05-18T18:07:27.3386569Z patching file ath10k-5.15/core.c
2023-05-18T18:07:27.3393316Z 
2023-05-18T18:07:27.3394325Z Applying ./patches/201-ath10k-add-LED-and-GPIO-controlling-support-for-various-chipsets.patch using plaintext: 
2023-05-18T18:07:27.3407415Z patching file ath10k-5.15/Kconfig
2023-05-18T18:07:27.3408360Z patching file ath10k-5.15/Makefile
2023-05-18T18:07:27.3412223Z patching file ath10k-5.15/core.c
2023-05-18T18:07:27.3417096Z patching file ath10k-5.15/core.h
2023-05-18T18:07:27.3419364Z patching file ath10k-5.15/hw.h
2023-05-18T18:07:27.3421012Z patching file ath10k-5.15/leds.c
2023-05-18T18:07:27.3422020Z patching file ath10k-5.15/leds.h
2023-05-18T18:07:27.3427581Z patching file ath10k-5.15/mac.c
2023-05-18T18:07:27.3436508Z patching file ath10k-5.15/wmi-ops.h
2023-05-18T18:07:27.3440050Z patching file ath10k-5.15/wmi-tlv.c
2023-05-18T18:07:27.3448623Z patching file ath10k-5.15/wmi.c
2023-05-18T18:07:27.3460517Z patching file ath10k-5.15/wmi.h
2023-05-18T18:07:27.3469402Z 
2023-05-18T18:07:27.3469794Z Applying ./patches/202-ath10k-use-tpt-trigger-by-default.patch using plaintext: 
2023-05-18T18:07:27.3479880Z patching file ath10k-5.15/core.h
2023-05-18T18:07:27.3483601Z patching file ath10k-5.15/leds.c
2023-05-18T18:07:27.3492133Z patching file ath10k-5.15/mac.c
2023-05-18T18:07:27.3503110Z 
2023-05-18T18:07:27.3503709Z Applying ./patches/300-ath10k-ct-Fix-spectral-scan-NULL-pointer.patch using plaintext: 
2023-05-18T18:07:27.3518143Z patching file ath10k-5.15/spectral.c
2023-05-18T18:07:27.3523108Z 
2023-05-18T18:07:27.3523673Z Applying ./patches/960-0010-ath10k-limit-htt-rx-ring-size.patch using plaintext: 
2023-05-18T18:07:27.3538643Z patching file ath10k-5.15/htt.h
2023-05-18T18:07:27.3542122Z 
2023-05-18T18:07:27.3542755Z Applying ./patches/960-0011-ath10k-limit-pci-buffer-size.patch using plaintext: 
2023-05-18T18:07:27.3557997Z patching file ath10k-5.15/pci.c
2023-05-18T18:07:27.3606113Z touch /workdir/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/ath10k-ct-regular/ath10k-ct-2022-05-13-f808496f/.prepared_33bd6a5bdaf3afe4c738ba76f5e01fd0_6664517399ebbbc92a37c5bb081b5c53
2023-05-18T18:07:27.3631254Z make[3]: *** No rule to make target '/workdir/openwrt/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/usr/include/mac80211-backport/backport/autoconf.h', needed by '/workdir/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/ath10k-ct-regular/ath10k-ct-2022-05-13-f808496f/.configured_68b329da9893e34099c7d8ad5cb9c940'.  Stop.
2023-05-18T18:07:27.3632158Z make[3]: Leaving directory '/workdir/openwrt/package/kernel/ath10k-ct'
2023-05-18T18:07:27.3632840Z time: package/kernel/ath10k-ct/regular/compile#0.23#0.16#0.34
2023-05-18T18:07:27.3638139Z     ERROR: package/kernel/ath10k-ct failed to build (build variant: regular).
2023-05-18T18:07:27.3647906Z make[2]: Leaving directory '/workdir/openwrt'
2023-05-18T18:07:27.3662016Z make[2]: *** [package/Makefile:116: package/kernel/ath10k-ct/compile] Error 1
2023-05-18T18:07:27.3662426Z make[1]: Leaving directory '/workdir/openwrt'
2023-05-18T18:07:27.3663205Z make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.package_compile] Error 2
2023-05-18T18:07:27.3665810Z make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 2
2023-05-18T18:07:27.3774833Z ##[error]Process completed with exit code 2.

编译acrh17的也是一样问题，这个是不是内核与补丁版本不一致造成的？内核是5.4补丁是5.15？

AE86JAY, 2023-05-19:
> 2023-05-18T18:07:27.0148556Z make[2]: Circular package/kernel/ath10k-ct/compile <- package/kernel/mac80211/compile dependency dropped. 2023-05-18T18:07:27.0257334Z make[3]: Entering directory '/workdir/openwrt/package/kernel/ath10k-ct' 2023-05-18T18:07:27.2265318Z touch /workdir/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/ath10k-ct-regular/ath10k-ct-2022-05-13-f808496f/.prepared_33bd6a5bdaf3afe4c738ba76f5e01fd0_6664517399ebbbc92a37c5bb081b5c53_check 2023-05-18T18:07:27.2290663Z . /workdir/openwrt/include/shell.sh; xzcat /workdir/openwrt/dl/ath10k-ct-2022-05-13-f808496f.tar.xz | tar -C /workdir/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/ath10k-ct-regular/ath10k-ct-2022-05-13-f808496f/.. -xf - 2023-05-18T18:07:27.3194722Z [ ! -d ./src/ ] || cp -fpR ./src/. /workdir/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/ath10k-ct-regular/ath10k-ct-2022-05-13-f808496f 2023-05-18T18:07:27.3253119Z 2023-05-18T18:07:27.3253854Z Applying ./patches/100-api_update.patch using plaintext: 2023-05-18T18:07:27.3269474Z patching file ath10k-5.15/mac.c 2023-05-18T18:07:27.3282384Z patching file ath10k-5.15/txrx.c 2023-05-18T18:07:27.3290097Z patching file ath10k-5.15/wmi.c 2023-05-18T18:07:27.3306400Z patching file ath10k-5.15/htt_rx.c 2023-05-18T18:07:27.3313663Z patching file ath10k-5.15/wmi-tlv.c 2023-05-18T18:07:27.3320788Z patching file ath10k-5.15/core.c 2023-05-18T18:07:27.3327818Z patching file ath10k-5.15/pci.c 2023-05-18T18:07:27.3333710Z patching file ath10k-5.15/sdio.c 2023-05-18T18:07:27.3338456Z patching file ath10k-5.15/snoc.c 2023-05-18T18:07:27.3343987Z 2023-05-18T18:07:27.3344591Z Applying ./patches/120-ath10k-fetch-calibration-data-via-nvmem-subsystem.patch using plaintext: 2023-05-18T18:07:27.3357278Z patching file ath10k-5.15/core.c 2023-05-18T18:07:27.3365267Z patching file ath10k-5.15/core.h 2023-05-18T18:07:27.3371334Z 2023-05-18T18:07:27.3373262Z Applying ./patches/130-ath10k-read-qcom-coexist-support-as-a-u32.patch using plaintext: 2023-05-18T18:07:27.3386569Z patching file ath10k-5.15/core.c 2023-05-18T18:07:27.3393316Z 2023-05-18T18:07:27.3394325Z Applying ./patches/201-ath10k-add-LED-and-GPIO-controlling-support-for-various-chipsets.patch using plaintext: 2023-05-18T18:07:27.3407415Z patching file ath10k-5.15/Kconfig 2023-05-18T18:07:27.3408360Z patching file ath10k-5.15/Makefile 2023-05-18T18:07:27.3412223Z patching file ath10k-5.15/core.c 2023-05-18T18:07:27.3417096Z patching file ath10k-5.15/core.h 2023-05-18T18:07:27.3419364Z patching file ath10k-5.15/hw.h 2023-05-18T18:07:27.3421012Z patching file ath10k-5.15/leds.c 2023-05-18T18:07:27.3422020Z patching file ath10k-5.15/leds.h 2023-05-18T18:07:27.3427581Z patching file ath10k-5.15/mac.c 2023-05-18T18:07:27.3436508Z patching file ath10k-5.15/wmi-ops.h 2023-05-18T18:07:27.3440050Z patching file ath10k-5.15/wmi-tlv.c 2023-05-18T18:07:27.3448623Z patching file ath10k-5.15/wmi.c 2023-05-18T18:07:27.3460517Z patching file ath10k-5.15/wmi.h 2023-05-18T18:07:27.3469402Z 2023-05-18T18:07:27.3469794Z Applying ./patches/202-ath10k-use-tpt-trigger-by-default.patch using plaintext: 2023-05-18T18:07:27.3479880Z patching file ath10k-5.15/core.h 2023-05-18T18:07:27.3483601Z patching file ath10k-5.15/leds.c 2023-05-18T18:07:27.3492133Z patching file ath10k-5.15/mac.c 2023-05-18T18:07:27.3503110Z 2023-05-18T18:07:27.3503709Z Applying ./patches/300-ath10k-ct-Fix-spectral-scan-NULL-pointer.patch using plaintext: 2023-05-18T18:07:27.3518143Z patching file ath10k-5.15/spectral.c 2023-05-18T18:07:27.3523108Z 2023-05-18T18:07:27.3523673Z Applying ./patches/960-0010-ath10k-limit-htt-rx-ring-size.patch using plaintext: 2023-05-18T18:07:27.3538643Z patching file ath10k-5.15/htt.h 2023-05-18T18:07:27.3542122Z 2023-05-18T18:07:27.3542755Z Applying ./patches/960-0011-ath10k-limit-pci-buffer-size.patch using plaintext: 2023-05-18T18:07:27.3557997Z patching file ath10k-5.15/pci.c 2023-05-18T18:07:27.3606113Z touch /workdir/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/ath10k-ct-regular/ath10k-ct-2022-05-13-f808496f/.prepared_33bd6a5bdaf3afe4c738ba76f5e01fd0_6664517399ebbbc92a37c5bb081b5c53 2023-05-18T18:07:27.3631254Z make[3]: *** No rule to make target '/workdir/openwrt/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/usr/include/mac80211-backport/backport/autoconf.h', needed by '/workdir/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/ath10k-ct-regular/ath10k-ct-2022-05-13-f808496f/.configured_68b329da9893e34099c7d8ad5cb9c940'. Stop. 2023-05-18T18:07:27.3632158Z make[3]: Leaving directory '/workdir/openwrt/package/kernel/ath10k-ct' 2023-05-18T18:07:27.3632840Z time: package/kernel/ath10k-ct/regular/compile#0.23#0.16#0.34 2023-05-18T18:07:27.3638139Z ERROR: package/kernel/ath10k-ct failed to build (build variant: regular). 2023-05-18T18:07:27.3647906Z make[2]: Leaving directory '/workdir/openwrt' 2023-05-18T18:07:27.3662016Z make[2]: *** [package/Makefile:116: package/kernel/ath10k-ct/compile] Error 1 2023-05-18T18:07:27.3662426Z make[1]: Leaving directory '/workdir/openwrt' 2023-05-18T18:07:27.3663205Z make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.package_compile] Error 2 2023-05-18T18:07:27.3665810Z make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 2 2023-05-18T18:07:27.3774833Z ##[error]Process completed with exit code 2.
> 
> 编译acrh17的也是一样问题，这个是不是内核与补丁版本不一致造成的？内核是5.4补丁是5.15？相同的.config用官方或其他大佬ipq40xx内核是5.15的源码编译却可正常编译通过。

或者说[ath10k-ct](https://github.com/coolsnowwolf/lede/tree/master/package/kernel/ath10k)已放弃对5.4内核的支持？

Luupw3d, 2023-05-25:
在[此commit](https://github.com/coolsnowwolf/lede/commit/5b6c90520b3ba350b549ef214c3d98fc12c8f82d)后，该问题仍未修复，主要表现为选择luci_app_easymesh时仍报相同错误。

AE86JAY, 2023-05-25:
> 

是的，没修复easymesh冲突

commit 5b6c9052 (2023-05-19): mac80211: fixes nl80211 support in kernel 5.4

Run-tested-on: P&W R619AC 128M
Fixes: #11172, #11191