coolsnowwolf/lede #11173: 最新编译的不能自动获取ip地址，

nb12nb34, 2023-05-01:
x86 内核5.15 最新的，全新刷入硬路由 硬件配置4125 ，i225 4口，问题入题，不能自动获取ip地址，这个问题好像去年就有了，今天又在线编译了下，还是这个问题，，不知道是不是我编译配置问题。 这个是重启后日志如下：

Mon May  1 17:04:08 2023 kern.info kernel: [   16.840672] usbcore: registered new interface driver usblp
Mon May  1 17:04:08 2023 kern.info kernel: [   16.854906] xt_time: kernel timezone is -0000
Mon May  1 17:04:08 2023 kern.info kernel: [   17.011794] [drm] amdgpu kernel modesetting enabled.
Mon May  1 17:04:08 2023 kern.info kernel: [   17.018382] usbcore: registered new interface driver aqc111
Mon May  1 17:04:08 2023 kern.info kernel: [   17.024801] usbcore: registered new interface driver ax88179_178a
Mon May  1 17:04:08 2023 kern.info kernel: [   17.032184] PPP generic driver version 2.4.2
Mon May  1 17:04:08 2023 kern.info kernel: [   17.037120] PPP MPPE Compression module registered
Mon May  1 17:04:08 2023 kern.info kernel: [   17.042537] NET: Registered PF_PPPOX protocol family
Mon May  1 17:04:08 2023 user.info kernel: [   17.048548] kmodloader: done loading kernel modules from /etc/modules.d/*
Mon May  1 17:04:08 2023 kern.info kernel: [   18.293284] EXT4-fs (sda3): mounted filesystem with ordered data mode. Opts: . Quota mode: disabled.
Mon May  1 17:04:08 2023 daemon.notice procd: /etc/rc.d/S13openssl: Generating engines.cnf
Mon May  1 17:04:08 2023 daemon.notice procd: /etc/rc.d/S13openssl: Enabling engine devcrypto
Mon May  1 17:04:08 2023 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Mon May  1 17:04:08 2023 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Mon May  1 17:04:09 2023 user.notice : Added device handler type: 8021ad
Mon May  1 17:04:09 2023 user.notice : Added device handler type: 8021q
Mon May  1 17:04:09 2023 user.notice : Added device handler type: macvlan
Mon May  1 17:04:09 2023 user.notice : Added device handler type: veth
Mon May  1 17:04:09 2023 user.notice : Added device handler type: bridge
Mon May  1 17:04:09 2023 user.notice : Added device handler type: Network device
Mon May  1 17:04:09 2023 user.notice : Added device handler type: tunnel
Mon May  1 17:04:09 2023 daemon.notice procd: /etc/rc.d/S50autoreboot: Command failed: ubus call service delete { "name": "cron" } (Not found)
Mon May  1 17:04:09 2023 daemon.notice procd: /etc/rc.d/S50autoreboot: Auto REBOOT has started.
Mon May  1 17:04:09 2023 cron.err crond[2004]: crond (busybox 1.36.0) started, log level 5
Mon May  1 17:04:09 2023 kern.info kernel: [   19.947852] br-lan: port 1(eth0) entered blocking state
Mon May  1 17:04:09 2023 kern.info kernel: [   19.953515] br-lan: port 1(eth0) entered disabled state
Mon May  1 17:04:09 2023 kern.info kernel: [   19.959236] device eth0 entered promiscuous mode
Mon May  1 17:04:09 2023 daemon.notice netifd: Interface 'lan' is enabled
Mon May  1 17:04:09 2023 daemon.notice netifd: Interface 'lan' is setting up now
Mon May  1 17:04:09 2023 daemon.notice netifd: Interface 'lan' is now up
Mon May  1 17:04:09 2023 daemon.notice netifd: Interface 'loopback' is enabled
Mon May  1 17:04:09 2023 daemon.notice netifd: Interface 'loopback' is setting up now
Mon May  1 17:04:09 2023 daemon.notice netifd: Interface 'loopback' is now up
Mon May  1 17:04:09 2023 daemon.notice netifd: Interface 'wan' is enabled
Mon May  1 17:04:09 2023 daemon.notice netifd: Network device 'lo' link is up
Mon May  1 17:04:09 2023 daemon.notice netifd: Interface 'loopback' has link connectivity
Mon May  1 17:04:09 2023 user.notice firewall: Reloading firewall due to ifup of lan (br-lan)
Mon May  1 17:04:09 2023 cron.err crond[2552]: crond (busybox 1.36.0) started, log level 5
Mon May  1 17:04:10 2023 daemon.info avahi-daemon[2348]: Found user 'nobody' (UID 65534) and group 'nogroup' (GID 65534).
Mon May  1 17:04:10 2023 daemon.info avahi-daemon[2348]: Successfully dropped root privileges.
Mon May  1 17:04:10 2023 daemon.info avahi-daemon[2348]: avahi-daemon 0.8 starting up.
Mon May  1 17:04:10 2023 daemon.warn avahi-daemon[2348]: WARNING: No NSS support for mDNS detected, consider installing nss-mdns!
Mon May  1 17:04:10 2023 daemon.info avahi-daemon[2348]: No service file found in /etc/avahi/services.
Mon May  1 17:04:10 2023 daemon.info avahi-daemon[2348]: Joining mDNS multicast group on interface br-lan.IPv6 with address fddb:8669:ebca::1.
Mon May  1 17:04:10 2023 daemon.info avahi-daemon[2348]: New relevant interface br-lan.IPv6 for mDNS.
Mon May  1 17:04:10 2023 daemon.info avahi-daemon[2348]: Joining mDNS multicast group on interface br-lan.IPv4 with address 192.168.0.1.
Mon May  1 17:04:10 2023 daemon.info avahi-daemon[2348]: New relevant interface br-lan.IPv4 for mDNS.
Mon May  1 17:04:10 2023 daemon.info avahi-daemon[2348]: Joining mDNS multicast group on interface lo.IPv6 with address ::1.
Mon May  1 17:04:10 2023 daemon.info avahi-daemon[2348]: New relevant interface lo.IPv6 for mDNS.
Mon May  1 17:04:10 2023 daemon.info avahi-daemon[2348]: Joining mDNS multicast group on interface lo.IPv4 with address 127.0.0.1.
Mon May  1 17:04:10 2023 daemon.info avahi-daemon[2348]: New relevant interface lo.IPv4 for mDNS.
Mon May  1 17:04:10 2023 daemon.info avahi-daemon[2348]: Network interface enumeration completed.
Mon May  1 17:04:10 2023 daemon.info avahi-daemon[2348]: Registering new address record for fddb:8669:ebca::1 on br-lan.*.
Mon May  1 17:04:10 2023 daemon.info avahi-daemon[2348]: Registering new address record for 192.168.0.1 on br-lan.IPv4.
Mon May  1 17:04:10 2023 daemon.info avahi-daemon[2348]: Registering new address record for ::1 on lo.*.
Mon May  1 17:04:10 2023 daemon.info avahi-daemon[2348]: Registering new address record for 127.0.0.1 on lo.IPv4.
Mon May  1 17:04:10 2023 daemon.info avahi-daemon[2348]: Registering HINFO record with values 'X86_64'/'LINUX'.
Mon May  1 17:04:10 2023 user.notice nlbwmon: Reloading nlbwmon due to ifup of lan (br-lan)
Mon May  1 17:04:10 2023 user.notice upnp daemon: external interface not found, not starting
Mon May  1 17:04:10 2023 daemon.err miniupnpd[2768]: could not open lease file: /var/upnp.leases
Mon May  1 17:04:10 2023 daemon.notice miniupnpd[2768]: HTTP listening on port 5000
Mon May  1 17:04:10 2023 daemon.notice miniupnpd[2768]: Listening for NAT-PMP/PCP traffic on port 5351
Mon May  1 17:04:10 2023 daemon.info nginx_init: using /etc/nginx/uci.conf (the test is ok)
Mon May  1 17:04:10 2023 cron.err crond[2947]: crond (busybox 1.36.0) started, log level 5
Mon May  1 17:04:10 2023 user.notice ucitrack: Setting up /etc/config/network reload dependency on /etc/config/dhcp
Mon May  1 17:04:10 2023 user.notice ucitrack: Setting up /etc/config/network reload dependency on /etc/config/radvd
Mon May  1 17:04:10 2023 user.notice ucitrack: Setting up /etc/config/wireless reload dependency on /etc/config/network
Mon May  1 17:04:10 2023 user.notice ucitrack: Setting up /etc/config/firewall reload dependency on /etc/config/luci-splash
[... 234 lines cut ...]
Mon May  1 17:04:37 2023 daemon.info netdata[8178]: NETDATA_HOST_OS_ID=openwrt
Mon May  1 17:04:37 2023 daemon.info netdata[8178]: NETDATA_HOST_OS_ID_LIKE=lede openwrt
Mon May  1 17:04:37 2023 daemon.info netdata[8178]: NETDATA_HOST_OS_VERSION=SNAPSHOT
Mon May  1 17:04:37 2023 daemon.info netdata[8178]: NETDATA_HOST_OS_VERSION_ID=snapshot
Mon May  1 17:04:37 2023 daemon.info netdata[8178]: NETDATA_HOST_OS_DETECTION=/etc/os-release
Mon May  1 17:04:37 2023 daemon.info netdata[8178]: NETDATA_HOST_IS_K8S_NODE=false
Mon May  1 17:04:37 2023 daemon.info netdata[8178]: NETDATA_SYSTEM_KERNEL_NAME=Linux
Mon May  1 17:04:37 2023 daemon.info netdata[8178]: NETDATA_SYSTEM_KERNEL_VERSION=5.15.105
Mon May  1 17:04:37 2023 daemon.info netdata[8178]: NETDATA_SYSTEM_ARCHITECTURE=x86_64
Mon May  1 17:04:37 2023 daemon.info netdata[8178]: NETDATA_SYSTEM_VIRTUALIZATION=none
Mon May  1 17:04:37 2023 daemon.info netdata[8178]: NETDATA_SYSTEM_VIRT_DETECTION=none
Mon May  1 17:04:37 2023 daemon.info netdata[8178]: NETDATA_SYSTEM_CONTAINER=unknown
Mon May  1 17:04:37 2023 daemon.info netdata[8178]: NETDATA_SYSTEM_CONTAINER_DETECTION=none
Mon May  1 17:04:37 2023 daemon.info netdata[8178]: NETDATA_SYSTEM_CPU_LOGICAL_CPU_COUNT=4
Mon May  1 17:04:37 2023 daemon.info netdata[8178]: NETDATA_SYSTEM_CPU_VENDOR=GenuineIntel
Mon May  1 17:04:37 2023 daemon.info netdata[8178]: NETDATA_SYSTEM_CPU_MODEL=Intel(R) Celeron(R) J4125 CPU @ 2.00GHz
Mon May  1 17:04:37 2023 daemon.info netdata[8178]: NETDATA_SYSTEM_CPU_FREQ=2700000
[... cut ...]
Mon May  1 17:04:37 2023 daemon.info netdata[8178]: NETDATA_SYSTEM_TOTAL_DISK_SIZE=120040980480
Mon May  1 17:04:37 2023 daemon.info netdata[8178]: NETDATA_SYSTEM_DISK_DETECTION=sysfs
Mon May  1 17:04:37 2023 daemon.info netdata[8178]: SQLite database /var/cache/netdata/netdata-meta.db initialization
Mon May  1 17:04:37 2023 daemon.info netdata[8178]: SQLite database initialization completed
Mon May  1 17:04:37 2023 daemon.info netdata[8178]: CONFIG: cannot load user config '/etc/netdata/stream.conf'. Will try stock config.

nb12nb34, 2023-05-01:
Mon May  1 17:04:37 2023 daemon.info netdata[8178]: Host 'OpenWrt' (at registry as 'OpenWrt') with guid '32ed103a-e7ff-11ed-a75b-60beb400613c' initialized, os 'linux', timezone 'UTC', tags '', program_name 'netdata', program_version '20200416-3476-g7c84256de', update every 2, memory mode save, history entries 3996, streaming disabled (to '' with api key ''), health disabled, cache_dir '/var/cache/netdata', varlib_dir '/var/lib/netdata', health_log '/var/lib/netdata/health/health-log.db', alarms default handler '/usr/lib/netdata/plugins.d/alarm-notify.sh', alarms default recipient 'root'
Mon May  1 17:04:37 2023 daemon.err netdata[8178]: child pid 8825 exited with code 127.
Mon May  1 17:04:37 2023 daemon.err netdata[8178]: /usr/lib/netdata/plugins.d/get-kubernetes-labels.sh exited abnormally. No kubernetes labels will be added to the host.
Mon May  1 17:04:37 2023 daemon.err netdata[8178]: PROCFILE: Cannot open file '/etc/netdata/.environment'
Mon May  1 17:04:37 2023 cron.err crond[8850]: crond (busybox 1.36.0) started, log level 5
Mon May  1 17:04:37 2023 daemon.info netdata[8178]: thread created with task id 8893
Mon May  1 17:04:37 2023 daemon.info netdata[8178]: set name of thread 8893 to PLUGIN[idlejitt
Mon May  1 17:04:37 2023 daemon.info netdata[8178]: thread created with task id 8896
Mon May  1 17:04:37 2023 daemon.info netdata[8178]: set name of thread 8896 to PLUGINSD
Mon May  1 17:04:37 2023 daemon.info netdata[8178]: thread created with task id 8894
Mon May  1 17:04:37 2023 daemon.info netdata[8178]: set name of thread 8894 to GLOBAL_STATS
Mon May  1 17:04:37 2023 daemon.info netdata[8178]: thread created with task id 8898
Mon May  1 17:04:37 2023 daemon.info netdata[8178]: set name of thread 8898 to STATSD
Mon May  1 17:04:37 2023 daemon.info netdata[8178]: thread created with task id 8899
Mon May  1 17:04:37 2023 daemon.info netdata[8178]: set name of thread 8899 to BACKENDS
Mon May  1 17:04:37 2023 daemon.info netdata[8178]: Initializing command server.
Mon May  1 17:04:37 2023 daemon.info netdata[8178]: thread created with task id 8901
Mon May  1 17:04:37 2023 daemon.info netdata[8178]: set name of thread 8901 to WEB_SERVER[stat
Mon May  1 17:04:37 2023 daemon.info netdata[8178]: thread created with task id 8904
Mon May  1 17:04:37 2023 daemon.info netdata[8178]: set name of thread 8904 to PLUGIN[diskspac
Mon May  1 17:04:37 2023 daemon.info netdata[8178]: thread created with task id 8897
Mon May  1 17:04:37 2023 daemon.info netdata[8178]: set name of thread 8897 to SERVICE
Mon May  1 17:04:37 2023 daemon.info netdata[8178]: thread created with task id 8902
Mon May  1 17:04:37 2023 daemon.info netdata[8178]: thread created with task id 8905
Mon May  1 17:04:37 2023 daemon.info netdata[8178]: starting worker 2
Mon May  1 17:04:37 2023 daemon.info netdata[8178]: set name of thread 8905 to PLUGIN[proc]
Mon May  1 17:04:37 2023 daemon.info netdata[8178]: starting worker 3
Mon May  1 17:04:37 2023 daemon.info netdata[8178]: cleaning up...
Mon May  1 17:04:37 2023 daemon.info netdata[8178]: thread with task id 8899 finished
Mon May  1 17:04:37 2023 daemon.info netdata[8178]: thread created with task id 8907
Mon May  1 17:04:37 2023 daemon.info netdata[8178]: set name of thread 8907 to WEB_SERVER[stat
Mon May  1 17:04:37 2023 daemon.info netdata[8178]: thread created with task id 8908
Mon May  1 17:04:37 2023 daemon.info netdata[8178]: POLLFD: LISTENER: listening on 'tcp:0.0.0.0:19999'
Mon May  1 17:04:37 2023 daemon.info netdata[8178]: set name of thread 8908 to WEB_SERVER[stat
Mon May  1 17:04:37 2023 daemon.info netdata[8178]: POLLFD: LISTENER: listening on 'tcp:0.0.0.0:19999'
Mon May  1 17:04:37 2023 daemon.info netdata[8178]: POLLFD: LISTENER: listening on 'tcp:[::]:19999'
Mon May  1 17:04:37 2023 daemon.info netdata[8178]: starting worker 4
Mon May  1 17:04:37 2023 daemon.info netdata[8178]: POLLFD: LISTENER: listening on 'tcp:0.0.0.0:19999'
Mon May  1 17:04:37 2023 daemon.info netdata[8178]: POLLFD: LISTENER: listening on 'tcp:[::]:19999'
Mon May  1 17:04:37 2023 daemon.info netdata[8178]: set name of thread 8902 to PLUGIN[timex]
Mon May  1 17:04:37 2023 daemon.info netdata[8178]: thread created with task id 8911
Mon May  1 17:04:37 2023 daemon.info netdata[8178]: set name of thread 8911 to WEB_SERVER[stat
Mon May  1 17:04:37 2023 daemon.info netdata[8178]: thread created with task id 8903
Mon May  1 17:04:37 2023 daemon.info netdata[8178]: set name of thread 8903 to PLUGIN[tc]
Mon May  1 17:04:37 2023 daemon.info netdata[8178]: POLLFD: LISTENER: listening on 'tcp:[::]:19999'
Mon May  1 17:04:37 2023 daemon.info netdata[8178]: netdata initialization completed. Enjoy real-time performance monitoring!
Mon May  1 17:04:37 2023 daemon.err netdata[8178]: This agent doesn't have ACLK.
Mon May  1 17:04:37 2023 daemon.info netdata[8178]: thread created with task id 8895
Mon May  1 17:04:37 2023 daemon.info netdata[8178]: set name of thread 8895 to HEALTH
Mon May  1 17:04:37 2023 daemon.info netdata[8178]: POLLFD: LISTENER: listening on 'tcp:0.0.0.0:19999'
Mon May  1 17:04:37 2023 daemon.info netdata[8178]: POLLFD: LISTENER: listening on 'tcp:[::]:19999'
Mon May  1 17:04:37 2023 daemon.info netdata[8178]: Initializing file /var/cache/netdata/system.idlejitter/main.db.
Mon May  1 17:04:37 2023 daemon.info netdata[8178]: thread created with task id 8910
Mon May  1 17:04:37 2023 daemon.info netdata[8178]: set name of thread 8910 to PLUGINSD[perf]
[... cut ...]
Mon May  1 17:05:04 2023 kern.err kernel: [   74.990594] exFAT-fs (sda128): failed to recognize exfat type
Mon May  1 17:06:52 2023 daemon.err uhttpd[2135]: sh: out of range
Mon May  1 17:06:52 2023 daemon.err uhttpd[2135]: sh: out of range
Mon May  1 17:07:40 2023 daemon.notice miniupnpd[6681]: shutting down MiniUPnPd
Mon May  1 17:07:40 2023 daemon.err uhttpd[2135]: Command failed: ubus call service delete { "name": "miniupnpd" } (Not found)
Mon May  1 17:08:02 2023 daemon.err uhttpd[2135]: Command failed: ubus call service delete { "name": "miniupnpd" } (Not found)

zsdbbn, 2023-05-01:
一样的问题

coolsnowwolf, 2026-09-05:
日志中可见 i225 的物理链路已建立，且同时存在 PPPoE、nginx 端口冲突和多个自定义服务报错；issue 没有提供 `/etc/config/network`、接口实际协议（DHCP/PPPoE）、`ubus call network.interface.wan status` 或 DHCP/PPPoE 失败日志，无法判断是配置、上游网络还是驱动问题。当前没有可验证的代码修复，补充最小配置和完整网络日志后可重新提交。

closed 2026-09-05 by coolsnowwolf