coolsnowwolf/lede #11176: SSR Plus启动后dnsmasq无法启动

Landidi666, 2023-05-02:
已经有了一个issue和我有一样的问题#5335 
但是并没有最终的解决方案，故有此issue

SSR Plus在启动时系统日志如下：
_Wed May  3 02:58:06 2023 cron.err crond[6220]: crond (busybox 1.36.0) started, log level 5
Wed May  3 02:58:17 2023 daemon.info dnsmasq[6005]: exiting on receipt of SIGTERM
Wed May  3 02:58:21 2023 daemon.crit dnsmasq[6879]: recompile with HAVE_IPSET defined to enable ipset directives at line 1 of /tmp/dnsmasq.ssr/blacklist_forward.conf
Wed May  3 02:58:21 2023 daemon.crit dnsmasq[6879]: FAILED to start up_
Dnsmasq会一直崩溃，导致SSR Plus启动后无法访问网络
另外，L大在#5335里提到了“DNSMASQ 没有选上 IPSET 的支持“，那该怎么选上呢？

Dnsmasq：dnsmasq-full	2.89-99
路由器：Redmi AX6S
固件：ImmortalWrt SNAPSHOT r26026-fb5f368275

minlang112, 2023-05-03:
但凡谷歌一下就会知道……原版dnsmasq并不支持ipset，需要更换为dnsmasq-full

Landidi666, 2023-05-03:
> 但凡谷歌一下就会知道……原版dnsmasq并不支持ipset，需要更换为dnsmasq-full

我已经写了Dnsmasq：dnsmasq-full 2.89-99，如果只是因为用的原版的话我就不会大费周章地发issue了

hcym, 2023-05-03:
这个出问题，路由器就是一块石头。我的一直正常，就是默认设置，偶尔也会用6.1内核编译

rrkai, 2023-05-03:
SSR+我觉得bug不少.还是helloword好用

closed 2023-05-03 by coolsnowwolf