coolsnowwolf/lede #11177: 编译的固件luci 网络>接口 没有设备选项，无法通过luci配置vlan。请教编译时需要在哪里设置启用。

dongc123, 2023-05-03:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [X ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：编译的固件luci 网络>接口 没有设备选项，无法通过luci配置vlan。请教编译时需要在哪里设置启用。附原版openwrt

### (2) 路由器型号和固件版本

A：软路由

### (3) 详细日志

A：
[image]

iam993, 2023-05-03:
什么平台，你装原版的有设备选项吗？

dongc123, 2023-05-03:
官方原版的有
[image]

iam993, 2023-05-03:
在此页面，您可以配置网络接口。您可以勾选“桥接接口”，并输入由空格分隔的多个网络接口的名称来桥接多个接口。接口名称中可以使用 VLAN 记号 INTERFACE.VLANNR（例如：eth0.1）。

这是接口配置里写着的原话，不懂的可以先问问百度。

eagle-lk, 2023-05-07:
设备的具体型号，版本？OpenWrt主线架构已经采用DSA，所以不再提供交换机配置。需要配置交换机VLAN的需要修改代码。

dongc123, 2023-05-07:
设备是x86的，编译的最新版，请问需要编译时修改哪里了？

eagle-lk, 2023-05-08:
我没编译过x86的版本，我用的型号是Newifi D1，默认是采用DSA架构，网络菜单中是没有“交换机”菜单项的。要开启交换机菜单，需要修改设备的dts文件，以及设备相关的启动文件，对应Newifi D1，这两个文件分别是：
lede/target/linux/ramips/dts/mt7621_lenovo_newifi-d1.dts
lede/target/linux/ramips/mt7621/base-files/etc/board.d/02_network
已经开启交换机配置的都在02_network中初始化是增加了ucidef_add_switch "switch0" 命令，同时dts需要匹配。

eagle-lk, 2023-05-08:
> 设备是x86的，编译的最新版，请问需要编译时修改哪里了？

抱歉，可能误解你的意思了。我有一台AX6s，使用默认的编译设置，Openwrt主线版本，kernel 5.15.110，你说的配置是网络接口br-lan的默认配置。

shenlijun, 2023-05-17:
luci的问题吧，lean这边的好像没有你要的那个界面

Nicepaul, 2023-05-22:
同问

dongc123, 2023-05-22:
> 

lean openwrt luci是18版的没有vlan配置，需要这个智能换原版的openwrt.

feifei435, 2023-11-28:
@dongc123 没有别的解决办法了吗，可以升级luci吗

coolsnowwolf, 2026-09-05:
该现象取决于目标硬件使用的网络架构：旧 swconfig 设备在“网络→交换机”配置 VLAN，DSA 设备则在“网络→接口”编辑网桥并在“设备”字段选择端口/创建 `br-lan.X`。软路由未提供型号、目标 profile 或 `/etc/config/network`，无法判断是 DSA 正常界面差异、未安装 `luci-mod-network`，还是自定义配置问题。当前 LuCI 已支持 DSA VLAN 配置；请安装 `luci-mod-network` 并按目标架构配置，补充具体硬件后再反馈。关闭该咨询 issue。

closed 2026-09-05 by coolsnowwolf