coolsnowwolf/lede #11182: 更新源码后，无线不可用

mitsukileung, 2023-05-05:
Ax6 K3

[image]

AndyLin2020, 2023-05-06:
ASUS RT-ACRH17也一样

Y-ZD-heheka, 2023-05-06:
7921＋1

xfmly, 2023-05-06:
rac2v1k +1
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.4.242 (runner@fv-az246-371) (gcc version 11.3.0 (OpenWrt GCC 11.3.0 r5946-b6e9dac2c)) #0 SMP Sat May 6 03:29:11 2023
[    0.000000] CPU: ARMv7 Processor [512f04d0] revision 0 (ARMv7), cr=10c5787d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
[    0.000000] OF: fdt: Machine model: Askey RT4230W REV6
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] On node 0 totalpages: 253952
[    0.000000]   Normal zone: 1728 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 196608 pages, LIFO batch:63
[    0.000000]   HighMem zone: 57344 pages, LIFO batch:15
[    0.000000] percpu: Embedded 15 pages/cpu s30348 r8192 d22900 u61440
[    0.000000] pcpu-alloc: s30348 r8192 d22900 u61440 alloc=15*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 252224
[    0.000000] Kernel command line: rootfstype=squashfs noinitrd
[    0.000000] Bootloader command line (ignored): 
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 996172K/1015808K available (6447K kernel code, 198K rwdata, 1552K rodata, 1024K init, 231K bss, 19636K reserved, 0K cma-reserved, 229376K highmem)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=2, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] clocksource: dg_timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 305801671480 ns
[    0.000006] sched_clock: 32 bits at 6MHz, resolution 160ns, wraps every 343597383600ns
[    0.000020] Switching to timer-based delay loop, resolution 160ns
[    0.000343] Calibrating delay loop (skipped), value calculated using timer frequency.. 12.50 BogoMIPS (lpj=62500)
[    0.000365] pid_max: default: 32768 minimum: 301
[    0.000562] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.000583] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.001628] CPU: Testing write buffer coherency: ok
[    0.002956] Setting up static identity map for 0x42300000 - 0x42300060
[    0.003162] rcu: Hierarchical SRCU implementation.
[    0.003745] smp: Bringing up secondary CPUs ...
[    0.005743] smp: Brought up 1 node, 2 CPUs
[    0.005756] SMP: Total of 2 processors activated (25.00 BogoMIPS).
[    0.005765] CPU: All CPU(s) started in SVC mode.
[    0.017044] VFP support v0.3: implementor 51 architecture 64 part 4d variant 2 rev 0
[    0.017204] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.017234] futex hash table entries: 512 (order: 3, 32768 bytes, linear)
[    0.017390] pinctrl core: initialized pinctrl subsystem
[    0.019062] NET: Registered protocol family 16
[    0.019390] DMA: preallocated 256 KiB pool for atomic coherent allocations
[    0.020825] cpuidle: using governor ladder
[    0.020893] cpuidle: using governor menu
[    0.056744] usbcore: registered new interface driver usbfs
[    0.056827] usbcore: registered new interface driver hub
[    0.056899] usbcore: registered new device driver usb
[    0.056960] pps_core: LinuxPPS API ver. 1 registered
[    0.056972] pps_core: Software ver. 5.3.6 - Copyright 2005-2007 Rodolfo Giometti <giometti@linux.it>
[    0.057005] PTP clock support registered
[    0.057855] workqueue: max_active 576 requested for napi_workq is out of range, clamping between 1 and 512
[    0.059088] clocksource: Switched to clocksource dg_timer
[    0.059977] thermal_sys: Registered thermal governor 'step_wise'
[    0.062490] NET: Registered protocol family 2
[    0.062746] IP idents hash table entries: 16384 (order: 5, 131072 bytes, linear)
[    0.063971] tcp_listen_portaddr_hash hash table entries: 512 (order: 0, 6144 bytes, linear)
[    0.064020] TCP established hash table entries: 8192 (order: 3, 32768 bytes, linear)
[    0.064079] TCP bind hash table entries: 8192 (order: 4, 65536 bytes, linear)
[    0.064172] TCP: Hash tables configured (established 8192 bind 8192)
[    0.064278] UDP hash table entries: 512 (order: 2, 16384 bytes, linear)
[    0.064318] UDP-Lite hash table entries: 512 (order: 2, 16384 bytes, linear)
[    0.064603] NET: Registered protocol family 1
[    0.064654] PCI: CLS 0 bytes, default 64
[    0.066025] workingset: timestamp_bits=14 max_order=18 bucket_order=4
[    0.072712] squashfs: version 4.0 (2009/01/31) Phillip Lougher
[... 366 lines cut ...]
[   83.108313] add stats
[   83.690551] ipq8064-mdio 37000000.mdio eth1: 1000 Mbps Full Duplex
[   84.819545] IPv6: ADDRCONF(NETDEV_CHANGE): eth1: link becomes ready
[   84.821014] br-lan: port 1(eth1.1) entered blocking state
[   84.824698] br-lan: port 1(eth1.1) entered disabled state
[   84.830677] device eth1.1 entered promiscuous mode
[   84.835566] device eth1 entered promiscuous mode
[   84.847176] br-lan: port 1(eth1.1) entered blocking state
[   84.847199] br-lan: port 1(eth1.1) entered forwarding state
[... cut ...]
[  100.260887] ath10k_pci 0000:01:00.0: 10.4 wmi init: vdevs: 16  peers: 48  tid: 96
[  100.260913] ath10k_pci 0000:01:00.0: msdu-desc: 2500  skid: 32
[  100.343103] ath10k_pci 0000:01:00.0: wmi print 'P 48/48 V 16 K 144 PH 176 T 186  msdu-desc: 2500  sw-crypt: 0 ct-sta: 0'
[  100.343944] ath10k_pci 0000:01:00.0: wmi print 'free: 84920 iram: 13156 sram: 11224'
[  100.728956] ath10k_pci 0000:01:00.0: rts threshold -1
[  100.730583] ath10k_pci 0000:01:00.0: Firmware lacks feature flag indicating a retry limit of > 2 is OK, requested limit: 4

btguigui, 2023-05-07:
MobiPromo CM520-79F，更新后无线也挂了

zhdot, 2023-05-07:
Linksys EA8500 也挂了

Y-ZD-heheka, 2023-05-07:
> 7921＋1

今天更新源码后7921、7922可以用了，但QCNFA765依然不能用

fengjiaqi927, 2023-05-10:
mips mt7628dan has the same error on wireless

particlewong, 2023-05-11:
竞斗云也炸了，今天最新源码依旧无线不能用

zwillhill, 2023-05-12:
同k3，还以为是云编译改配置出问题了，刷了个四月份云编译版本，有无线，但是有线挂了

mitsukileung, 2023-05-12:
> 同k3，还以为是云编译改配置出问题了，刷了个四月份云编译版本，有无线，但是有线挂了

我的4月份有线没有问题

mitsukileung, 2023-05-14:
ipq807x / Ax6

跳过这个PR https://github.com/coolsnowwolf/lede/commit/31673339464df901c9c5a70dcc8ba00953149ab1 可以顺利编译，并且WIFI正常了

zhdot, 2023-05-15:
> ipq807x / Ax6 Ax3600
> 
> 跳过这个PR [3167333](https://github.com/coolsnowwolf/lede/commit/31673339464df901c9c5a70dcc8ba00953149ab1)可以顺势编译，并且WIFI正常了
> 
> [image]
> 
> [image]

编译时 怎么跳过呢？

fengqi, 2023-05-16:
> > ipq807x / Ax6 Ax3600
> > 跳过这个PR [3167333](https://github.com/coolsnowwolf/lede/commit/31673339464df901c9c5a70dcc8ba00953149ab1)可以顺势编译，并且WIFI正常了
> > [image]
> > [image]
> 
> 编译时 怎么跳过呢？

试试 git revert 31673339464df901c9c5a70dcc8ba00953149ab1

zhdot, 2023-05-16:
> > > ipq807x / Ax6 Ax3600
> > > 跳过这个PR [3167333](https://github.com/coolsnowwolf/lede/commit/31673339464df901c9c5a70dcc8ba00953149ab1)可以顺势编译，并且WIFI正常了
> > > [image]
> > > [image]
> > 
> > 
> > 编译时 怎么跳过呢？
> 
> 试试 `git revert 31673339464df901c9c5a70dcc8ba00953149ab1`

error: could not revert 316733394... mac80211: update to v6.1.24
hint: after resolving the conflicts, mark the corrected paths
hint: with 'git add <paths>' or 'git rm <paths>'
hint: and commit the result with 'git commit'

报错了 小白一个 不知道怎么弄啊

taibei123, 2023-05-18:
acrh17同样问题，不可用

mitsukileung, 2023-05-18:
重新拉库不带设置文件再编译出来的固件没有“无线”的选项。

系统日志：

Wed May 17 13:25:26 2023 kern.info kernel: [    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
Wed May 17 13:25:26 2023 kern.notice kernel: [    0.000000] Linux version 5.15.110 (koysuke@koysuke) (aarch64-openwrt-linux-musl-gcc (OpenWrt GCC 11.3.0 r5971-a6f1cc5b5) 11.3.0, GNU ld (GNU Binutils) 2.37) #0 SMP Wed May 17 05:25:11 2023
Wed May 17 13:25:26 2023 kern.info kernel: [    0.000000] Machine model: Redmi AX6
Wed May 17 13:25:26 2023 kern.info kernel: [    0.000000] Zone ranges:
Wed May 17 13:25:26 2023 kern.info kernel: [    0.000000]   DMA      [mem 0x0000000040000000-0x000000005fffffff]
Wed May 17 13:25:26 2023 kern.info kernel: [    0.000000]   DMA32    empty
Wed May 17 13:25:26 2023 kern.info kernel: [    0.000000]   Normal   empty
Wed May 17 13:25:26 2023 kern.info kernel: [    0.000000] Movable zone start for each node
Wed May 17 13:25:26 2023 kern.info kernel: [    0.000000] Early memory node ranges
Wed May 17 13:25:26 2023 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000040000000-0x0000000040ffffff]
Wed May 17 13:25:26 2023 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000041000000-0x000000004a3fffff]
Wed May 17 13:25:26 2023 kern.info kernel: [    0.000000]   node   0: [mem 0x000000004a400000-0x000000004a4fffff]
Wed May 17 13:25:26 2023 kern.info kernel: [    0.000000]   node   0: [mem 0x000000004a500000-0x000000004a5fffff]
Wed May 17 13:25:26 2023 kern.info kernel: [    0.000000]   node   0: [mem 0x000000004a600000-0x000000004e8fffff]
Wed May 17 13:25:26 2023 kern.info kernel: [    0.000000]   node   0: [mem 0x000000004e900000-0x000000005fffffff]
Wed May 17 13:25:26 2023 kern.info kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000040000000-0x000000005fffffff]
Wed May 17 13:25:26 2023 kern.info kernel: [    0.000000] psci: probing for conduit method from DT.
Wed May 17 13:25:26 2023 kern.info kernel: [    0.000000] psci: PSCIv1.0 detected in firmware.
Wed May 17 13:25:26 2023 kern.info kernel: [    0.000000] psci: Using standard PSCI v0.2 function IDs
Wed May 17 13:25:26 2023 kern.info kernel: [    0.000000] psci: MIGRATE_INFO_TYPE not supported.
Wed May 17 13:25:26 2023 kern.info kernel: [    0.000000] psci: SMC Calling Convention v1.0
Wed May 17 13:25:26 2023 kern.info kernel: [    0.000000] psci: OSI mode supported.
Wed May 17 13:25:26 2023 kern.info kernel: [    0.000000] percpu: Embedded 17 pages/cpu s30424 r8192 d31016 u69632
Wed May 17 13:25:26 2023 kern.debug kernel: [    0.000000] pcpu-alloc: s30424 r8192 d31016 u69632 alloc=17*4096
Wed May 17 13:25:26 2023 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
Wed May 17 13:25:26 2023 kern.info kernel: [    0.000000] Detected VIPT I-cache on CPU0
Wed May 17 13:25:26 2023 kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 129024
Wed May 17 13:25:26 2023 kern.notice kernel: [    0.000000] Kernel command line: console=ttyMSM0,115200n8 ubi.mtd=rootfs root=mtd:ubi_rootfs rootfstype=squashfs rootwait root=/dev/ubiblock0_1
Wed May 17 13:25:26 2023 kern.info kernel: [    0.000000] Dentry cache hash table entries: 65536 (order: 7, 524288 bytes, linear)
Wed May 17 13:25:26 2023 kern.info kernel: [    0.000000] Inode-cache hash table entries: 32768 (order: 6, 262144 bytes, linear)
Wed May 17 13:25:26 2023 kern.info kernel: [    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
Wed May 17 13:25:26 2023 kern.info kernel: [    0.000000] Memory: 415648K/524288K available (7680K kernel code, 896K rwdata, 2032K rodata, 1216K init, 288K bss, 108640K reserved, 0K cma-reserved)
Wed May 17 13:25:26 2023 kern.info kernel: [    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
Wed May 17 13:25:26 2023 kern.info kernel: [    0.000000] rcu: Hierarchical RCU implementation.
Wed May 17 13:25:26 2023 kern.info kernel: [    0.000000] 	Tracing variant of Tasks RCU enabled.
Wed May 17 13:25:26 2023 kern.info kernel: [    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
Wed May 17 13:25:26 2023 kern.info kernel: [    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
Wed May 17 13:25:26 2023 kern.info kernel: [    0.000000] Root IRQ handler: gic_handle_irq
Wed May 17 13:25:26 2023 kern.info kernel: [    0.000000] GICv2m: range[mem 0x0b00a000-0x0b00affc], SPI[448:479]
Wed May 17 13:25:26 2023 kern.info kernel: [    0.000000] arch_timer: cp15 and mmio timer(s) running at 19.20MHz (virt/virt).
Wed May 17 13:25:26 2023 kern.info kernel: [    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
Wed May 17 13:25:26 2023 kern.info kernel: [    0.000000] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
Wed May 17 13:25:26 2023 kern.info kernel: [    0.000187] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
Wed May 17 13:25:26 2023 kern.info kernel: [    0.000204] pid_max: default: 32768 minimum: 301
Wed May 17 13:25:26 2023 kern.info kernel: [    0.000353] Mount-cache hash table entries: 1024 (order: 1, 8192 bytes, linear)
Wed May 17 13:25:26 2023 kern.info kernel: [    0.000369] Mountpoint-cache hash table entries: 1024 (order: 1, 8192 bytes, linear)
Wed May 17 13:25:26 2023 kern.info kernel: [    0.002055] rcu: Hierarchical SRCU implementation.
[... cut ...]
[   31.874782] br-lan: port 2(eth2) entered disabled state
[   31.879056] device eth2 entered promiscuous mode
[   31.887681] br-lan: port 3(eth3) entered blocking state
[   31.888847] br-lan: port 3(eth3) entered disabled state
[   31.894133] device eth3 entered promiscuous mode
[  264.792247] nss-dp 3a001400.dp3 eth1: PHY Link up speed: 1000
[  264.792336] br-lan: port 1(eth1) entered blocking state
[  264.796987] br-lan: port 1(eth1) entered forwarding state
[  264.802388] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready

fichenx, 2023-05-26:
+1
5月4日编译时没问题，5月4日以后一直到现在编译的都没有无线

zclkkk, 2023-05-26:
是不是这个patchhttps://github.com/coolsnowwolf/lede/commit/aa4e4dae343c785bbe3f65dfda32114af357cf74

mitsukileung, 2023-05-27:
> 是不是这个patchhttps://github.com/coolsnowwolf/lede/commit/aa4e4dae343c785bbe3f65dfda32114af357cf74

不知道呀
回滚到 https://github.com/coolsnowwolf/lede/commit/26f3634631244eedfbdfd61ea6cc3a2d7f5d896f 这个PR就可以正常编译，无线正常，往后的就不行了

ddvcx, 2023-05-28:
同样问题

xiaiohuan, 2023-06-08:
昨天也遇到同样的问题  用以前的老版本源码编译现在直接报错无法编译 拉最新的编译 直接没无线

xiaiohuan, 2023-06-09:
目前解决了 退回源码  拉去最新源码 然后  cd lede 在 git checkout 26f3634631244eedfbdfd61ea6cc3a2d7f5d896f    在 git checkout -b 26f363463  切换就行了

xiaiohuan, 2023-06-09:
> 目前解决了 退回源码 拉去最新源码 然后 cd lede 在 git checkout [26f3634](https://github.com/coolsnowwolf/lede/commit/26f3634631244eedfbdfd61ea6cc3a2d7f5d896f) 在 git checkout -b [26f3634](https://github.com/coolsnowwolf/lede/commit/
[image]
26f3634631244eedfbdfd61ea6cc3a2d7f5d896f)

WYC-2020, 2023-06-09:
哈哈可怜的孩子，因该是他们升级了mac80211 导致的，毕竟一下从5.15 升级到了6.1  很多机型肯定都还没测试自然会出现各种奇葩问题

mitsukileung, 2023-06-10:
想用新版可以用 immortalwrt 来编译，immortalwrt升级后没有无WIFI这样的问题，但SSRP无法科学上网

mitsukileung, 2023-06-12:
> 请尝试替换成官方提供的无线 fw: https://github.com/openwrt/firmware_qca-wireless/raw/main/board-redmi_ax6.ipq8074

好的，我试试

xiaiohuan, 2023-06-12:
> 想用新版可以用immortalwrt 来编，immortalwrt升级后没有无WIFI这种的问题 但是SSRP无法科学网上，内存会用得更多

我是过用他的编译完 刷了直接不启动

mitsukileung, 2023-06-12:
> > 想用新版可以用immortalwrt 来编，immortalwrt升级后没有无WIFI这种的问题 但是SSRP无法科学网上，内存会用得更多
> 
> 我是过用他的编译完 刷了直接不启动

用 xiaomi_ax3600-stock 和 redmi_ax6-stock 可以正常启动，建议用23.05。好像出了RC1了
但我用SSRP不知为何翻不了墙，用passwall 和 homeproxy 可以

mitsukileung, 2023-06-12:
> > 
> 
> 开源源码 都吃内存 改了1G就没事了 edmi_ax6-stock xiaomi_ax3600-stock 的 库链接有不 发一个我看看

https://github.com/immortalwrt/immortalwrt/tree/openwrt-23.05

fichenx, 2023-06-27:
快两个月了，这个问题修复了吗？

mitsukileung, 2023-06-28:
> 快两个月了，这个问题修复了吗？

不清楚，这段时间在用 immortalwrt

fengqi, 2023-06-28:
都是老机型大概率不会修复了，也没必要追新，用旧版吧。

M4TRIX04, 2023-09-22:
这个问题会解决吗？

closed 2026-09-05 by coolsnowwolf