coolsnowwolf/lede #11185: 5月份更新代码编译报错No rule to make target package/system/opkg/host/compile

dqylyln, 2023-05-06:
## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ X] 没有类似的issue

## 2. 详细叙述
5月更新代码后编译报错，编译环境确定没问题，因为是每个月固定编译一次，最近一次成功过编译是4月份。

### (1) 具体问题

A：更新代码后编译报错：
*** No rule to make target package/system/opkg/host/compile", needed by package/compile
Stop.

### (2) 路由器型号和固件版本

A：X86/64  编译环境Linux V-UBUNTU 5.15.0-70-generic #77-Ubuntu SMP Tue Mar 21 14:02:37 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux

### (3) 详细日志

LD
ymlinux
SORTTAB wml inux
SYSHAP System. map
VOFFSET arch/x86/boat/compressed/../voffset
CC
arch/x86/boot/compressed/misc.0
DB.JCOPY arch/x86/boot/compressed/wml inux.bin
XZKERN arch/x86/boot/compressed/wml inux.bin. xz
MKPIGGY arch/x86/boot/compressed/piggy.s
AS
arch/x86/boot/compressed/piggy.o
LD
arch/x86/boot/compressed/wmlinux
ZOFFSET arch/x86/boot/zoffset.h
AS
LD
arch/x86/boot/header.o
arch/x86/boot/setup.elf
DBJCOPY arch/x86/boat/setup.bin
DB JCOPY arch/x86/boot/vmlinux.bin
BUILD arch/x86/boot/bzImage
ernel: arch/x86/boat/bzImage is ready (#2)
MODPOST modules-only.symvers
GEN
Module.sumvers
Bake [5]: Leaving directory ‘/root/lede/build_dir/target-x86_64_musl/1inux-×86_64/linux-5.15.110
If .conf ig did not change, use the previous timestamp to avoid package rebuilds
-s /root/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.110/.conf ig /root/lede/build_dir/target-x86_64_mus1/1inux-x86_64/linux-5.15.110/.conf ig.m
dules. save && mw /root/lede/build-dir/target-x86-64-musl/linux-x86_64/1inux-5.15.110/.config.modules. save /root/lede/build_dir/target-x86_64-musl/1inux-x86_64/
inux-5.15.110/.config；cp -fpR /root/lede/build_dir/target-x86_64_mus1/ 1inux-x86_64/1inux-5.15.110/.conf ig /root/lede/build_dir/target-x86_64_musl/linux-x86_64
1inux-5.15.110/.conf ig.modules.save
ouch
/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.110/.madules
iake -C image Cconp JLe. TARGET, BUILD5..
rake [5]： Enter ing directory‘/root/lede/target/linux/x86/image
rake[5]: Nothing to be done for compile.
lake [5]: Leaving directory ‘/root/lede/target/linux/x86/image
iake [4]: Leaving directony /root/lede/target/linux/x86
iake [3]: Leaving directory /root/lede/target/linux
ime: target/linux/compile#28.59#7.44#36.02
iake[2l: Leaving directory'/root/lede
xport MAKEFLAGS= ;make 一u -r buildinfo
iake[2]: Entering directory /root/lede
make [3] diffconfig buildversion feedsversion
make [4] scripts/conf ig/conf
rake [2J: Leaving directory /root/lede
iake [2J: Enter ing directory /root/lede
m -rf /root/lede/staging_dir/target-x86_64_mus 1/root-x86
iake [2]: Leaving directory /root/lede
rake [2]： Entering directory /root/lede
iake[2]： *** No rule to make target package/system/opkg/host/compile, needed by package/compile.stop.
iake [2]： Leaving directory /root/lede
iake[1]：米*米 [package/Hakefile:110: /root/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error
iake[1J: Leaving directory /root/lede
iake：***[/root/lede/ inc lude/toplevel.mk:231: wor ld] Error 2

a786458540, 2023-05-06:
和你一样情况，我是x86正常，但是编r66s的出错了不知道为啥

WYC-2020, 2023-05-06:
说了几万次 j1 日志

dqylyln, 2023-05-06:
> 说了几万次 j1 日志

日志打印到终端不太好复制，所以只截取了错误部分，编译是j1了。我在从新编译下把日志重定向到一个文件里。

AE86JAY, 2023-05-06:
有没加luci-app-easymesh？反正我的一加这插件就出错。

dqylyln, 2023-05-06:
[build.log](https://github.com/coolsnowwolf/lede/files/11411972/build.log)

补充下j1 日志

dqylyln, 2023-05-06:
> 有没加luci-app-easymesh？反正我的一加这插件就出错。

没有加这个，这玩意我都不用。

himho, 2023-05-06:
我今天X86也出现这个问题了。先不折腾了！

WYC-2020, 2023-05-07:
> > 说了几万次 j1 日志
> 
> 日志打印到终端不太好复制，所以只截取了错误部分，编译是j1了。我在从新编译下把日志重定向到一个文件里。

你这个单线程日志也太假了吧，看不出哪里错误，哈哈

dqylyln, 2023-05-07:
> > > 说了几万次 j1 日志
> > 
> > 
> > 日志打印到终端不太好复制，所以只截取了错误部分，编译是j1了。我在从新编译下把日志重定向到一个文件里。
> 
> 你这个单线程日志也太假了吧，看不出哪里错误，哈哈

这样输出的日志，难道方式不对？
make -j1 V=s > build.log 2>&1

WYC-2020, 2023-05-08:
> 

因为你这个日志看不到错误位置，假单线程假日志哈哈

closed 2023-05-11 by aiamadeus

dqylyln, 2023-06-16:
问题找到了，是用了https://github.com/kenzok8/small-package 这个源导致的，切换成https://github.com/kenzok8/openwrt-packages 这个源编译就好了。

zhang-mr, 2023-06-22:
> 问题找到了，是用了https://github.com/kenzok8/small-package 这个源导致的，切换成https://github.com/kenzok8/openwrt-packages 这个源编译就好了。

解决了我的问题，希望有更多人看到这个