coolsnowwolf/lede #11187: 编译固件时报错误2

MgCI2, 2023-05-06:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：编译固件时报错误2

### (2) 路由器型号和固件版本

A：x86_64

### (3) 详细日志

A：
“Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location https://mirror2.openwrt.org/sources/ipt2socks-1.1.3.tar.gz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (22) The requested URL returned error: 404 Not Found
Download failed.
No more mirrors to try - giving up.
make[3]: *** [Makefile:46: /home/ubuntu/lede/dl/ipt2socks-1.1.3.tar.gz] Error 2
make[3]: Leaving directory '/home/ubuntu/lede/feeds/small/ipt2socks'
time: package/feeds/small/ipt2socks/compile#0.21#0.08#2.13
    ERROR: package/feeds/small/ipt2socks failed to build.
make[2]: *** [package/Makefile:116: package/feeds/small/ipt2socks/compile] Error 1
make[2]: Leaving directory '/home/ubuntu/lede'
make[1]: *** [package/Makefile:110: /home/ubuntu/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/ubuntu/lede'
make: *** [/home/ubuntu/lede/include/toplevel.mk:231: world] Error 2”
[image]

Gzxhwq, 2023-05-06:
到你这个small的feeds下面去提才对

MgCI2, 2023-05-07:
怎么提

WYC-2020, 2023-05-07:
你说怎么提，这里怎么提的到哪里就怎么提，一看就是url链接失效而已，自己修复一下就行的，不知道怎么修直接吧xiaorouji下面相同的ip2sokcket直接替换这个small 下面的就行

mitsukileung, 2023-05-07:
> 怎么提

把 /lede/feeds/helloworld/ipt2socks/Makefile 里的 PKG_HASH:=73a2498dc95934c225d358707e7f7d060b5ce81aa45260ada09cbd15207d27d1 替换成 PKG_HASH:=5279eb1cb7555cf9292423cc9f672dc43e6e214b3411a6df26a6a1cfa59d88b7 就可以了

binge8, 2023-05-07:
等合并就好了 https://github.com/fw876/helloworld/pull/1191

wilson-chan2019, 2023-05-07:
cd lede/dl
wget https://sources.immortalwrt.org/ipt2socks-1.1.3.tar.gz
再重新编译

wuqingxin1987, 2023-05-09:
删掉dl库里面的iptsockes再make download -j8

closed 2023-05-10 by MgCI2