coolsnowwolf/lede #11191: 竞斗云编译固件WiFi无法正常使用

OldCoding, 2023-05-07:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：竞斗云编译刷入后没有WiFi，无线的界面显示未开启或未关联

### (2) 路由器型号和固件版本

A：竞斗云2.0，当天最新源码

### (3) 详细日志

A：
Sun May  7 19:49:33 2023 kern.info kernel: [    0.000000] Booting Linux on physical CPU 0x0
Sun May  7 19:49:33 2023 kern.notice kernel: [    0.000000] Linux version 5.4.242 (runner@fv-az395-607) (gcc version 11.3.0 (OpenWrt GCC 11.3.0 r5946-4fbb5202f)) #0 SMP Sat May 6 18:00:08 2023
Sun May  7 19:49:33 2023 kern.info kernel: [    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c5387d
Sun May  7 19:49:33 2023 kern.info kernel: [    0.000000] CPU: div instructions available: patching division code
Sun May  7 19:49:33 2023 kern.info kernel: [    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
Sun May  7 19:49:33 2023 kern.info kernel: [    0.000000] OF: fdt: Machine model: P&W R619AC 128M
Sun May  7 19:49:33 2023 kern.info kernel: [    0.000000] Memory policy: Data cache writealloc
Sun May  7 19:49:33 2023 kern.debug kernel: [    0.000000] On node 0 totalpages: 130560
Sun May  7 19:49:33 2023 kern.debug kernel: [    0.000000]   Normal zone: 1152 pages used for memmap
Sun May  7 19:49:33 2023 kern.debug kernel: [    0.000000]   Normal zone: 0 pages reserved
Sun May  7 19:49:33 2023 kern.debug kernel: [    0.000000]   Normal zone: 130560 pages, LIFO batch:31
Sun May  7 19:49:33 2023 kern.info kernel: [    0.000000] percpu: Embedded 15 pages/cpu s30348 r8192 d22900 u61440
Sun May  7 19:49:33 2023 kern.debug kernel: [    0.000000] pcpu-alloc: s30348 r8192 d22900 u61440 alloc=15*4096
Sun May  7 19:49:33 2023 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
Sun May  7 19:49:33 2023 kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 129408
Sun May  7 19:49:33 2023 kern.notice kernel: [    0.000000] Kernel command line: ubi.mtd=rootfs root=mtd:ubi_rootfs rootfstype=squashfs rootwait root=/dev/ubiblock0_1 rootfstype=squashfs
Sun May  7 19:49:33 2023 kern.info kernel: [    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
Sun May  7 19:49:33 2023 kern.info kernel: [    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
Sun May  7 19:49:33 2023 kern.info kernel: [    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
Sun May  7 19:49:33 2023 kern.info kernel: [    0.000000] Memory: 506736K/522240K available (5931K kernel code, 188K rwdata, 1484K rodata, 1024K init, 235K bss, 15504K reserved, 0K cma-reserved, 0K highmem)
Sun May  7 19:49:33 2023 kern.info kernel: [    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
Sun May  7 19:49:33 2023 kern.info kernel: [    0.000000] rcu: Hierarchical RCU implementation.
Sun May  7 19:49:33 2023 kern.info kernel: [    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
Sun May  7 19:49:33 2023 kern.info kernel: [    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
Sun May  7 19:49:33 2023 kern.info kernel: [    0.000000] arch_timer: cp15 timer(s) running at 48.00MHz (virt).
Sun May  7 19:49:33 2023 kern.info kernel: [    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xb11fd3bfb, max_idle_ns: 440795203732 ns
Sun May  7 19:49:33 2023 kern.info kernel: [    0.000007] sched_clock: 56 bits at 48MHz, resolution 20ns, wraps every 4398046511096ns
Sun May  7 19:49:33 2023 kern.info kernel: [    0.000020] Switching to timer-based delay loop, resolution 20ns
Sun May  7 19:49:33 2023 kern.info kernel: [    0.000309] Calibrating delay loop (skipped), value calculated using timer frequency.. 96.00 BogoMIPS (lpj=480000)
Sun May  7 19:49:33 2023 kern.info kernel: [    0.000327] pid_max: default: 32768 minimum: 301
Sun May  7 19:49:33 2023 kern.info kernel: [    0.000515] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Sun May  7 19:49:33 2023 kern.info kernel: [    0.000533] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Sun May  7 19:49:33 2023 kern.info kernel: [    0.001527] CPU: Testing write buffer coherency: ok
Sun May  7 19:49:33 2023 kern.info kernel: [    0.002819] Setting up static identity map for 0x80300000 - 0x80300060
Sun May  7 19:49:33 2023 kern.info kernel: [    0.002993] rcu: Hierarchical SRCU implementation.
Sun May  7 19:49:33 2023 kern.info kernel: [    0.003612] smp: Bringing up secondary CPUs ...
Sun May  7 19:49:33 2023 kern.info kernel: [    0.007157] smp: Brought up 1 node, 4 CPUs
Sun May  7 19:49:33 2023 kern.info kernel: [    0.007176] SMP: Total of 4 processors activated (384.00 BogoMIPS).
Sun May  7 19:49:33 2023 kern.info kernel: [    0.007183] CPU: All CPU(s) started in SVC mode.
Sun May  7 19:49:33 2023 kern.info kernel: [    0.012128] VFP support v0.3: implementor 41 architecture 2 part 30 variant 7 rev 5
Sun May  7 19:49:33 2023 kern.info kernel: [    0.012290] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
Sun May  7 19:49:33 2023 kern.info kernel: [    0.012313] futex hash table entries: 1024 (order: 4, 65536 bytes, linear)
Sun May  7 19:49:33 2023 kern.info kernel: [    0.012553] pinctrl core: initialized pinctrl subsystem
Sun May  7 19:49:33 2023 kern.info kernel: [    0.014190] NET: Registered protocol family 16
Sun May  7 19:49:33 2023 kern.info kernel: [    0.014547] DMA: preallocated 256 KiB pool for atomic coherent allocations
Sun May  7 19:49:33 2023 kern.info kernel: [    0.015798] cpuidle: using governor ladder
Sun May  7 19:49:33 2023 kern.info kernel: [    0.015848] cpuidle: using governor menu
Sun May  7 19:49:33 2023 kern.info kernel: [    0.016519] pstore: Registered ramoops as persistent store backend
Sun May  7 19:49:33 2023 kern.info kernel: [    0.016535] ramoops: using 0x80000@0x83f00000, ecc: 0
Sun May  7 19:49:33 2023 kern.info kernel: [    0.037301] cryptd: max_cpu_qlen set to 1000
[... 486 lines cut ...]
Sun May  7 19:52:29 2023 daemon.err nmbd[7575]:   This response was from IP 192.168.10.2, reporting an IP address of 192.168.10.2.
Sun May  7 19:52:30 2023 daemon.err smbd[7574]: [2023/05/07 19:52:30.635844,  0] ../../lib/util/become_daemon.c:135(daemon_ready)
Sun May  7 19:52:30 2023 daemon.err smbd[7574]:   daemon_ready: daemon 'smbd' finished starting up and ready to serve connections
Sun May  7 19:52:32 2023 daemon.info avahi-daemon[4825]: Joining mDNS multicast group on interface ztwfupnfjb.IPv4 with address 192.168.192.2.
Sun May  7 19:52:32 2023 daemon.info avahi-daemon[4825]: New relevant interface ztwfupnfjb.IPv4 for mDNS.
Sun May  7 19:52:32 2023 daemon.info avahi-daemon[4825]: Registering new address record for 192.168.192.2 on ztwfupnfjb.IPv4.
Sun May  7 19:52:33 2023 daemon.info avahi-daemon[4825]: Joining mDNS multicast group on interface ztwfupnfjb.IPv6 with address fe80::90c4:60ff:fe2d:face.
Sun May  7 19:52:33 2023 daemon.info avahi-daemon[4825]: New relevant interface ztwfupnfjb.IPv6 for mDNS.
Sun May  7 19:52:33 2023 daemon.info avahi-daemon[4825]: Registering new address record for fe80::90c4:60ff:fe2d:face on ztwfupnfjb.*.
Sun May  7 19:52:33 2023 daemon.err uhttpd[3916]: awk: cmd. line:1: Network is down
Sun May  7 19:52:34 2023 cron.err crond[9251]: crond (busybox 1.36.0) started, log level 5
Sun May  7 19:52:35 2023 user.notice dnsmasq: found already running DHCP-server on interface 'br-lan' refusing to start, use 'option force 1' to override
Sun May  7 19:52:35 2023 cron.err crond[9617]: crond (busybox 1.36.0) started, log level 5
Sun May  7 19:52:35 2023 daemon.info procd: - init complete -
Sun May  7 19:52:38 2023 user.notice firewall: Reloading firewall due to ifup of wan (eth0.2)
Sun May  7 19:52:38 2023 daemon.err uhttpd[3916]: awk: cmd. line:1: Network is down
Sun May  7 19:52:39 2023 user.notice nlbwmon: Reloading nlbwmon due to ifup of wan (eth0.2)
Sun May  7 19:52:43 2023 daemon.err uhttpd[3916]: awk: cmd. line:1: Network is down
Sun May  7 19:52:48 2023 daemon.err uhttpd[3916]: awk: cmd. line:1: Network is down
Sun May  7 19:52:48 2023 daemon.info procd: Instance haproxy::instance1 s in a crash loop 6 crashes, 0 seconds since last crash
Sun May  7 19:52:51 2023 user.notice firewall: Reloading firewall due to ifup of wan6 (eth0.2)
Sun May  7 19:52:53 2023 user.notice nlbwmon: Reloading nlbwmon due to ifup of wan6 (eth0.2)
Sun May  7 19:52:53 2023 daemon.err uhttpd[3916]: awk: cmd. line:1: Network is down
Sun May  7 19:52:58 2023 daemon.err uhttpd[3916]: awk: cmd. line:1: Network is down
Sun May  7 19:53:03 2023 daemon.err uhttpd[3916]: awk: cmd. line:1: Network is down

hubbylei, 2023-05-08:
网件R7800无线也挂了，日志也是这样 T^T

Luupw3d, 2023-05-13:
同为竞斗云，出现相同问题，日志类似。

wilson-chan2019, 2023-05-13:
acrh17一样问题

closed 2023-05-19 by aiamadeus