coolsnowwolf/lede #11192: xiaomi R3G 最新版本wifi失效

eagle-lk, 2023-05-07:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
小米R3G最新版本刷入后无线失效
Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue
其他型号有类似问题，不知是否为Ramips/MTK mt7261的普遍问题。
## 2. 详细叙述
最近可用版本为R23.4.1，不可用版本为R23.5.1
### (1) 具体问题
从R23.3.3 sysupgrade，无线部分可用，能搜索，但作为AP无法被发现。
清除配置后全新刷入，整个无线部分不可用。路由器其他功能正常。
回退到R23.3.3，功能正常。
A：

### (2) 路由器型号和固件版本
xiaomi R3G，R23.5.1，23.5.7日编译。
A：

### (3) 详细日志
`Sun May  7 16:33:48 2023 kern.notice kernel: [    0.000000] Linux version 5.4.242 (eagle@DESKTOP-0PEJ6PS) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r5859-0747f72a7)) #0 SMP Sun May 7 08:33:20 2023
Sun May  7 16:33:48 2023 kern.info kernel: [    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
Sun May  7 16:33:48 2023 kern.info kernel: [    0.000000] printk: bootconsole [early0] enabled
Sun May  7 16:33:48 2023 kern.info kernel: [    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
Sun May  7 16:33:48 2023 kern.info kernel: [    0.000000] MIPS: machine is Xiaomi Mi Router 3G
Sun May  7 16:33:48 2023 kern.info kernel: [    0.000000] Initrd not found or empty - disabling initrd
Sun May  7 16:33:48 2023 kern.info kernel: [    0.000000] VPE topology {2,2} total 4
Sun May  7 16:33:48 2023 kern.warn kernel: [    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
Sun May  7 16:33:48 2023 kern.warn kernel: [    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
Sun May  7 16:33:48 2023 kern.warn kernel: [    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
Sun May  7 16:33:48 2023 kern.info kernel: [    0.000000] Zone ranges:
Sun May  7 16:33:48 2023 kern.info kernel: [    0.000000]   Normal   [mem 0x0000000000000000-0x000000000fffffff]
Sun May  7 16:33:48 2023 kern.info kernel: [    0.000000]   HighMem  empty
Sun May  7 16:33:48 2023 kern.info kernel: [    0.000000] Movable zone start for each node
Sun May  7 16:33:48 2023 kern.info kernel: [    0.000000] Early memory node ranges
Sun May  7 16:33:48 2023 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000000000000-0x000000000fffffff]
Sun May  7 16:33:48 2023 kern.info kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000000fffffff]
Sun May  7 16:33:48 2023 kern.debug kernel: [    0.000000] On node 0 totalpages: 65536
Sun May  7 16:33:48 2023 kern.debug kernel: [    0.000000]   Normal zone: 576 pages used for memmap
Sun May  7 16:33:48 2023 kern.debug kernel: [    0.000000]   Normal zone: 0 pages reserved
Sun May  7 16:33:48 2023 kern.debug kernel: [    0.000000]   Normal zone: 65536 pages, LIFO batch:15
Sun May  7 16:33:48 2023 kern.info kernel: [    0.000000] percpu: Embedded 14 pages/cpu s26640 r8192 d22512 u57344
Sun May  7 16:33:48 2023 kern.debug kernel: [    0.000000] pcpu-alloc: s26640 r8192 d22512 u57344 alloc=14*4096
Sun May  7 16:33:48 2023 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
Sun May  7 16:33:48 2023 kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 64960
Sun May  7 16:33:48 2023 kern.notice kernel: [    0.000000] Kernel command line: console=ttyS0,115200n8 rootfstype=squashfs,jffs2
Sun May  7 16:33:48 2023 kern.info kernel: [    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
Sun May  7 16:33:48 2023 kern.info kernel: [    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
Sun May  7 16:33:48 2023 kern.info kernel: [    0.000000] Writing ErrCtl register=00014008
Sun May  7 16:33:48 2023 kern.info kernel: [    0.000000] Readback ErrCtl register=00014008
Sun May  7 16:33:48 2023 kern.info kernel: [    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
Sun May  7 16:33:48 2023 kern.info kernel: [    0.000000] Memory: 250020K/262144K available (6209K kernel code, 274K rwdata, 1304K rodata, 1288K init, 235K bss, 12124K reserved, 0K cma-reserved, 0K highmem)
Sun May  7 16:33:48 2023 kern.info kernel: [    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
Sun May  7 16:33:48 2023 kern.info kernel: [    0.000000] rcu: Hierarchical RCU implementation.
Sun May  7 16:33:48 2023 kern.info kernel: [    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
Sun May  7 16:33:48 2023 kern.info kernel: [    0.000000] NR_IRQS: 256
Sun May  7 16:33:48 2023 kern.info kernel: [    0.000000] CPU Clock: 880MHz
Sun May  7 16:33:48 2023 kern.info kernel: [    0.000000] clocksource: GIC: mask: 0xffffffffffffffff max_cycles: 0xcaf478abb4, max_idle_ns: 440795247997 ns
Sun May  7 16:33:48 2023 kern.info kernel: [    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 4343773742 ns
Sun May  7 16:33:48 2023 kern.info kernel: [    0.000009] sched_clock: 32 bits at 440MHz, resolution 2ns, wraps every 4880645118ns
Sun May  7 16:33:48 2023 kern.info kernel: [    0.007889] Calibrating delay loop... 586.13 BogoMIPS (lpj=2930688)
Sun May  7 16:33:48 2023 kern.info kernel: [    0.074057] pid_max: default: 32768 minimum: 301
Sun May  7 16:33:48 2023 kern.info kernel: [    0.078857] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Sun May  7 16:33:48 2023 kern.info kernel: [    0.086058] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Sun May  7 16:33:48 2023 kern.info kernel: [    0.096928] rcu: Hierarchical SRCU implementation.
Sun May  7 16:33:48 2023 kern.info kernel: [    0.102439] smp: Bringing up secondary CPUs ...
Sun May  7 16:33:48 2023 kern.warn kernel: [    5.210338] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
Sun May  7 16:33:48 2023 kern.warn kernel: [    5.210347] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
Sun May  7 16:33:48 2023 kern.warn kernel: [    5.210360] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
Sun May  7 16:33:48 2023 kern.info kernel: [    5.210466] CPU1 revision is: 0001992f (MIPS 1004Kc)
[... 542 lines cut ...]
Sun May  7 16:35:40 2023 user.err : jail: prctl(PR_CAPBSET_DROP, 38) failed: Invalid argument
Sun May  7 16:35:41 2023 user.err : jail: prctl(PR_CAPBSET_DROP, 38) failed: Invalid argument
Sun May  7 16:35:41 2023 user.err : jail: prctl(PR_CAPBSET_DROP, 38) failed: Invalid argument
Sun May  7 16:35:42 2023 user.err : jail: prctl(PR_CAPBSET_DROP, 38) failed: Invalid argument
Sun May  7 16:35:42 2023 user.err : jail: prctl(PR_CAPBSET_DROP, 38) failed: Invalid argument
Sun May  7 16:35:43 2023 user.err : jail: prctl(PR_CAPBSET_DROP, 38) failed: Invalid argument
Sun May  7 16:35:43 2023 user.err : jail: prctl(PR_CAPBSET_DROP, 38) failed: Invalid argument
Sun May  7 16:35:44 2023 user.err : jail: prctl(PR_CAPBSET_DROP, 38) failed: Invalid argument
Sun May  7 16:35:44 2023 user.err : jail: prctl(PR_CAPBSET_DROP, 38) failed: Invalid argument
Sun May  7 16:35:45 2023 user.err : jail: prctl(PR_CAPBSET_DROP, 38) failed: Invalid argument
Sun May  7 16:35:45 2023 user.err : jail: prctl(PR_CAPBSET_DROP, 38) failed: Invalid argument
Sun May  7 16:35:46 2023 user.err : jail: prctl(PR_CAPBSET_DROP, 38) failed: Invalid argument
Sun May  7 16:35:46 2023 user.err : jail: prctl(PR_CAPBSET_DROP, 38) failed: Invalid argument
Sun May  7 16:35:47 2023 user.err : jail: prctl(PR_CAPBSET_DROP, 38) failed: Invalid argument
Sun May  7 16:35:47 2023 user.err : jail: prctl(PR_CAPBSET_DROP, 38) failed: Invalid argument
Sun May  7 16:35:48 2023 user.err : jail: prctl(PR_CAPBSET_DROP, 38) failed: Invalid argument
Sun May  7 16:35:48 2023 user.err : jail: prctl(PR_CAPBSET_DROP, 38) failed: Invalid argument
Sun May  7 16:35:49 2023 user.err : jail: prctl(PR_CAPBSET_DROP, 38) failed: Invalid argument
Sun May  7 16:35:49 2023 user.err : jail: prctl(PR_CAPBSET_DROP, 38) failed: Invalid argument
Sun May  7 16:35:50 2023 user.err : jail: prctl(PR_CAPBSET_DROP, 38) failed: Invalid argument
Sun May  7 16:35:50 2023 user.err : jail: prctl(PR_CAPBSET_DROP, 38) failed: Invalid argument
Sun May  7 16:35:51 2023 user.err : jail: prctl(PR_CAPBSET_DROP, 38) failed: Invalid argument
Sun May  7 16:35:51 2023 user.err : jail: prctl(PR_CAPBSET_DROP, 38) failed: Invalid argument
`
A：

eagle-lk, 2023-05-08:
看了一下其他型号的问题反馈，wan口是否连接上级设备结果都一样

coolsnowwolf, 2023-05-08:
fixed

closed 2023-05-08 by coolsnowwolf