coolsnowwolf/lede #11193: 使能无线相关组件编译报错

sensiki, 2023-05-08:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
git clone https://github.com/coolsnowwolf/lede
cd lede
./scripts/feeds update -a
./scripts/feeds install -a
make menuconfig
然后选择
FriendlyARM NanoPi R5S
再使能
PACKAGE_kmod-cfg80211
PACKAGE_kmod-mac80211
PACKAGE_hostapd
CONFIG_PACKAGE_wpa-supplicant
然后进行编译，编译不过，报错看下面详细信息

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：
/home/sen/test/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/lib/gcc/aarch64-openwrt-linux-musl/11.3.0/../../../../aarch64-openwrt-linux-musl/bin/ld: /home/sen/test/lede/build_dir/target-aarch64_generic_musl/hostapd-supplicant-full-internal/hostapd-2022-07-29-b704dc72/build/wpa_supplicant/src/ap/acs.o: in function `acs_request_scan':
acs.c:(.text+0x488): undefined reference to `hostapd_driver_scan'
/home/sen/test/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/lib/gcc/aarch64-openwrt-linux-musl/11.3.0/../../../../aarch64-openwrt-linux-musl/bin/ld: acs.c:(.text+0x494): undefined reference to `hostapd_driver_scan'
/home/sen/test/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/lib/gcc/aarch64-openwrt-linux-musl/11.3.0/../../../../aarch64-openwrt-linux-musl/bin/ld: /home/sen/test/lede/build_dir/target-aarch64_generic_musl/hostapd-supplicant-full-internal/hostapd-2022-07-29-b704dc72/build/wpa_supplicant/src/ap/acs.o: in function `acs_scan_complete':
acs.c:(.text+0x564): undefined reference to `hostapd_disable_iface'
/home/sen/test/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/lib/gcc/aarch64-openwrt-linux-musl/11.3.0/../../../../aarch64-openwrt-linux-musl/bin/ld: acs.c:(.text+0x570): undefined reference to `hostapd_disable_iface'
/home/sen/test/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/lib/gcc/aarch64-openwrt-linux-musl/11.3.0/../../../../aarch64-openwrt-linux-musl/bin/ld: /home/sen/test/lede/build_dir/target-aarch64_generic_musl/hostapd-supplicant-full-internal/hostapd-2022-07-29-b704dc72/build/wpa_supplicant/src/ap/acs.o: in function `acs_init':
acs.c:(.text+0x1198): undefined reference to `hostapd_drv_do_acs'
/home/sen/test/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/lib/gcc/aarch64-openwrt-linux-musl/11.3.0/../../../../aarch64-openwrt-linux-musl/bin/ld: acs.c:(.text+0x119c): undefined reference to `hostapd_drv_do_acs'
/home/sen/test/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/lib/gcc/aarch64-openwrt-linux-musl/11.3.0/../../../../aarch64-openwrt-linux-musl/bin/ld: acs.c:(.text+0x11f0): undefined reference to `hostapd_set_state'
/home/sen/test/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/lib/gcc/aarch64-openwrt-linux-musl/11.3.0/../../../../aarch64-openwrt-linux-musl/bin/ld: acs.c:(.text+0x11fc): undefined reference to `hostapd_set_state'
collect2: error: ld returned 1 exit status
make[4]: *** [Makefile:2037: wpa_supplicant] Error 1
make[4]: Leaving directory '/home/sen/test/lede/build_dir/target-aarch64_generic_musl/hostapd-supplicant-full-internal/hostapd-2022-07-29-b704dc72/wpa_supplicant'
make[3]: *** [Makefile:790: /home/sen/test/lede/build_dir/target-aarch64_generic_musl/hostapd-supplicant-full-internal/hostapd-2022-07-29-b704dc72/.built] Error 2
make[3]: Leaving directory '/home/sen/test/lede/package/network/services/hostapd'
time: package/network/services/hostapd/supplicant-full-internal/compile#39.90#2.67#41.44
    ERROR: package/network/services/hostapd failed to build (build variant: supplicant-full-internal).
make[2]: *** [package/Makefile:119: package/network/services/hostapd/compile] Error 1
make[2]: Leaving directory '/home/sen/test/lede'
make[1]: *** [package/Makefile:110: /home/sen/test/lede/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/sen/test/lede'
make: *** [/home/sen/test/lede/include/toplevel.mk:231: world] Error 2

sensiki, 2023-05-08:
> make clean

不行，新拉代码。做我写的步骤进行配置，直接编译报错

eagle-lk, 2023-05-08:
下午注意到新的代码内核切换到了5.10.176，编译通过，无线部分的操作恢复（）。实测性能比之前的版本有很大的差距。5G部分大约300Mb/s

eagle-lk, 2023-05-08:
有线部分HWNAT好像也失效了，LAN/WAN 整体的转发能力只有300多一点。之前R23.3.3(5.4.234)版及R23.4.1(5.4.241)整体转发可以达到950Mb左右，5G部分可以到500M以上。

boyhaibo, 2023-05-08:
同无线报错
```
make[2]: Entering directory '/home/runner/work/OpenWrt/OpenWrt/workdir/openwrt/feeds/packages/lang/perl'
echo "perl" >> /home/runner/work/OpenWrt/OpenWrt/workdir/openwrt/staging_dir/target-aarch64_generic_musl/pkginfo/perl.default.install
echo "perlbase-base" >> /home/runner/work/OpenWrt/OpenWrt/workdir/openwrt/staging_dir/target-aarch64_generic_musl/pkginfo/perl.default.install
echo "perlbase-bytes" >> /home/runner/work/OpenWrt/OpenWrt/workdir/openwrt/staging_dir/target-aarch64_generic_musl/pkginfo/perl.default.install
echo "perlbase-charnames" >> /home/runner/work/OpenWrt/OpenWrt/workdir/openwrt/staging_dir/target-aarch64_generic_musl/pkginfo/perl.default.install
echo "perlbase-class" >> /home/runner/work/OpenWrt/OpenWrt/workdir/openwrt/staging_dir/target-aarch64_generic_musl/pkginfo/perl.default.install
echo "perlbase-config" >> /home/runner/work/OpenWrt/OpenWrt/workdir/openwrt/staging_dir/target-aarch64_generic_musl/pkginfo/perl.default.install
echo "perlbase-cwd" >> /home/runner/work/OpenWrt/OpenWrt/workdir/openwrt/staging_dir/target-aarch64_generic_musl/pkginfo/perl.default.install
echo "perlbase-dynaloader" >> /home/runner/work/OpenWrt/OpenWrt/workdir/openwrt/staging_dir/target-aarch64_generic_musl/pkginfo/perl.default.install
echo "perlbase-errno" >> /home/runner/work/OpenWrt/OpenWrt/workdir/openwrt/staging_dir/target-aarch64_generic_musl/pkginfo/perl.default.install
echo "perlbase-essential" >> /home/runner/work/OpenWrt/OpenWrt/workdir/openwrt/staging_dir/target-aarch64_generic_musl/pkginfo/perl.default.install
echo "perlbase-fcntl" >> /home/runner/work/OpenWrt/OpenWrt/workdir/openwrt/staging_dir/target-aarch64_generic_musl/pkginfo/perl.default.install
echo "perlbase-file" >> /home/runner/work/OpenWrt/OpenWrt/workdir/openwrt/staging_dir/target-aarch64_generic_musl/pkginfo/perl.default.install
echo "perlbase-filehandle" >> /home/runner/work/OpenWrt/OpenWrt/workdir/openwrt/staging_dir/target-aarch64_generic_musl/pkginfo/perl.default.install
echo "perlbase-getopt" >> /home/runner/work/OpenWrt/OpenWrt/workdir/openwrt/staging_dir/target-aarch64_generic_musl/pkginfo/perl.default.install
echo "perlbase-i18n" >> /home/runner/work/OpenWrt/OpenWrt/workdir/openwrt/staging_dir/target-aarch64_generic_musl/pkginfo/perl.default.install
echo "perlbase-integer" >> /home/runner/work/OpenWrt/OpenWrt/workdir/openwrt/staging_dir/target-aarch64_generic_musl/pkginfo/perl.default.install
echo "perlbase-io" >> /home/runner/work/OpenWrt/OpenWrt/workdir/openwrt/staging_dir/target-aarch64_generic_musl/pkginfo/perl.default.install
echo "perlbase-list" >> /home/runner/work/OpenWrt/OpenWrt/workdir/openwrt/staging_dir/target-aarch64_generic_musl/pkginfo/perl.default.install
echo "perlbase-locale" >> /home/runner/work/OpenWrt/OpenWrt/workdir/openwrt/staging_dir/target-aarch64_generic_musl/pkginfo/perl.default.install
echo "perlbase-params" >> /home/runner/work/OpenWrt/OpenWrt/workdir/openwrt/staging_dir/target-aarch64_generic_musl/pkginfo/perl.default.install
echo "perlbase-posix" >> /home/runner/work/OpenWrt/OpenWrt/workdir/openwrt/staging_dir/target-aarch64_generic_musl/pkginfo/perl.default.install
echo "perlbase-re" >> /home/runner/work/OpenWrt/OpenWrt/workdir/openwrt/staging_dir/target-aarch64_generic_musl/pkginfo/perl.default.install
echo "perlbase-scalar" >> /home/runner/work/OpenWrt/OpenWrt/workdir/openwrt/staging_dir/target-aarch64_generic_musl/pkginfo/perl.default.install
echo "perlbase-selectsaver" >> /home/runner/work/OpenWrt/OpenWrt/workdir/openwrt/staging_dir/target-aarch64_generic_musl/pkginfo/perl.default.install
echo "perlbase-socket" >> /home/runner/work/OpenWrt/OpenWrt/workdir/openwrt/staging_dir/target-aarch64_generic_musl/pkginfo/perl.default.install
echo "perlbase-symbol" >> /home/runner/work/OpenWrt/OpenWrt/workdir/openwrt/staging_dir/target-aarch64_generic_musl/pkginfo/perl.default.install
echo "perlbase-tie" >> /home/runner/work/OpenWrt/OpenWrt/workdir/openwrt/staging_dir/target-aarch64_generic_musl/pkginfo/perl.default.install
echo "perlbase-time" >> /home/runner/work/OpenWrt/OpenWrt/workdir/openwrt/staging_dir/target-aarch64_generic_musl/pkginfo/perl.default.install
echo "perlbase-unicode" >> /home/runner/work/OpenWrt/OpenWrt/workdir/openwrt/staging_dir/target-aarch64_generic_musl/pkginfo/perl.default.install
echo "perlbase-unicore" >> /home/runner/work/OpenWrt/OpenWrt/workdir/openwrt/staging_dir/target-aarch64_generic_musl/pkginfo/perl.default.install
echo "perlbase-utf8" >> /home/runner/work/OpenWrt/OpenWrt/workdir/openwrt/staging_dir/target-aarch64_generic_musl/pkginfo/perl.default.install
echo "perlbase-xsloader" >> /home/runner/work/OpenWrt/OpenWrt/workdir/openwrt/staging_dir/target-aarch64_generic_musl/pkginfo/perl.default.install
make[2]: Leaving directory '/home/runner/work/OpenWrt/OpenWrt/workdir/openwrt/feeds/packages/lang/perl'
time: package/feeds/packages/perl/compile#0.50#0.24#0.65
make[2]: Entering directory '/home/runner/work/OpenWrt/OpenWrt/workdir/openwrt/feeds/packages/lang/perl-http-date'
echo "perl-http-date" >> /home/runner/work/OpenWrt/OpenWrt/workdir/openwrt/staging_dir/target-aarch64_generic_musl/pkginfo/perl-http-date.default.install
make[2]: Leaving directory '/home/runner/work/OpenWrt/OpenWrt/workdir/openwrt/feeds/packages/lang/perl-http-date'
time: package/feeds/packages/perl-http-date/compile#0.16#0.06#0.21
make[2]: Entering directory '/home/runner/work/OpenWrt/OpenWrt/workdir/openwrt/feeds/packages/utils/pigz'
[... cut ...]
make[2]: Leaving directory '/home/runner/work/OpenWrt/OpenWrt/workdir/openwrt/package/network/services/hostapd'
time: package/network/services/hostapd/wpad-basic/compile#0.53#0.20#0.72
    ERROR: package/network/services/hostapd failed to build (build variant: wpad-basic).
make[1]: *** [package/Makefile:116: package/network/services/hostapd/compile] Error 1
make[1]: Leaving directory '/home/runner/work/OpenWrt/OpenWrt/workdir/openwrt'
make: *** [/home/runner/work/OpenWrt/OpenWrt/workdir/openwrt/include/toplevel.mk:231: package/compile] Error 2
Error: Process completed with exit code 2.
```

yhl452493373, 2023-05-20:
无线和nftable部分功能都会编译不过去。。

hammerfunctor, 2023-06-08:
Same here for r5c.
<details>
<summary>.config</summary>
<br>

```toml
CONFIG_MODULES=y
CONFIG_HAVE_DOT_CONFIG=y
CONFIG_HOST_OS_LINUX=y
CONFIG_TARGET_rockchip=y
CONFIG_TARGET_rockchip_armv8=y
CONFIG_TARGET_rockchip_armv8_DEVICE_friendlyarm_nanopi-r5c=y
CONFIG_HAS_SUBTARGETS=y
CONFIG_HAS_DEVICES=y
CONFIG_TARGET_BOARD="rockchip"
CONFIG_TARGET_SUBTARGET="armv8"
CONFIG_TARGET_PROFILE="DEVICE_friendlyarm_nanopi-r5c"
CONFIG_TARGET_ARCH_PACKAGES="aarch64_generic"
CONFIG_DEFAULT_TARGET_OPTIMIZATION="-Os -pipe -mcpu=generic"
CONFIG_CPU_TYPE="generic"
CONFIG_LINUX_5_15=y
CONFIG_DEFAULT_autocore-arm=y
CONFIG_DEFAULT_automount=y
CONFIG_DEFAULT_base-files=y
CONFIG_DEFAULT_block-mount=y
CONFIG_DEFAULT_busybox=y
CONFIG_DEFAULT_ca-bundle=y
CONFIG_DEFAULT_ca-certificates=y
CONFIG_DEFAULT_coremark=y
CONFIG_DEFAULT_curl=y
CONFIG_DEFAULT_ddns-scripts_aliyun=y
CONFIG_DEFAULT_ddns-scripts_dnspod=y
CONFIG_DEFAULT_default-settings=y
CONFIG_DEFAULT_dnsmasq-full=y
CONFIG_DEFAULT_dropbear=y
CONFIG_DEFAULT_e2fsprogs=y
CONFIG_DEFAULT_ethtool=y
CONFIG_DEFAULT_firewall=y
CONFIG_DEFAULT_fstools=y
CONFIG_DEFAULT_haveged=y
CONFIG_DEFAULT_htop=y
CONFIG_DEFAULT_ip-full=y
CONFIG_DEFAULT_ipset=y
CONFIG_DEFAULT_iptables=y
CONFIG_DEFAULT_iptables-mod-extra=y
CONFIG_DEFAULT_iptables-mod-tproxy=y
CONFIG_DEFAULT_kmod-gpio-button-hotplug=y
CONFIG_DEFAULT_kmod-ipt-raw=y
CONFIG_DEFAULT_kmod-nf-nathelper=y
CONFIG_DEFAULT_kmod-nf-nathelper-extra=y
CONFIG_DEFAULT_kmod-nvme=y
CONFIG_DEFAULT_kmod-r8125=y
CONFIG_DEFAULT_kmod-scsi-core=y
CONFIG_DEFAULT_kmod-tun=y
CONFIG_DEFAULT_libc=y
CONFIG_DEFAULT_libgcc=y
CONFIG_DEFAULT_libustream-openssl=y
CONFIG_DEFAULT_logd=y
CONFIG_DEFAULT_luci=y
CONFIG_DEFAULT_luci-app-accesscontrol=y
CONFIG_DEFAULT_luci-app-arpbind=y
CONFIG_DEFAULT_luci-app-autoreboot=y
CONFIG_DEFAULT_luci-app-ddns=y
CONFIG_DEFAULT_luci-app-filetransfer=y
CONFIG_DEFAULT_luci-app-nlbwmon=y
CONFIG_DEFAULT_luci-app-ssr-plus=y
CONFIG_DEFAULT_luci-app-turboacc=y
CONFIG_DEFAULT_luci-app-upnp=y
CONFIG_DEFAULT_luci-app-vlmcsd=y
CONFIG_DEFAULT_luci-app-vsftpd=y
[... 946 lines cut ...]
```sh
/home/huzf/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/lib/gcc/aarch64-openwrt-linux-musl/11.3.0/../../../../aarch64-openwrt-linux-musl/bin/ld: /home/huzf/lede/build_dir/target-aarch64_generic_musl/hostapd-supplicant-full-internal/hostapd-2022-07-29-b704dc72/build/wpa_supplicant/src/ap/acs.o: in function `acs_request_scan':
acs.c:(.text+0x488): undefined reference to `hostapd_driver_scan'
/home/huzf/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/lib/gcc/aarch64-openwrt-linux-musl/11.3.0/../../../../aarch64-openwrt-linux-musl/bin/ld: acs.c:(.text+0x494): undefined reference to `hostapd_driver_scan'
/home/huzf/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/lib/gcc/aarch64-openwrt-linux-musl/11.3.0/../../../../aarch64-openwrt-linux-musl/bin/ld: /home/huzf/lede/build_dir/target-aarch64_generic_musl/hostapd-supplicant-full-internal/hostapd-2022-07-29-b704dc72/build/wpa_supplicant/src/ap/acs.o: in function `acs_scan_complete':
acs.c:(.text+0x564): undefined reference to `hostapd_disable_iface'
/home/huzf/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/lib/gcc/aarch64-openwrt-linux-musl/11.3.0/../../../../aarch64-openwrt-linux-musl/bin/ld: acs.c:(.text+0x570): undefined reference to `hostapd_disable_iface'
/home/huzf/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/lib/gcc/aarch64-openwrt-linux-musl/11.3.0/../../../../aarch64-openwrt-linux-musl/bin/ld: /home/huzf/lede/build_dir/target-aarch64_generic_musl/hostapd-supplicant-full-internal/hostapd-2022-07-29-b704dc72/build/wpa_supplicant/src/ap/acs.o: in function `acs_init':
acs.c:(.text+0x1198): undefined reference to `hostapd_drv_do_acs'
/home/huzf/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/lib/gcc/aarch64-openwrt-linux-musl/11.3.0/../../../../aarch64-openwrt-linux-musl/bin/ld: acs.c:(.text+0x119c): undefined reference to `hostapd_drv_do_acs'
/home/huzf/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/lib/gcc/aarch64-openwrt-linux-musl/11.3.0/../../../../aarch64-openwrt-linux-musl/bin/ld: acs.c:(.text+0x11f0): undefined reference to `hostapd_set_state'
/home/huzf/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/lib/gcc/aarch64-openwrt-linux-musl/11.3.0/../../../../aarch64-openwrt-linux-musl/bin/ld: acs.c:(.text+0x11fc): undefined reference to `hostapd_set_state'
collect2: error: ld returned 1 exit status
make[4]: *** [Makefile:2037: wpa_supplicant] Error 1
make[4]: Leaving directory '/home/huzf/lede/build_dir/target-aarch64_generic_musl/hostapd-supplicant-full-internal/hostapd-2022-07-29-b704dc72/wpa_supplicant'
make[3]: *** [Makefile:790: /home/huzf/lede/build_dir/target-aarch64_generic_musl/hostapd-supplicant-full-internal/hostapd-2022-07-29-b704dc72/.built] Error 2
make[3]: Leaving directory '/home/huzf/lede/package/network/services/hostapd'
time: package/network/services/hostapd/supplicant-full-internal/compile#22.47#1.54#23.44
    ERROR: package/network/services/hostapd failed to build (build variant: supplicant-full-internal).
make[2]: *** [package/Makefile:119: package/network/services/hostapd/compile] Error 1
make[2]: Leaving directory '/home/huzf/lede'
make[1]: *** [package/Makefile:110: /home/huzf/lede/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/huzf/lede'
make: *** [/home/huzf/lede/include/toplevel.mk:231: world] Error 2
```

ShireCiel, 2023-06-12:
The problem was initially found.
When you failed first time.
Edit file "build_dir/target-x86_64_musl/hostapd-supplicant-full-internal/hostapd-2023-03-29-bb945b98/wpa_supplicant/.config". Make "CONFIG_AP=y" and save it.
Recompile it. You will pass it.
If help you solove your problem, don't remember give a 👍 to me.

问题初步找到。
第一次编译失败得时候 编译文件"build_dir/target-x86_64_musl/hostapd-supplicant-full-internal/hostapd-2023-03-29-bb945b98/wpa_supplicant/.config". 并修改“CONFIG_AP=y”暴露出来。
重新编译。就能过了。
如果有帮助解决问题，别忘了点个赞噢。

@hammerfunctor @yhl452493373

closed 2026-09-05 by coolsnowwolf