coolsnowwolf/lede #11195: X86雲編譯“ppp compile”及“hostapd compile”出錯

dbsjacky, 2023-05-09:
X86雲編譯“ppp compile”及“hostapd compile”出錯

是否搜索了issue (使用 "x" 选择)
      没有类似的issue

 X86雲編譯“ppp compile”及“hostapd compile”出錯

路由器型号和固件版本  X86

详细日志

A：ppp compile；
    make[3] -C feeds/luci/protocols/luci-proto-bonding compile
 make[3] -C package/network/services/hostapd compile
 make[3] -C package/network/services/hostapd compile
 make[3] -C package/network/services/odhcpd compile
 make[3] -C package/network/services/ppp compile
    ERROR: package/network/services/hostapd failed to build (build variant: supplicant-full-internal).
make -r world: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 1
make[2]: Entering directory '/workdir/openwrt/scripts/config'

B：hostapd compile；
    make[3] -C feeds/passwallluci/luci-app-passwall compile
 make[3] -C package/firmware/wireless-regdb compile
 make[3] -C package/network/services/hostapd compile
 make[3] -C package/network/services/hostapd compile
    ERROR: package/network/services/hostapd failed to build (build variant: supplicant-full-internal).
make -r world: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 1
make[2]: Entering directory '/workdir/openwrt/scripts/config'
make[2]: 'conf' is up to date.
make[2]: Leaving directory '/workdir/openwrt/scripts/config'

WYC-2020, 2023-05-12:
j1 日志提供看下

dbsjacky, 2023-06-05:
acs.c:(.text+0x3a3): undefined reference to `hostapd_driver_scan'
/home/da/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/lib/gcc/x86_64-openwrt-linux-musl/11.3.0/../../../../x86_64-openwrt-linux-musl/bin/ld: /home/da/lede/build_dir/target-x86_64_musl/hostapd-supplicant-full-internal/hostapd-2022-07-29-b704dc72/build/wpa_supplicant/src/ap/acs.o: in function `acs_scan_complete':
acs.c:(.text+0xe55): undefined reference to `hostapd_disable_iface'
/home/da/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/lib/gcc/x86_64-openwrt-linux-musl/11.3.0/../../../../x86_64-openwrt-linux-musl/bin/ld: /home/da/lede/build_dir/target-x86_64_musl/hostapd-supplicant-full-internal/hostapd-2022-07-29-b704dc72/build/wpa_supplicant/src/ap/acs.o: in function `acs_init':
acs.c:(.text+0xed1): undefined reference to `hostapd_drv_do_acs'
/home/da/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/lib/gcc/x86_64-openwrt-linux-musl/11.3.0/../../../../x86_64-openwrt-linux-musl/bin/ld: acs.c:(.text+0xf1c): undefined reference to `hostapd_set_state'
collect2: error: ld returned 1 exit status
make[4]: *** [Makefile:2037: wpa_supplicant] Error 1
make[4]: Leaving directory '/home/da/lede/build_dir/target-x86_64_musl/hostapd-supplicant-full-internal/hostapd-2022-07-29-b704dc72/wpa_supplicant'
make[3]: *** [Makefile:790: /home/da/lede/build_dir/target-x86_64_musl/hostapd-supplicant-full-internal/hostapd-2022-07-29-b704dc72/.built] Error 2
make[3]: Leaving directory '/home/da/lede/package/network/services/hostapd'
time: package/network/services/hostapd/supplicant-full-internal/compile#33.67#4.47#37.15
    ERROR: package/network/services/hostapd failed to build (build variant: supplicant-full-internal).
make[2]: *** [package/Makefile:119: package/network/services/hostapd/compile] Error 1
make[2]: Leaving directory '/home/da/lede'
make[1]: *** [package/Makefile:110: /home/da/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/da/lede'
make: *** [/home/da/lede/include/toplevel.mk:231：world] 錯誤 2

> j1 日誌提供看下

acs.c:(.text+0x3a3): undefined reference to `hostapd_driver_scan'
/home/da/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/lib/gcc/x86_64-openwrt-linux-musl/11.3.0/../../../../x86_64-openwrt-linux-musl/bin/ld: /home/da/lede/build_dir/target-x86_64_musl/hostapd-supplicant-full-internal/hostapd-2022-07-29-b704dc72/build/wpa_supplicant/src/ap/acs.o: in function `acs_scan_complete':
acs.c:(.text+0xe55): undefined reference to `hostapd_disable_iface'
/home/da/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/lib/gcc/x86_64-openwrt-linux-musl/11.3.0/../../../../x86_64-openwrt-linux-musl/bin/ld: /home/da/lede/build_dir/target-x86_64_musl/hostapd-supplicant-full-internal/hostapd-2022-07-29-b704dc72/build/wpa_supplicant/src/ap/acs.o: in function `acs_init':
acs.c:(.text+0xed1): undefined reference to `hostapd_drv_do_acs'
/home/da/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/lib/gcc/x86_64-openwrt-linux-musl/11.3.0/../../../../x86_64-openwrt-linux-musl/bin/ld: acs.c:(.text+0xf1c): undefined reference to `hostapd_set_state'
collect2: error: ld returned 1 exit status
make[4]: *** [Makefile:2037: wpa_supplicant] Error 1
make[4]: Leaving directory '/home/da/lede/build_dir/target-x86_64_musl/hostapd-supplicant-full-internal/hostapd-2022-07-29-b704dc72/wpa_supplicant'
make[3]: *** [Makefile:790: /home/da/lede/build_dir/target-x86_64_musl/hostapd-supplicant-full-internal/hostapd-2022-07-29-b704dc72/.built] Error 2
make[3]: Leaving directory '/home/da/lede/package/network/services/hostapd'
time: package/network/services/hostapd/supplicant-full-internal/compile#33.67#4.47#37.15
    ERROR: package/network/services/hostapd failed to build (build variant: supplicant-full-internal).
make[2]: *** [package/Makefile:119: package/network/services/hostapd/compile] Error 1
make[2]: Leaving directory '/home/da/lede'
make[1]: *** [package/Makefile:110: /home/da/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/da/lede'
make: *** [/home/da/lede/include/toplevel.mk:231：world] 錯誤 2

closed 2023-07-31 by dbsjacky