coolsnowwolf/lede #11197: `ERROR: package/feeds/helloworld/ipt2socks failed to build.`

devioa, 2023-05-11:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题
SSRP组件依赖`package/feeds/helloworld/ipt2socks`无法`make download`
已确认并非网络环境导致的

### (2) 路由器型号和固件版本
x86

### (3) 详细日志
```
make[2] -C feeds/helloworld/dns2socks download
 make[2] -C feeds/helloworld/dns2tcp download
 make[2] -C feeds/helloworld/chinadns-ng download
 make[2] -C feeds/helloworld/lua-neturl download
 make[2] -C feeds/helloworld/ipt2socks download
 make[2] -C feeds/helloworld/luci-app-ssr-plus download
 make[2] -C feeds/helloworld/microsocks download
 make[2] -C feeds/helloworld/shadowsocksr-libev download
 make[2] -C feeds/helloworld/simple-obfs download
 make[2] -C feeds/helloworld/tcping download
 make[2] -C feeds/helloworld/trojan download
    ERROR: package/feeds/helloworld/ipt2socks failed to build.
 make[2] -C package/boot/grub2 download
 make[2] -C package/boot/grub2 download
make package/download: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
 make[1] target/download
 make[2] -C target/linux download
```
```
install -d -m0755 /home/devioa/lede/bin/packages/x86_64/helloworld
/home/devioa/lede/staging_dir/host/bin/fakeroot /home/devioa/lede/staging_dir/host/bin/bash /home/devioa/lede/scripts/ipkg-build -m "" /home/devioa/lede/build_dir/target-x86_64_musl/dns2tcp-1.1.0/ipkg-x86_64/dns2tcp /home/devioa/lede/bin/packages/x86_64/helloworld
Packaged contents of /home/devioa/lede/build_dir/target-x86_64_musl/dns2tcp-1.1.0/ipkg-x86_64/dns2tcp into /home/devioa/lede/bin/packages/x86_64/helloworld/dns2tcp_1.1.0-1_x86_64.ipk
echo "dns2tcp" >> /home/devioa/lede/staging_dir/target-x86_64_musl/pkginfo/dns2tcp.default.install
make[3]: Leaving directory '/home/devioa/lede/feeds/helloworld/dns2tcp'
time: package/feeds/helloworld/dns2tcp/compile#0.12#0.01#0.12
make[3]: Entering directory '/home/devioa/lede/feeds/helloworld/ipt2socks'
mkdir -p /home/devioa/lede/dl
SHELL= flock /home/devioa/lede/tmp/.ipt2socks-1.1.3.tar.gz.flock -c '   /home/devioa/lede/scripts/download.pl "/home/devioa/lede/dl" "ipt2socks-1.1.3.tar.gz" "5279eb1cb7555cf9292423cc9f672dc43e6e214b3411a6df26a6a1cfa59d88b7" "" "https://codeload.github.com/zfl9/ipt2socks/tar.gz/v1.1.3?"    '
Hash of the local file ipt2socks-1.1.3.tar.gz does not match (file: 73a2498dc95934c225d358707e7f7d060b5ce81aa45260ada09cbd15207d27d1, requested: 5279eb1cb7555cf9292423cc9f672dc43e6e214b3411a6df26a6a1cfa59d88b7) - deleting download.
make[3]: *** [Makefile:46: /home/devioa/lede/dl/ipt2socks-1.1.3.tar.gz] Error 2
make[3]: Leaving directory '/home/devioa/lede/feeds/helloworld/ipt2socks'
time: package/feeds/helloworld/ipt2socks/compile#0.07#0.00#0.07
    ERROR: package/feeds/helloworld/ipt2socks failed to build.
make[2]: *** [package/Makefile:116: package/feeds/helloworld/ipt2socks/compile] Error 1
make[2]: Leaving directory '/home/devioa/lede'
make[1]: *** [package/Makefile:110: /home/devioa/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/devioa/lede'
make: *** [/home/devioa/lede/include/toplevel.mk:231: world] Error 2
```

WYC-2020, 2023-05-11:
[image]
不是不能下载而是下载后hash 校验不过，这玩意 我是直接吧hash校验关闭了，改成skip 即可，因为这个下载地址有好几个，有的库不同的下载地址下载下来 hash不是一样的就会导致这个问题

devioa, 2023-05-11:
> [image] 不是不能下载而是下载后hash 校验不过，这玩意 我是直接吧hash校验关闭了，改成skip 即可，因为这个下载地址有好几个，有的库不同的下载地址下载下来 hash不是一样的就会导致这个问题

where and how to do

devioa, 2023-05-11:
> [image] 不是不能下载而是下载后hash 校验不过，这玩意 我是直接吧hash校验关闭了，改成skip 即可，因为这个下载地址有好几个，有的库不同的下载地址下载下来 hash不是一样的就会导致这个问题

这个？
https://github.com/coolsnowwolf/lede/blob/4fc044df145c793de11c2552d336958c3b91c8d1/Makefile#L109

devioa, 2023-05-11:
> > [image] 不是不能下载而是下载后hash 校验不过，这玩意 我是直接吧hash校验关闭了，改成skip 即可，因为这个下载地址有好几个，有的库不同的下载地址下载下来 hash不是一样的就会导致这个问题
> 
> 这个？
> 
> https://github.com/coolsnowwolf/lede/blob/4fc044df145c793de11c2552d336958c3b91c8d1/Makefile#L109

试了很显然也不是，建议如果可以提供问题的解决方案，至少让人知道从哪儿入手以及如何去修改
而不是解释一大堆这个问题的缘由，很显然提出issues的人只想知道问题如何解决而不是很想知道为什么会出现这个问题

devioa, 2023-05-11:
> 你可真是人才，稍微看过库升级的提交都知道在哪里改，在这个包的目录下，makefile里面，hash修改成skip，不知道你看这个英语也知道搜索关键字嘛，直接搜索那个hash也知道怎么改

你自己都在[Discussions](https://github.com/coolsnowwolf/lede/discussions/11194#discussioncomment-5847313)说了最近commit这么乱，而我正常一两个月pull一次我怎么知道是哪个commit引起的？既然有问题，那么我自然是先在issues里post
另外`PKG_HASH`这个参数，一般没有足够相关基础的人
都只会认为这只有哈希值这一种情况而不是还可以传入类似`skip`之类的method吧？
搜关键字我会，但是你一没告知操作二没告知位置，多少有点无米之炊了

> 试了很显然也不是，建议如果可以提供问题的解决方案，至少让人知道从哪儿入手以及如何去修改
而不是解释一大堆这个问题的缘由，很显然提出issues的人只想知道问题如何解决而不是很想知道为什么会出现这个问题

这段话仅仅只是友情给个建议，你不用这幅看不起人的口吻来和我对话，没必要和谁彰显你的傲慢，大家都是普通人
我有问题，你有解决方法，好，那么非常感谢你能提供，如果能直接说明该怎么做在哪儿做就更好了

Greeting and Thanks.

closed 2023-05-11 by devioa

devioa, 2023-05-11:
> > > 你可真是人才，稍微看过库升级的提交都知道在哪里改，在这个包的目录下，makefile里面，hash修改成skip，不知道你看这个英语也知道搜索关键字嘛，直接搜索那个hash也知道怎么改
> > 
> > 
> > 你自己都在[Discussions](https://github.com/coolsnowwolf/lede/discussions/11194#discussioncomment-5847313)说了最近commit这么乱，而我正常一两个月pull一次我怎么知道是哪个commit引起的？既然有问题，那么我自然是先在issues里post 另外`PKG_HASH`这个参数，一般没有足够相关基础的人 都只会认为这只有哈希值这一种情况而不是还可以传入类似`skip`之类的method吧？ 搜关键字我会，但是你一没告知操作二没告知位置，多少有点无米之炊了
> > > 试了很显然也不是，建议如果可以提供问题的解决方案，至少让人知道从哪儿入手以及如何去修改
> > > 而不是解释一大堆这个问题的缘由，很显然提出issues的人只想知道问题如何解决而不是很想知道为什么会出现这个问题
> > 
> > 
> > 这段话仅仅只是友情给个建议，你不用这幅看不起人的口吻来和我对话，没必要和谁彰显你的傲慢，大家都是普通人 我有问题，你有解决方法，好，那么非常感谢你能提供，如果能直接说明该怎么做在哪儿做就更好了
> > Greeting and Thanks.
> 
> 谁叫大佬NB，我们不NB呢？被大佬奚落几句，能拿到解决方案，习惯就好:P

你这种想法多少有点卑微了，还生而为人呢，没谁欠着谁 :)

WYC-2020, 2023-05-12:
> 谁叫大佬NB，我们不NB呢？被大佬奚落几句，能拿到解决方案，习惯就好:P

谁对谁错自己知道，我倒是没看出你那里卑微，还一脸的责怪给你答复的人为啥不提供详细的步骤，为啥
1这个库一看就是隔壁的，你不应该在这里提
2隔壁关闭的issue一堆这个的解决方法自己不去搜索看
3都在走编译了，已经告诉你修改方法，自己不会请再问，别一脸责怪别人没给你详细步骤
4在授人以鱼不如授人以渔的告诉原因，你却不要在意知道，这算我多此一举，工具人
 本就不打算答复你的，看你还在这里得理不让人的样子看不惯

devioa, 2023-05-12:
> 谁对谁错自己知道，我倒是没看出你那里卑微，还一脸的责怪给你答复的人为啥不提供详细的步骤，为啥 1这个库一看就是隔壁的，你不应该在这里提 2隔壁关闭的issue一堆这个的解决方法自己不去搜索看 3都在走编译了，已经告诉你修改方法，自己不会请再问，别一脸责怪别人没给你详细步骤 4在授人以鱼不如授人以渔的告诉原因，你却不要在意知道，这算我多此一举，工具人 [...]

1. 请问我为什么要卑微？我不卑不亢开个issues有什么问题？
2. 请问我哪儿责怪了任何人？我一没骂人二没阴阳怪气，反倒是你一副傲慢的态度说`你可真是人才，稍微看过库升级的提交都知道在哪里改`
3. 其次，我平常只关注`lede`主库并不关注`package`或`luci`其它库，我日常工作很忙，我只会把我有限的精力放在最应当关注的事情上，而且issues置顶贴有说此类issues`不应该`在这里post么？
4. 再者，你的回答我没有看出`渔`在哪里，只看到一堆毫无必要的解释，没有直接点题地回答，反而让人去猜
5. 关于你说的`自己不会请再问`，请看你自己的第三条comment
6. 最后，整篇下来我的任何一段文字有任何地方体现出`得理不让人的样子`么？不建议你生造概念，我只是把该说的说了，不会像一些人一样把自己的位置放得很卑微，你不是我的领导不是我长辈，我没有任何义务被任何人所谓的`奚落`且我也不建议你这样去做，毕竟生而为人，人人有自尊且应当`自尊`:)

devioa, 2023-05-12:
> 开源项目本身和所有用户都是相辅相成的，所有人都在一个社区里根据自身的能力共同维护一个项目这是开源的意义之一，并不建议任何人在这里施行`官僚主义`的那套以上凌下的态度去看待这个社区里的每一个人，开源社区里每一个人都是平等的。
我有问题且并不是在我主观意志的操作下造成的、且我已经提供了足够清晰的`Troubleshoting Logs`来供仓库维护者或有能力解答的爱好者参考并判断问题，那么与之对等的，我自然也希望能够得到足够清晰且准确的解决方法来解决这个问题。

以上是我提出此`issues`的初心，如果对此你有任何误解，请见谅。
我并无任何毫无必要、所谓`得理不让人`的争论之心。
此`issues`并不欢迎任何抱着动机不纯前来争论的人，并且此issues已被解决且关闭、归档供后来遇到相同问题的人。
所以点到为止。