coolsnowwolf/lede #11198: kernel: add support for threaded network backlog processing 这个功能在5.15上会kernel panic

GreatMichaelLee, 2023-05-11:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：复现步骤：
step1: 编译x86 5.15最新基线代码
step2: 在rc.local里加入echo 1 > /proc/sys/net/core/backlog_threaded 使系统启动时启用此功能
step3: 终端反复输出Kernel panic的日志，系统崩溃
step4: 如取消在rc.local里加入 echo 1 > /proc/sys/net/core/backlog_threaded 去使能该功能，则系统能正常运作。

### (2) 路由器型号和固件版本

A：x86软路由: VMWARE 8.0.1 ESXI+ N5105 4*2.5G 
kernel 5.15.110 + OpenWrt R23.5.1

### (3) 详细日志

A：终端反复输出Kernel panic的日志，系统崩溃

Thu May 11 16:09:51 2023 kern.warn kernel: [   28.981479] ------------[ cut here ]------------
Thu May 11 16:09:51 2023 kern.warn kernel: [   28.984894] WARNING: CPU: 2 PID: 7977 at napi_threaded_poll+0xc8/0xd0
Thu May 11 16:09:51 2023 kern.warn kernel: [   28.985805] Modules linked in: ksmbd pppoe ppp_async wireguard snd_usb_audio snd_hda_intel snd_hda_codec_via snd_hda_codec_realtek snd_hda_codec_hdmi s
Thu May 11 16:09:51 2023 kern.warn kernel: [   28.985870]  poly1305_x86_64 pcnet32 nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject_bridge nft_reject nft_redir nft_quota nft_objref nft_nu
Thu May 11 16:09:51 2023 kern.warn kernel: [   28.998110]  sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact confi
Thu May 11 16:09:51 2023 kern.warn kernel: [   29.011677]  rpcsec_gss_krb5 auth_rpcgss veth tun snd_rawmidi snd_seq_device snd_pcm_oss snd_mixer_oss snd_hwdep snd_compress snd_pcm snd_timer snd sou
Thu May 11 16:09:51 2023 kern.warn kernel: [   29.048489] CPU: 2 PID: 7977 Comm: napi/backlog-3 Not tainted 5.15.110 #0
Thu May 11 16:09:51 2023 kern.warn kernel: [   29.049950] Hardware name: VMware, Inc. VMware7,1/440BX Desktop Reference Platform, BIOS VMW71.00V.20829224.B64.2211211837 11/21/2022
Thu May 11 16:09:51 2023 kern.warn kernel: [   29.055552] RIP: 0010:napi_threaded_poll+0xc8/0xd0
Thu May 11 16:09:51 2023 kern.warn kernel: [   29.056901] Code: 00 00 48 8d 75 e7 48 89 df e8 34 fe ff ff be 00 02 00 00 48 c7 c7 e1 ba 72 81 e8 23 44 99 ff 80 7d e7 00 75 c9 e9 51 ff ff ff <0f> 0b
Thu May 11 16:09:51 2023 kern.warn kernel: [   29.059972] RSP: 0000:ffffc9000730fee0 EFLAGS: 00010286
Thu May 11 16:09:51 2023 kern.warn kernel: [   29.062217] RAX: 0000000000000000 RBX: ffff88807dda2c90 RCX: 0000000000000000
Thu May 11 16:09:51 2023 kern.warn kernel: [   29.065518] RDX: 0000000000000000 RSI: 0000000000000002 RDI: ffff88807dd21b80
Thu May 11 16:09:51 2023 kern.warn kernel: [   29.067054] RBP: ffffc9000730ff00 R08: 0000000000000800 R09: 0000000000000000
Thu May 11 16:09:51 2023 kern.warn kernel: [   29.068521] R10: 0000000000000001 R11: 0000000000000000 R12: ffff888003a70000
Thu May 11 16:09:51 2023 kern.warn kernel: [   29.069947] R13: 0000000000000001 R14: ffffc900072ffc90 R15: ffff88807dda2c90
Thu May 11 16:09:51 2023 kern.warn kernel: [   29.071388] FS:  0000000000000000(0000) GS:ffff88807dd00000(0000) knlGS:0000000000000000
Thu May 11 16:09:51 2023 kern.warn kernel: [   29.075454] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Thu May 11 16:09:51 2023 kern.warn kernel: [   29.076836] CR2: 00007fe50e4b2000 CR3: 00000000109b2000 CR4: 0000000000350ee0
Thu May 11 16:09:51 2023 kern.warn kernel: [   29.078339] Call Trace:
Thu May 11 16:09:51 2023 kern.warn kernel: [   29.079632]  <TASK>
Thu May 11 16:09:51 2023 kern.warn kernel: [   29.080860]  ? __kthread_parkme+0x46/0x60
Thu May 11 16:09:51 2023 kern.warn kernel: [   29.082282]  kthread+0x125/0x150
Thu May 11 16:09:51 2023 kern.warn kernel: [   29.086354]  ? __napi_poll+0x120/0x120
Thu May 11 16:09:51 2023 kern.warn kernel: [   29.087705]  ? set_kthread_struct+0x40/0x40
Thu May 11 16:09:51 2023 kern.warn kernel: [   29.088922]  ret_from_fork+0x1f/0x30
Thu May 11 16:09:51 2023 kern.warn kernel: [   29.090117]  </TASK>
Thu May 11 16:09:51 2023 kern.warn kernel: [   29.091212] ---[ end trace 6abe307e2eccae15 ]---

WYC-2020, 2023-05-11:
6.1不行是因为中断的方式被改过，6.1以下目测没什么问题，5.15我试过没有你这种情况，这玩意开启会有一堆backlog，你ps看就知道，反正我不喜欢没开哈哈

GreatMichaelLee, 2023-05-11:
我这6.1跟5.15是同样的现象…

closed 2023-05-11 by aiamadeus

GreatMichaelLee, 2023-05-11:
> 6.1不行是因为中断的方式被改过，6.1以下目测没什么问题，5.15我试过没有你这种情况，这玩意开启会有一堆backlog，你ps看就知道，反正我不喜欢没开哈哈

5.15上测了，修复了，6.1上改了中断方式是NA了么？能不能也patch一下啊....想用6.1,感谢大佬们@aiamadeus @WYC-2020