coolsnowwolf/lede #11199: gcc 13 下 zoneinfo 编译失败（需降级至 gcc 12）

KevinMX, 2023-05-11:
## 1.关于你要提交的问题

`zoneinfo` 包在 `gcc 13.1.1` 下无法通过编译。

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

软件包 `zoneinfo` 在 `gcc 13` 下无法通过编译。尝试降级主机 `gcc` 至 `12` 后编译通过。

Host: Arch Linux on WSL 2, 已安装 `README` 中的提示设置了环境变量。

针对已更新至 `gcc 13` 的发行版: 支持 `gcc 13` 之前，请降级至 `gcc 12.2.1`。

上游主线似乎刚刚合并了 `gcc 13` 支持，见 https://github.com/openwrt/openwrt/pull/12489 ，不知道这边有何打算。

e.g. 启用了 `archlinuxcn` 源的情况下进行软件包降级：

```bash
sudo pacman -S downgrade
sudo downgrade gcc-libs gcc
# 选择两个软件包的 12.2.1-4 版本
# 重新进行构建
```

### (2) 路由器型号和固件版本

P&W R619AC 128M, coolsnowwolf/lede master branch at 4fc044df145c793de11c2552d336958c3b91c8d1

理论上影响所有 target？

### (3) 详细日志

尝试单独构建 `zoneinfo` 失败：

<details><summary>日志已折叠，点击展开</summary><p>

```bash
mx at ArchWSL in ~/lede on master*
$ make package/feeds/packages/zoneinfo/compile V=s
make[2]: Entering directory '/home/mx/lede/scripts/config'
make[2]: 'conf' is up to date.
make[2]: Leaving directory '/home/mx/lede/scripts/config'
make[1]: Entering directory '/home/mx/lede'
make[2]: Entering directory '/home/mx/lede/package/libs/toolchain'
echo "libc" >> /home/mx/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/pkginfo/toolchain.default.install
echo "libgcc" >> /home/mx/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/pkginfo/toolchain.default.install
echo "libatomic" >> /home/mx/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/pkginfo/toolchain.default.install
echo "libstdcpp" >> /home/mx/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/pkginfo/toolchain.default.install
echo "libpthread" >> /home/mx/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/pkginfo/toolchain.default.install
echo "librt" >> /home/mx/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/pkginfo/toolchain.default.install
make[2]: Leaving directory '/home/mx/lede/package/libs/toolchain'
time: package/libs/toolchain/compile#0.13#0.00#0.16
make[2]: Entering directory '/home/mx/lede/feeds/packages/utils/zoneinfo'
rm -f /home/mx/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/zoneinfo-2022g/.built
touch /home/mx/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/zoneinfo-2022g/.built_check
CFLAGS="-O2 -I/home/mx/lede/staging_dir/host/include -I/home/mx/lede/staging_dir/hostpkg/include -I/home/mx/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/host/include" make -C /home/mx/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/zoneinfo-2022g -f Makefile  CC="gcc" LD="\$(CC)" CPPFLAGS="-I/home/mx/lede/staging_dir/host/include -I/home/mx/lede/staging_dir/hostpkg/include -I/home/mx/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/host/include -DHAVE_SNPRINTF=1 -DZIC_BLOAT_DEFAULT='\"fat\"'" LDFLAGS="-L/home/mx/lede/staging_dir/host/lib -L/home/mx/lede/staging_dir/hostpkg/lib -L/home/mx/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/host/lib" TOPDIR="/home/mx/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/zoneinfo-2022g/ipkg-install" TZDIR="/home/mx/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/zoneinfo-2022g/ipkg-install/zoneinfo" install
make[3]: Entering directory '/home/mx/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/zoneinfo-2022g'
gcc  -I/home/mx/lede/staging_dir/host/include -I/home/mx/lede/staging_dir/hostpkg/include -I/home/mx/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/host/include -DHAVE_SNPRINTF=1 -DZIC_BLOAT_DEFAULT='"fat"'  -c -o zic.o zic.c
zic.c:462:1: warning: 'noreturn' attribute ignored [-Wattributes]
  462 | static ATTRIBUTE_NORETURN void
      | ^~~~~~
zic.c:462:27: error: expected identifier or '(' before 'void'
  462 | static ATTRIBUTE_NORETURN void
      |                           ^~~~
zic.c:469:1: warning: 'noreturn' attribute ignored [-Wattributes]
  469 | static ATTRIBUTE_NORETURN void
[... 179 lines cut ...]
mkdir -p /home/mx/lede/bin/targets/ipq40xx/generic/packages /home/mx/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/zoneinfo-2022g/ipkg-arm_cortex-a7_neon-vfpv4/zoneinfo-asia/CONTROL /home/mx/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/pkginfo
install -d -m0755 /home/mx/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/zoneinfo-2022g/ipkg-arm_cortex-a7_neon-vfpv4/zoneinfo-asia/usr/share/zoneinfo
for i in Asia Japan Singapore Hongkong ; do cp -fpR /home/mx/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/zoneinfo-2022g/ipkg-install/zoneinfo/$i /home/mx/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/zoneinfo-2022g/ipkg-arm_cortex-a7_neon-vfpv4/zoneinfo-asia/usr/share/zoneinfo ; done
find /home/mx/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/zoneinfo-2022g/ipkg-arm_cortex-a7_neon-vfpv4/zoneinfo-asia -name 'CVS' -o -name '.svn' -o -name '.#*' -o -name '*~'| xargs -r rm -rf
export CROSS="arm-openwrt-linux-muslgnueabi-"  NO_RENAME=1 ; NM="arm-openwrt-linux-muslgnueabi-nm" STRIP="/home/mx/lede/staging_dir/host/bin/sstrip -z" STRIP_KMOD="/home/mx/lede/scripts/strip-kmod.sh" PATCHELF="/home/mx/lede/staging_dir/host/bin/patchelf" /home/mx/lede/scripts/rstrip.sh /home/mx/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/zoneinfo-2022g/ipkg-arm_cortex-a7_neon-vfpv4/zoneinfo-asia
(cd /home/mx/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/zoneinfo-2022g/ipkg-arm_cortex-a7_neon-vfpv4/zoneinfo-asia/CONTROL; ( echo "$CONTROL"; printf "Description: "; echo "$DESCRIPTION" | sed -e 's,^[[:space:]]*, ,g'; ) > control; chmod 644 control; ( echo "#!/bin/sh"; echo "[ \"\${IPKG_NO_SCRIPT}\" = \"1\" ] && exit 0"; echo "[ -s "\${IPKG_INSTROOT}/lib/functions.sh" ] || exit 0"; echo ". \${IPKG_INSTROOT}/lib/functions.sh"; echo "default_postinst \$0 \$@"; ) > postinst; ( echo "#!/bin/sh"; echo "[ -s "\${IPKG_INSTROOT}/lib/functions.sh" ] || exit 0"; echo ". \${IPKG_INSTROOT}/lib/functions.sh"; echo "default_prerm \$0 \$@"; ) > prerm; chmod 0755 postinst prerm;  )
install -d -m0755 /home/mx/lede/bin/packages/arm_cortex-a7_neon-vfpv4/packages
/home/mx/lede/staging_dir/host/bin/fakeroot /home/mx/lede/staging_dir/host/bin/bash /home/mx/lede/scripts/ipkg-build -m "" /home/mx/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/zoneinfo-2022g/ipkg-arm_cortex-a7_neon-vfpv4/zoneinfo-asia /home/mx/lede/bin/packages/arm_cortex-a7_neon-vfpv4/packages
Packaged contents of /home/mx/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/zoneinfo-2022g/ipkg-arm_cortex-a7_neon-vfpv4/zoneinfo-asia into /home/mx/lede/bin/packages/arm_cortex-a7_neon-vfpv4/packages/zoneinfo-asia_2022g-1_arm_cortex-a7_neon-vfpv4.ipk
rm -rf /home/mx/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/zoneinfo-2022g/.pkgdir/zoneinfo-asia.installed /home/mx/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/zoneinfo-2022g/.pkgdir/zoneinfo-asia
mkdir -p /home/mx/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/zoneinfo-2022g/.pkgdir/zoneinfo-asia
install -d -m0755 /home/mx/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/zoneinfo-2022g/.pkgdir/zoneinfo-asia/usr/share/zoneinfo
for i in Asia Japan Singapore Hongkong ; do cp -fpR /home/mx/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/zoneinfo-2022g/ipkg-install/zoneinfo/$i /home/mx/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/zoneinfo-2022g/.pkgdir/zoneinfo-asia/usr/share/zoneinfo ; done
touch /home/mx/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/zoneinfo-2022g/.pkgdir/zoneinfo-asia.installed
mkdir -p /home/mx/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/stamp
SHELL= flock /home/mx/lede/tmp/.root-copy.flock -c 'cp -fpR /home/mx/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/zoneinfo-2022g/.pkgdir/zoneinfo-asia/. /home/mx/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/'
touch /home/mx/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/stamp/.zoneinfo-asia_installed
echo "zoneinfo-asia" >> /home/mx/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/pkginfo/zoneinfo.default.install
make[2]: Leaving directory '/home/mx/lede/feeds/packages/utils/zoneinfo'
time: package/feeds/packages/zoneinfo/compile#1.04#0.11#1.14
make[1]: Leaving directory '/home/mx/lede'
mx at ArchWSL in ~/lede on master*
$
```
</p>

WYC-2020, 2023-05-11:
没函数声明而已，都不能算bug，这边没和，你应该去官方反馈，虽然我合并了，gcc13可以明确告诉你很多库都需要加一些选项才能编译过

KevinMX, 2023-05-11:
> 没函数声明而已，都不能算bug，这边没和，你应该去官方反馈，虽然我合并了，gcc13可以明确告诉你很多库都需要加一些选项才能编译过

主要是给用上新版本编译器的用户提个醒咯。

上游那边我周末有时间再看看。

WYC-2020, 2023-05-11:
flag 加一个这个 -Wno-implicit-function-declaration 应该就可以过

KevinMX, 2023-05-20:
> flag 加一个这个 -Wno-implicit-function-declaration 应该就可以过

尝试加到 CFLAGS & CPPFLAGS 里，似乎并不行。

观察到 openwrt 上游似乎也在处理 GCC 13 带来的问题（而且还不小），https://github.com/openwrt/packages/issues/20994 ，稍候我会尝试一下 openwrt 主线版本的 zoneinfo 能不能编译过，如果也复现的话就直接提到上游了。

这边的 issue 暂时先挂起吧。感谢大佬指点。

WYC-2020, 2023-05-20:
> > flag 加一个这个 -Wno-implicit-function-declaration 应该就可以过
> 
> 尝试加到 CFLAGS & CPPFLAGS 里，似乎并不行。
> 
> 观察到 openwrt 上游似乎也在处理 GCC 13 带来的问题（而且还不小），[openwrt/packages#20994](https://github.com/openwrt/packages/issues/20994) ，稍候我会尝试一下 openwrt 主线版本的 zoneinfo 能不能编译过，如果也复现的话就直接提到上游了。
> 
> 这边的 issue 暂时先挂起吧。感谢大佬指点。
> 
> Update: 主线 openwrt 那边的 zoneinfo 在 GCC 13 下是正常过编译的。

是的我这边升级到13编译也是没问题的

closed 2023-05-20 by aiamadeus