coolsnowwolf/lede #11200: x86虚拟机流量分载可能有问题

liwenjie119, 2023-05-11:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：启动后查看Turbo ACC 网络加速设置，先能看到流量分载未运行其他正常，等一会儿所有都是正在收集数据...
### (2) 路由器型号和固件版本

A：x86询价

### (3) 详细日志

A：[    0.000000] Linux version 5.15.110 (liwenjie@localhost) (x86_64-openwrt-linux-musl-gcc (OpenWrt GCC 11.3.0 r5955-4fc044df1) 11.3.0, GNU ld (GNU Binutils) 2.37) #0 SMP Wed May 10 14:00:09 2023
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz root=PARTUUID=8a603d76-7ff4-62fd-2280-9ac30b91fa02 rootwait console=tty0 console=ttyS0,115200n8 noinitrd
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
[    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[    0.000000] x86/fpu: Enabled xstate features 0x7, context size is 832 bytes, using 'standard' format.
[    0.000000] signal: max sigframe size: 1360
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009ffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x00000000007fffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000000800000-0x0000000000807fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x0000000000808000-0x000000000080afff] usable
[    0.000000] BIOS-e820: [mem 0x000000000080b000-0x000000000080bfff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000000080c000-0x000000000080ffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000000810000-0x00000000008fffff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x0000000000900000-0x000000003f8eefff] usable
[    0.000000] BIOS-e820: [mem 0x000000003f8ef000-0x000000003f9eefff] reserved
[    0.000000] BIOS-e820: [mem 0x000000003f9ef000-0x000000003faeefff] type 20
[    0.000000] BIOS-e820: [mem 0x000000003faef000-0x000000003fb6efff] reserved
[    0.000000] BIOS-e820: [mem 0x000000003fb6f000-0x000000003fb7efff] ACPI data
[    0.000000] BIOS-e820: [mem 0x000000003fb7f000-0x000000003fbfefff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000003fbff000-0x000000003ff1ffff] usable
[    0.000000] BIOS-e820: [mem 0x000000003ff20000-0x000000003ff3ffff] reserved
[    0.000000] BIOS-e820: [mem 0x000000003ff40000-0x000000003fffffff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000ffe00000-0x00000000ffffffff] reserved
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] efi: EFI v2.70 by EDK II
[    0.000000] efi: SMBIOS=0x3f9ab000 ACPI=0x3fb7e000 ACPI 2.0=0x3fb7e014 MEMATTR=0x3edf0018 
[    0.000000] SMBIOS 2.8 present.
[    0.000000] DMI: QEMU Standard PC (i440FX + PIIX, 1996), BIOS 0.0.0 02/06/2015
[    0.000000] Hypervisor detected: KVM
[    0.000000] kvm-clock: Using msrs 4b564d01 and 4b564d00
[    0.000000] kvm-clock: cpu 0, msr 448b001, primary cpu clock
[    0.000001] kvm-clock: using sched offset of 89270563311 cycles
[    0.000006] clocksource: kvm-clock: mask: 0xffffffffffffffff max_cycles: 0x1cd42e4dffb, max_idle_ns: 881590591483 ns
[    0.000013] tsc: Detected 2494.332 MHz processor
[    0.000122] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000125] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000133] last_pfn = 0x3ff20 max_arch_pfn = 0x400000000
[    0.000776] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WP  UC- WT  
[    0.024772] Secure boot disabled
[    0.024815] ACPI: Early table checksum verification disabled
[    0.024838] ACPI: RSDP 0x000000003FB7E014 000024 (v02 BOCHS )
[    0.024852] ACPI: XSDT 0x000000003FB7D0E8 000044 (v01 BOCHS  BXPC     00000001      01000013)
[    0.024870] ACPI: FACP 0x000000003FB7A000 000074 (v01 BOCHS  BXPC     00000001 BXPC 00000001)
[    0.024880] ACPI: DSDT 0x000000003FB7B000 0018E3 (v01 BOCHS  BXPC     00000001 BXPC 00000001)
[    0.024890] ACPI: FACS 0x000000003FBDD000 000040
[    0.024897] ACPI: APIC 0x000000003FB79000 0000B0 (v01 BOCHS  BXPC     00000001 BXPC 00000001)
[    0.024904] ACPI: WAET 0x000000003FB78000 000028 (v01 BOCHS  BXPC     00000001 BXPC 00000001)
[    0.024911] ACPI: BGRT 0x000000003FB77000 000038 (v01 INTEL  EDK2     00000002      01000013)
[    0.024915] ACPI: Reserving FACP table memory at [mem 0x3fb7a000-0x3fb7a073]
[... 483 lines cut ...]
[    8.147128] usbcore: registered new interface driver ums-karma
[    8.149672] usbcore: registered new interface driver ums-sddr09
[    8.152336] usbcore: registered new interface driver ums-sddr55
[    8.154857] usbcore: registered new interface driver ums-usbat
[    8.188453] xt_time: kernel timezone is -0000
[    8.330985] [drm] amdgpu kernel modesetting enabled.
[    8.335539] usbcore: registered new interface driver aqc111
[    8.338128] usbcore: registered new interface driver ax88179_178a
[    8.361536] PPP generic driver version 2.4.2
[    8.365477] PPP MPPE Compression module registered
[    8.367643] NET: Registered PF_PPPOX protocol family
[    8.378042] usbcore: registered new interface driver snd-usb-audio
[    8.381624] wireguard: WireGuard 1.0.0 loaded. See www.wireguard.com for information.
[    8.389617] wireguard: Copyright (C) 2015-2019 Jason A. Donenfeld <Jason@zx2c4.com>. All Rights Reserved.
[    8.395408] batman_adv: B.A.T.M.A.N. advanced 2022.3-openwrt-1 (compatibility version 15) loaded
[    8.438699] kmodloader: 1 module could not be probed
[    8.442653] kmodloader: - oaf - 0
[   11.311896] eth0: 0xffffc90000085000, 00:e0:4c:16:49:65, IRQ 44
[   14.479384] r8168: eth0: link up
[   14.482141] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[  103.999649] init oaf sysctl...ok
[  116.502864] NFSD: Using /var/lib/nfs/v4recovery as the NFSv4 state recovery directory
[  116.508711] NFSD: Using legacy client tracking operations.
[  116.511258] NFSD: starting 10-second grace period (net f0000000)
[  137.097313] TCP: request_sock_TCP: Possible SYN flooding on port 5000. Sending cookies.  Check SNMP counters.

WYC-2020, 2023-05-12:
看你 开了 oaf 和 那个wireguard  以前相同配置没有这个问题么？还是说 你是第一次编译这个配置，是的话 你可以关闭oaf和wireguard 在试试，有可能是什么不兼容，最好是全新安装，别保留什么配置之类的

liwenjie119, 2023-05-12:
> 看你 开了 oaf 和 那个wireguard 以前相同配置没有这个问题么？还是说 你是第一次编译这个配置，是的话 你可以关闭oaf和wireguard 在试试，有可能是什么不兼容，最好是全新安装，别保留什么配置之类的

oaf编译进去了但是没开，wireguard应该是默认编译带的，没有开启。全新的配置看了一下也有这个问题，等会儿去除oaf再编译一遍看看是不是还这样

liwenjie119, 2023-05-12:
> 看你 开了 oaf 和 那个wireguard 以前相同配置没有这个问题么？还是说 你是第一次编译这个配置，是的话 你可以关闭oaf和wireguard 在试试，有可能是什么不兼容，最好是全新安装，别保留什么配置之类的

确认了一下，至少不是oaf的问题，去除oaf编译仍有此问题。
[    0.000000] Linux version 5.15.110 (liwenjie@localhost) (x86_64-openwrt-linux-musl-gcc (OpenWrt GCC 11.3.0 r5955-4fc044df1) 11.3.0, GNU ld (GNU Binutils) 2.37) #0 SMP Fri May 12 06:55:48 2023
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz root=PARTUUID=ec469ec6-5fb9-a22c-8828-5ebc52041e02 rootwait console=tty0 console=ttyS0,115200n8 noinitrd
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
[    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[    0.000000] x86/fpu: Enabled xstate features 0x7, context size is 832 bytes, using 'standard' format.
[    0.000000] signal: max sigframe size: 1360
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009ffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x00000000007fffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000000800000-0x0000000000807fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x0000000000808000-0x000000000080afff] usable
[    0.000000] BIOS-e820: [mem 0x000000000080b000-0x000000000080bfff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000000080c000-0x000000000080ffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000000810000-0x00000000008fffff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x0000000000900000-0x000000003f8eefff] usable
[    0.000000] BIOS-e820: [mem 0x000000003f8ef000-0x000000003f9eefff] reserved
[    0.000000] BIOS-e820: [mem 0x000000003f9ef000-0x000000003faeefff] type 20
[    0.000000] BIOS-e820: [mem 0x000000003faef000-0x000000003fb6efff] reserved
[    0.000000] BIOS-e820: [mem 0x000000003fb6f000-0x000000003fb7efff] ACPI data
[    0.000000] BIOS-e820: [mem 0x000000003fb7f000-0x000000003fbfefff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000003fbff000-0x000000003ff1ffff] usable
[    0.000000] BIOS-e820: [mem 0x000000003ff20000-0x000000003ff3ffff] reserved
[    0.000000] BIOS-e820: [mem 0x000000003ff40000-0x000000003fffffff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000ffe00000-0x00000000ffffffff] reserved
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] efi: EFI v2.70 by EDK II
[    0.000000] efi: SMBIOS=0x3f9ab000 ACPI=0x3fb7e000 ACPI 2.0=0x3fb7e014 MEMATTR=0x3edf0018 
[    0.000000] SMBIOS 2.8 present.
[    0.000000] DMI: QEMU Standard PC (i440FX + PIIX, 1996), BIOS 0.0.0 02/06/2015
[    0.000000] Hypervisor detected: KVM
[    0.000000] kvm-clock: Using msrs 4b564d01 and 4b564d00
[    0.000000] kvm-clock: cpu 0, msr 448b001, primary cpu clock
[    0.000001] kvm-clock: using sched offset of 335917251617 cycles
[    0.000003] clocksource: kvm-clock: mask: 0xffffffffffffffff max_cycles: 0x1cd42e4dffb, max_idle_ns: 881590591483 ns
[    0.000007] tsc: Detected 2494.332 MHz processor
[    0.000125] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000128] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000135] last_pfn = 0x3ff20 max_arch_pfn = 0x400000000
[    0.001506] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WP  UC- WT  
[    0.026025] Secure boot disabled
[    0.026098] ACPI: Early table checksum verification disabled
[    0.026138] ACPI: RSDP 0x000000003FB7E014 000024 (v02 BOCHS )
[    0.026159] ACPI: XSDT 0x000000003FB7D0E8 000044 (v01 BOCHS  BXPC     00000001      01000013)
[    0.026180] ACPI: FACP 0x000000003FB7A000 000074 (v01 BOCHS  BXPC     00000001 BXPC 00000001)
[    0.026193] ACPI: DSDT 0x000000003FB7B000 0018E3 (v01 BOCHS  BXPC     00000001 BXPC 00000001)
[    0.026204] ACPI: FACS 0x000000003FBDD000 000040
[    0.026212] ACPI: APIC 0x000000003FB79000 0000B0 (v01 BOCHS  BXPC     00000001 BXPC 00000001)
[    0.026219] ACPI: WAET 0x000000003FB78000 000028 (v01 BOCHS  BXPC     00000001 BXPC 00000001)
[    0.026227] ACPI: BGRT 0x000000003FB77000 000038 (v01 INTEL  EDK2     00000002      01000013)
[    0.026230] ACPI: Reserving FACP table memory at [mem 0x3fb7a000-0x3fb7a073]
[    0.026236] ACPI: Reserving DSDT table memory at [mem 0x3fb7b000-0x3fb7c8e2]
[    0.026237] ACPI: Reserving FACS table memory at [mem 0x3fbdd000-0x3fbdd03f]
[    0.026238] ACPI: Reserving APIC table memory at [mem 0x3fb79000-0x3fb790af]
[    0.026239] ACPI: Reserving WAET table memory at [mem 0x3fb78000-0x3fb78027]
[    0.026240] ACPI: Reserving BGRT table memory at [mem 0x3fb77000-0x3fb77037]
[    0.026394] Zone ranges:
[    0.026400]   DMA      [mem 0x0000000000001000-0x0000000000ffffff]
[    0.026405]   DMA32    [mem 0x0000000001000000-0x000000003ff1ffff]
[    0.026410]   Normal   empty
[    0.026412] Movable zone start for each node
[    0.026413] Early memory node ranges
[    0.026414]   node   0: [mem 0x0000000000001000-0x000000000009ffff]
[    0.026416]   node   0: [mem 0x0000000000100000-0x00000000007fffff]
[    0.026418]   node   0: [mem 0x0000000000808000-0x000000000080afff]
[    0.026420]   node   0: [mem 0x000000000080c000-0x000000000080ffff]
[... 465 lines cut ...]
[    9.425183] usbcore: registered new interface driver ums-freecom
[    9.432351] usbcore: registered new interface driver ums-isd200
[    9.438981] usbcore: registered new interface driver ums-jumpshot
[    9.447592] usbcore: registered new interface driver ums-karma
[... cut ...]
[   10.163376] batman_adv: B.A.T.M.A.N. advanced 2022.3-openwrt-1 (compatibility version 15) loaded
[   10.192062] kmodloader: done loading kernel modules from /etc/modules.d/*
[   13.349988] eth0: 0xffffc90000085000, 00:e0:4c:16:49:65, IRQ 44
[   17.545724] r8168: eth0: link up
[   17.548567] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[  120.842141] NFSD: Using /var/lib/nfs/v4recovery as the NFSv4 state recovery directory
[  120.846412] NFSD: Using legacy client tracking operations.
[  120.848416] NFSD: starting 10-second grace period (net f0000000)

liwenjie119, 2023-06-02:
未知原因仍有问题

Fri Jun  2 23:09:49 2023 kern.info kernel: [    0.000008] tsc: Detected 2494.332 MHz processor
Fri Jun  2 23:09:49 2023 kern.debug kernel: [    0.000112] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
Fri Jun  2 23:09:49 2023 kern.debug kernel: [    0.000115] e820: remove [mem 0x000a0000-0x000fffff] usable
Fri Jun  2 23:09:49 2023 kern.info kernel: [    0.000123] last_pfn = 0x3ff20 max_arch_pfn = 0x400000000
Fri Jun  2 23:09:49 2023 kern.info kernel: [    0.000749] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WP  UC- WT
Fri Jun  2 23:09:49 2023 kern.info kernel: [    0.025580] Secure boot disabled
Fri Jun  2 23:09:49 2023 kern.info kernel: [    0.025621] ACPI: Early table checksum verification disabled
Fri Jun  2 23:09:49 2023 kern.info kernel: [    0.025648] ACPI: RSDP 0x000000003FB7E014 000024 (v02 BOCHS )
Fri Jun  2 23:09:49 2023 kern.info kernel: [    0.025660] ACPI: XSDT 0x000000003FB7D0E8 000044 (v01 BOCHS  BXPC     00000001      01000013)
Fri Jun  2 23:09:49 2023 kern.info kernel: [    0.025682] ACPI: FACP 0x000000003FB7A000 000074 (v01 BOCHS  BXPC     00000001 BXPC 00000001)
Fri Jun  2 23:09:49 2023 kern.info kernel: [    0.025695] ACPI: DSDT 0x000000003FB7B000 0018E3 (v01 BOCHS  BXPC     00000001 BXPC 00000001)
Fri Jun  2 23:09:49 2023 kern.info kernel: [    0.025707] ACPI: FACS 0x000000003FBDD000 000040
Fri Jun  2 23:09:49 2023 kern.info kernel: [    0.025732] ACPI: APIC 0x000000003FB79000 0000B0 (v01 BOCHS  BXPC     00000001 BXPC 00000001)
Fri Jun  2 23:09:49 2023 kern.info kernel: [    0.025745] ACPI: WAET 0x000000003FB78000 000028 (v01 BOCHS  BXPC     00000001 BXPC 00000001)
Fri Jun  2 23:09:49 2023 kern.info kernel: [    0.025757] ACPI: BGRT 0x000000003FB77000 000038 (v01 INTEL  EDK2     00000002      01000013)
Fri Jun  2 23:09:49 2023 kern.info kernel: [    0.025761] ACPI: Reserving FACP table memory at [mem 0x3fb7a000-0x3fb7a073]
Fri Jun  2 23:09:49 2023 kern.info kernel: [    0.025766] ACPI: Reserving DSDT table memory at [mem 0x3fb7b000-0x3fb7c8e2]
Fri Jun  2 23:09:49 2023 kern.info kernel: [    0.025767] ACPI: Reserving FACS table memory at [mem 0x3fbdd000-0x3fbdd03f]
Fri Jun  2 23:09:49 2023 kern.info kernel: [    0.025768] ACPI: Reserving APIC table memory at [mem 0x3fb79000-0x3fb790af]
Fri Jun  2 23:09:49 2023 kern.info kernel: [    0.025769] ACPI: Reserving WAET table memory at [mem 0x3fb78000-0x3fb78027]
Fri Jun  2 23:09:49 2023 kern.info kernel: [    0.025770] ACPI: Reserving BGRT table memory at [mem 0x3fb77000-0x3fb77037]
Fri Jun  2 23:09:49 2023 kern.info kernel: [    0.025931] Zone ranges:
Fri Jun  2 23:09:49 2023 kern.info kernel: [    0.025932]   DMA      [mem 0x0000000000001000-0x0000000000ffffff]
Fri Jun  2 23:09:49 2023 kern.info kernel: [    0.025935]   DMA32    [mem 0x0000000001000000-0x000000003ff1ffff]
Fri Jun  2 23:09:49 2023 kern.info kernel: [    0.025936]   Normal   empty
Fri Jun  2 23:09:49 2023 kern.info kernel: [    0.025937] Movable zone start for each node
Fri Jun  2 23:09:49 2023 kern.info kernel: [    0.025938] Early memory node ranges
Fri Jun  2 23:09:49 2023 kern.info kernel: [    0.025938]   node   0: [mem 0x0000000000001000-0x000000000009ffff]
Fri Jun  2 23:09:49 2023 kern.info kernel: [    0.025939]   node   0: [mem 0x0000000000100000-0x00000000007fffff]
Fri Jun  2 23:09:49 2023 kern.info kernel: [    0.025940]   node   0: [mem 0x0000000000808000-0x000000000080afff]
Fri Jun  2 23:09:49 2023 kern.info kernel: [    0.025941]   node   0: [mem 0x000000000080c000-0x000000000080ffff]
Fri Jun  2 23:09:49 2023 kern.info kernel: [    0.025941]   node   0: [mem 0x0000000000900000-0x000000003f8eefff]
Fri Jun  2 23:09:49 2023 kern.info kernel: [    0.025942]   node   0: [mem 0x000000003fbff000-0x000000003ff1ffff]
Fri Jun  2 23:09:49 2023 kern.info kernel: [    0.025946] Initmem setup node 0 [mem 0x0000000000001000-0x000000003ff1ffff]
Fri Jun  2 23:09:49 2023 kern.info kernel: [    0.025968] On node 0, zone DMA: 1 pages in unavailable ranges
Fri Jun  2 23:09:49 2023 kern.info kernel: [    0.026036] On node 0, zone DMA: 96 pages in unavailable ranges
Fri Jun  2 23:09:49 2023 kern.info kernel: [    0.026040] On node 0, zone DMA: 8 pages in unavailable ranges
Fri Jun  2 23:09:49 2023 kern.info kernel: [    0.026041] On node 0, zone DMA: 1 pages in unavailable ranges
Fri Jun  2 23:09:49 2023 kern.info kernel: [    0.026146] On node 0, zone DMA: 240 pages in unavailable ranges
Fri Jun  2 23:09:49 2023 kern.info kernel: [    0.032834] On node 0, zone DMA32: 784 pages in unavailable ranges
Fri Jun  2 23:09:49 2023 kern.info kernel: [    0.032847] On node 0, zone DMA32: 224 pages in unavailable ranges
Fri Jun  2 23:09:49 2023 kern.info kernel: [    0.033574] ACPI: PM-Timer IO Port: 0xb008
Fri Jun  2 23:09:49 2023 kern.info kernel: [    0.033600] ACPI: LAPIC_NMI (acpi_id[0xff] dfl dfl lint[0x1])
Fri Jun  2 23:09:49 2023 kern.info kernel: [    0.033660] IOAPIC[0]: apic_id 0, version 17, address 0xfec00000, GSI 0-23
Fri Jun  2 23:09:49 2023 kern.info kernel: [    0.033669] ACPI: INT_SRC_OVR (bus 0 bus_irq 0 global_irq 2 dfl dfl)
Fri Jun  2 23:09:49 2023 kern.info kernel: [    0.033671] ACPI: INT_SRC_OVR (bus 0 bus_irq 5 global_irq 5 high level)
Fri Jun  2 23:09:49 2023 kern.info kernel: [    0.033672] ACPI: INT_SRC_OVR (bus 0 bus_irq 9 global_irq 9 high level)
Fri Jun  2 23:09:49 2023 kern.info kernel: [    0.033678] ACPI: INT_SRC_OVR (bus 0 bus_irq 10 global_irq 10 high level)
[... cut ...]
Fri Jun  2 23:10:48 2023 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Fri Jun  2 23:10:58 2023 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Fri Jun  2 23:11:07 2023 daemon.info uwsgi-luci: sh: out of range
Fri Jun  2 23:11:07 2023 daemon.info uwsgi-luci: sh: out of range
Fri Jun  2 23:11:08 2023 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Fri Jun  2 23:11:12 2023 daemon.info uwsgi-luci: sh: out of range

commit d2460dfb (2022-12-22): treewide: sync with upstream (#10658)

* rules: fix broken commitcount on alpine system

To generate commitcount we use grep --max-count. This is not present on
alpine grep and cause wrong generation. Use -m as it's just the short
version of --max-count and more portable.

Fixes: #11200

* image-commands.mk: Be consistent in command invocation

Most/all other tools use the staging dir prefix, gzip should as well.

Acked-by: Christian Marangi <ansuelsmth@gmail.com>

* image: fix device profile specific COMPILE targets

Commit a01d23e75 ("image: always rebuild kernel loaders")
is a step in the r [... cut]
closed 2026-09-05 by coolsnowwolf