coolsnowwolf/lede #11201: wpad-basic编译失败

boyhaibo, 2023-05-12:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：虚拟机全新编译wpad-basic失败

### (2) 路由器型号和固件版本

A：arm64通用固件

### (3) 详细日志

A：
```
aarch64-openwrt-linux-musl-gcc -c -o /home/summer/lede/build_dir/target-aarch64_generic_musl/hostapd-wpad-basic/hostapd-2022-07-29-b704dc72/build/wpa_supplicant/events.o -I/home/summer/lede/staging_dir/target-aarch64_generic_musl/usr/include/libnl-tiny -I/home/summer/lede/build_dir/target-aarch64_generic_musl/hostapd-wpad-basic/hostapd-2022-07-29-b704dc72/src/crypto -I/home/summer/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/usr/include -I/home/summer/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/include/fortify -I/home/summer/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/include -DCONFIG_LIBNL20 -D_GNU_SOURCE -DCONFIG_MSG_MIN_PRIORITY=3 -Os -pipe -mcpu=generic -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/summer/lede/build_dir/target-aarch64_generic_musl/hostapd-wpad-basic/hostapd-2022-07-29-b704dc72=hostapd-2022-07-29-b704dc72 -Wformat -Werror=format-security -DPIC -fPIC -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/home/summer/lede/build_dir/target-aarch64_generic_musl/hostapd-wpad-basic/hostapd-2022-07-29-b704dc72/src -I/home/summer/lede/build_dir/target-aarch64_generic_musl/hostapd-wpad-basic/hostapd-2022-07-29-b704dc72/src/utils -DHOSTAPD -DCONFIG_ELOOP_EPOLL -DUBUS_SUPPORT -DCONFIG_NO_RADIUS -DCONFIG_NO_ACCOUNTING -DCONFIG_CTRL_IFACE_UNIX -DCONFIG_CTRL_IFACE -DCONFIG_RSN_PREAUTH -DCONFIG_OCV -DCONFIG_IEEE80211R -DCONFIG_IEEE80211R_AP -DCONFIG_ETH_P_OUI -DCONFIG_AIRTIME_POLICY -DCONFIG_IEEE80211AC -DCONFIG_IEEE80211AX -DCONFIG_MBO -DCONFIG_DRIVER_WIRED -DCONFIG_DRIVER_NL80211 -DCONFIG_DRIVER_WIRED -DCONFIG_DRIVER_NL80211 -DEAP_SERVER_IDENTITY -DCONFIG_TLS_INTERNAL -DCONFIG_CRYPTO_INTERNAL -DCONFIG_SHA256 -DCONFIG_HMAC_SHA256_KDF -DCONFIG_INTERNAL_SHA384 -DCONFIG_INTERNAL_SHA512 -DCONFIG_NO_RANDOM_POOL -DNEED_AP_MLME -DCONFIG_ACS -DCONFIG_DEBUG_SYSLOG -I/home/summer/lede/staging_dir/target-aarch64_generic_musl/usr/include/libnl-tiny -I/home/summer/lede/build_dir/target-aarch64_generic_musl/hostapd-wpad-basic/hostapd-2022-07-29-b704dc72/src/crypto -I/home/summer/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/usr/include -I/home/summer/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/include/fortify -I/home/summer/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/include -DCONFIG_LIBNL20 -D_GNU_SOURCE -DCONFIG_MSG_MIN_PRIORITY=3 -I/home/summer/lede/build_dir/target-aarch64_generic_musl/hostapd-wpad-basic/hostapd-2022-07-29-b704dc72/src -I/home/summer/lede/build_dir/target-aarch64_generic_musl/hostapd-wpad-basic/hostapd-2022-07-29-b704dc72/src/utils -DCONFIG_ELOOP_EPOLL -DUBUS_SUPPORT -DCONFIG_HT_OVERRIDES -DCONFIG_VHT_OVERRIDES -DCONFIG_BACKEND_FILE -DCONFIG_NO_CONFIG_WRITE -DCONFIG_OCV -DCONFIG_IEEE80211R -DCONFIG_WNM -DCONFIG_DRIVER_WIRED -DCONFIG_DRIVER_NL80211 -DCONFIG_MBO -DCONFIG_ACS -DCONFIG_TLS_INTERNAL -DCONFIG_CRYPTO_INTERNAL -DCONFIG_SHA256 -DCONFIG_INTERNAL_SHA384 -DCONFIG_INTERNAL_SHA512 -DCONFIG_NO_RANDOM_POOL -DCONFIG_CTRL_IFACE -DCONFIG_CTRL_IFACE_UNIX -DCONFIG_NO_LINUX_PACKET_SOCKET_WAR -DCONFIG_SME -DCONFIG_DEBUG_SYSLOG -DLOG_HOSTAPD=LOG_DAEMON events.c
aarch64-openwrt-linux-musl-gcc -c -o /home/summer/lede/build_dir/target-aarch64_generic_musl/hostapd-wpad-basic/hostapd-2022-07-29-b704dc72/build/wpa_supplicant/bssid_ignore.o -I/home/summer/lede/staging_dir/target-aarch64_generic_musl/usr/include/libnl-tiny -I/home/summer/lede/build_dir/target-aarch64_generic_musl/hostapd-wpad-basic/hostapd-2022-07-29-b704dc72/src/crypto -I/home/summer/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/usr/include -I/home/summer/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/include/fortify -I/home/summer/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/include -DCONFIG_LIBNL20 -D_GNU_SOURCE -DCONFIG_MSG_MIN_PRIORITY=3 -Os -pipe -mcpu=generic -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/summer/lede/build_dir/target-aarch64_generic_musl/hostapd-wpad-basic/hostapd-2022-07-29-b704dc72=hostapd-2022-07-29-b704dc72 -Wformat -Werror=format-security -DPIC -fPIC -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/home/summer/lede/build_dir/target-aarch64_generic_musl/hostapd-wpad-basic/hostapd-2022-07-29-b704dc72/src -I/home/summer/lede/build_dir/target-aarch64_generic_musl/hostapd-wpad-basic/hostapd-2022-07-29-b704dc72/src/utils -DHOSTAPD -DCONFIG_ELOOP_EPOLL -DUBUS_SUPPORT -DCONFIG_NO_RADIUS -DCONFIG_NO_ACCOUNTING -DCONFIG_CTRL_IFACE_UNIX -DCONFIG_CTRL_IFACE -DCONFIG_RSN_PREAUTH -DCONFIG_OCV -DCONFIG_IEEE80211R -DCONFIG_IEEE80211R_AP -DCONFIG_ETH_P_OUI -DCONFIG_AIRTIME_POLICY -DCONFIG_IEEE80211AC -DCONFIG_IEEE80211AX -DCONFIG_MBO -DCONFIG_DRIVER_WIRED -DCONFIG_DRIVER_NL80211 -DCONFIG_DRIVER_WIRED -DCONFIG_DRIVER_NL80211 -DEAP_SERVER_IDENTITY -DCONFIG_TLS_INTERNAL -DCONFIG_CRYPTO_INTERNAL -DCONFIG_SHA256 -DCONFIG_HMAC_SHA256_KDF -DCONFIG_INTERNAL_SHA384 -DCONFIG_INTERNAL_SHA512 -DCONFIG_NO_RANDOM_POOL -DNEED_AP_MLME -DCONFIG_ACS -DCONFIG_DEBUG_SYSLOG -I/home/summer/lede/staging_dir/target-aarch64_generic_musl/usr/include/libnl-tiny -I/home/summer/lede/build_dir/target-aarch64_generic_musl/hostapd-wpad-basic/hostapd-2022-07-29-b704dc72/src/crypto -I/home/summer/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/usr/include -I/home/summer/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/include/fortify -I/home/summer/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/include -DCONFIG_LIBNL20 -D_GNU_SOURCE -DCONFIG_MSG_MIN_PRIORITY=3 -I/home/summer/lede/build_dir/target-aarch64_generic_musl/hostapd-wpad-basic/hostapd-2022-07-29-b704dc72/src -I/home/summer/lede/build_dir/target-aarch64_generic_musl/hostapd-wpad-basic/hostapd-2022-07-29-b704dc72/src/utils -DCONFIG_ELOOP_EPOLL -DUBUS_SUPPORT -DCONFIG_HT_OVERRIDES -DCONFIG_VHT_OVERRIDES -DCONFIG_BACKEND_FILE -DCONFIG_NO_CONFIG_WRITE -DCONFIG_OCV -DCONFIG_IEEE80211R -DCONFIG_WNM -DCONFIG_DRIVER_WIRED -DCONFIG_DRIVER_NL80211 -DCONFIG_MBO -DCONFIG_ACS -DCONFIG_TLS_INTERNAL -DCONFIG_CRYPTO_INTERNAL -DCONFIG_SHA256 -DCONFIG_INTERNAL_SHA384 -DCONFIG_INTERNAL_SHA512 -DCONFIG_NO_RANDOM_POOL -DCONFIG_CTRL_IFACE -DCONFIG_CTRL_IFACE_UNIX -DCONFIG_NO_LINUX_PACKET_SOCKET_WAR -DCONFIG_SME -DCONFIG_DEBUG_SYSLOG -DLOG_HOSTAPD=LOG_DAEMON bssid_ignore.c
aarch64-openwrt-linux-musl-gcc -c -o /home/summer/lede/build_dir/target-aarch64_generic_musl/hostapd-wpad-basic/hostapd-2022-07-29-b704dc72/build/wpa_supplicant/wpas_glue.o -I/home/summer/lede/staging_dir/target-aarch64_generic_musl/usr/include/libnl-tiny -I/home/summer/lede/build_dir/target-aarch64_generic_musl/hostapd-wpad-basic/hostapd-2022-07-29-b704dc72/src/crypto -I/home/summer/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/usr/include -I/home/summer/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/include/fortify -I/home/summer/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/include -DCONFIG_LIBNL20 -D_GNU_SOURCE -DCONFIG_MSG_MIN_PRIORITY=3 -Os -pipe -mcpu=generic -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/summer/lede/build_dir/target-aarch64_generic_musl/hostapd-wpad-basic/hostapd-2022-07-29-b704dc72=hostapd-2022-07-29-b704dc72 -Wformat -Werror=format-security -DPIC -fPIC -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/home/summer/lede/build_dir/target-aarch64_generic_musl/hostapd-wpad-basic/hostapd-2022-07-29-b704dc72/src -I/home/summer/lede/build_dir/target-aarch64_generic_musl/hostapd-wpad-basic/hostapd-2022-07-29-b704dc72/src/utils -DHOSTAPD -DCONFIG_ELOOP_EPOLL -DUBUS_SUPPORT -DCONFIG_NO_RADIUS -DCONFIG_NO_ACCOUNTING -DCONFIG_CTRL_IFACE_UNIX -DCONFIG_CTRL_IFACE -DCONFIG_RSN_PREAUTH -DCONFIG_OCV -DCONFIG_IEEE80211R -DCONFIG_IEEE80211R_AP -DCONFIG_ETH_P_OUI -DCONFIG_AIRTIME_POLICY -DCONFIG_IEEE80211AC -DCONFIG_IEEE80211AX -DCONFIG_MBO -DCONFIG_DRIVER_WIRED -DCONFIG_DRIVER_NL80211 -DCONFIG_DRIVER_WIRED -DCONFIG_DRIVER_NL80211 -DEAP_SERVER_IDENTITY -DCONFIG_TLS_INTERNAL -DCONFIG_CRYPTO_INTERNAL -DCONFIG_SHA256 -DCONFIG_HMAC_SHA256_KDF -DCONFIG_INTERNAL_SHA384 -DCONFIG_INTERNAL_SHA512 -DCONFIG_NO_RANDOM_POOL -DNEED_AP_MLME -DCONFIG_ACS -DCONFIG_DEBUG_SYSLOG -I/home/summer/lede/staging_dir/target-aarch64_generic_musl/usr/include/libnl-tiny -I/home/summer/lede/build_dir/target-aarch64_generic_musl/hostapd-wpad-ba [... cut ...]
time: package/network/services/hostapd/wpad-basic/compile#56.21#7.38#62.21
    ERROR: package/network/services/hostapd failed to build (build variant: wpad-basic).
make[2]: *** [package/Makefile:116: package/network/services/hostapd/compile] Error 1
make[2]: Leaving directory '/home/summer/lede'
make[1]: *** [package/Makefile:110: /home/summer/lede/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/summer/lede'
make: *** [/home/summer/lede/include/toplevel.mk:231：world] 错误 2
```

WYC-2020, 2023-05-15:
我这边修复了一下那个hostapd-common 依赖问题，编译没问题，你试试
[image]

WYC-2020, 2023-05-15:
> 反馈bug/问题模板，提建议请删除
>
> ## 1.关于你要提交的问题
> Q：是否搜索了issue (使用 "x" 选择)
>
> * [ x] 没有类似的issue
>
> ## 2. 详细叙述
> ### (1) 具体问题
> A：虚拟机全新编译wpad-basic失败
>
> ### (2) 路由器型号和固件版本
> A：arm64通用固件
>
> ### (3) 详细日志
> A：
>
> ``` [...]

当然你不这样修 去手工选择也可以，不过我习惯 依赖什么就依赖上，不喜欢去手工选

boyhaibo, 2023-05-17:
> 我这边修复了一下那个hostapd-common 依赖问题，编译没问题，你试试 [image]

你好，我看了一下我的config文件cfg80211是勾选的
CONFIG_PACKAGE_kmod-cfg80211=y

boyhaibo, 2023-05-17:
> 我这边修复了一下那个hostapd-common 依赖问题，编译没问题，你试试 [image]

hostapd-common这个文件路径是什么，我并没有搜到这个文件

boyhaibo, 2023-05-17:
> 我这边修复了一下那个hostapd-common 依赖问题，编译没问题，你试试 [image]

这是我的配置文件
```
CONFIG_TARGET_armvirt=y
CONFIG_TARGET_armvirt_64=y
CONFIG_TARGET_armvirt_64_DEVICE_generic=y
# CONFIG_TARGET_IMAGES_GZIP is not set
# CONFIG_TARGET_ROOTFS_CPIOGZ is not set
# CONFIG_TARGET_ROOTFS_EXT4FS is not set
# CONFIG_TARGET_ROOTFS_INITRAMFS is not set
# CONFIG_TARGET_ROOTFS_SQUASHFS is not set
CONFIG_PACKAGE_perl=y
CONFIG_PACKAGE_perl-http-date=y
CONFIG_PACKAGE_perlbase-file=y
CONFIG_PACKAGE_perlbase-getopt=y
CONFIG_PACKAGE_perlbase-time=y
CONFIG_PACKAGE_perlbase-unicode=y
CONFIG_PACKAGE_perlbase-utf8=y
CONFIG_PACKAGE_blkid=y
CONFIG_PACKAGE_fdisk=y
CONFIG_PACKAGE_lsblk=y
CONFIG_PACKAGE_parted=y
CONFIG_PACKAGE_attr=y
CONFIG_PACKAGE_btrfs-progs=y
CONFIG_BTRFS_PROGS_ZSTD=y
CONFIG_PACKAGE_chattr=y
CONFIG_PACKAGE_dosfstools=y
CONFIG_PACKAGE_e2fsprogs=y
CONFIG_PACKAGE_f2fs-tools=y
CONFIG_PACKAGE_f2fsck=y
CONFIG_PACKAGE_lsattr=y
CONFIG_PACKAGE_mkf2fs=y
CONFIG_PACKAGE_xfs-fsck=y
CONFIG_PACKAGE_xfs-mkfs=y
CONFIG_PACKAGE_bsdtar=y
CONFIG_PACKAGE_pigz=y
CONFIG_PACKAGE_bash=y
CONFIG_PACKAGE_gawk=y
CONFIG_PACKAGE_getopt=y
CONFIG_PACKAGE_losetup=y
CONFIG_PACKAGE_tar=y
CONFIG_PACKAGE_uuidgen=y
CONFIG_PACKAGE_acpid=y
CONFIG_PACKAGE_kmod-brcmfmac=y
CONFIG_BRCMFMAC_SDIO=y
CONFIG_PACKAGE_kmod-brcmutil=y
CONFIG_PACKAGE_kmod-cfg80211=y
CONFIG_PACKAGE_kmod-mac80211=y
CONFIG_PACKAGE_hostapd-common=y
CONFIG_PACKAGE_wpa-cli=y
CONFIG_PACKAGE_wpad-basic=y
CONFIG_PACKAGE_iw=y
CONFIG_PACKAGE_ntfs3-mount=y
# CONFIG_PACKAGE_luci-app-accesscontrol is not set
CONFIG_PACKAGE_luci-app-adguardhome=y
CONFIG_PACKAGE_luci-app-alist=y
CONFIG_PACKAGE_luci-app-argon-config=y
CONFIG_PACKAGE_luci-app-ddns=y
CONFIG_PACKAGE_luci-app-diskman=y
CONFIG_PACKAGE_luci-app-dockerman=y
CONFIG_PACKAGE_luci-app-mosdns=y
CONFIG_PACKAGE_luci-app-nlbwmon=y
CONFIG_PACKAGE_luci-app-openclash=y
CONFIG_PACKAGE_luci-app-passwall=y
CONFIG_PACKAGE_luci-app-passwall_Iptables_Transparent_Proxy=y
CONFIG_PACKAGE_luci-app-passwall_INCLUDE_Brook=y
CONFIG_PACKAGE_luci-app-passwall_INCLUDE_Hysteria=y
CONFIG_PACKAGE_luci-app-passwall_INCLUDE_NaiveProxy=y
CONFIG_PACKAGE_luci-app-passwall_INCLUDE_Shadowsocks_Rust_Client=y
CONFIG_PACKAGE_luci-app-passwall_INCLUDE_ShadowsocksR_Libev_Server=y
CONFIG_PACKAGE_luci-app-passwall_INCLUDE_Trojan_GO=y
# CONFIG_PACKAGE_luci-app-passwall_INCLUDE_V2ray_Plugin is not set
CONFIG_PACKAGE_luci-app-samba4=y
CONFIG_PACKAGE_luci-app-ssr-plus=y
CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_ChinaDNS_NG=y
CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_Hysteria=y
CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_IPT2Socks=y
CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_Kcptun=y
CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_NaiveProxy=y
CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_Redsocks2=y
CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_Shadowsocks_Rust_Client=y
# CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_Shadowsocks_Libev_Server is not set
CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_Shadowsocks_Rust_Server=y
CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_ShadowsocksR_Libev_Server=y
CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_Trojan=y
CONFIG_PACKAGE_luci-app-ttyd=y
CONFIG_PACKAGE_luci-app-turboacc=y
CONFIG_PACKAGE_luci-app-turboacc_INCLUDE_OFFLOADING=y
# CONFIG_PACKAGE_luci-app-turboacc_INCLUDE_DNSFORWARDER is not set
# CONFIG_PACKAGE_luci-app-turboacc_INCLUDE_DNSPROXY is not set
CONFIG_PACKAGE_luci-app-unblockmusic=y
CONFIG_PACKAGE_luci-app-upnp=y
# CONFIG_PACKAGE_luci-app-vlmcsd is not set
# CONFIG_PACKAGE_luci-app-vsftpd is not set
# CONFIG_PACKAGE_luci-app-wol is not set
CONFIG_PACKAGE_luci-app-zerotier=y
CONFIG_PACKAGE_luci-theme-argon=y
CONFIG_PACKAGE_luci-theme-design=y
CONFIG_PACKAGE_ddns-scripts_cloudflare.com-v4=y
CONFIG_PACKAGE_docker-compose=y
CONFIG_PACKAGE_htop=y
CONFIG_PACKAGE_iperf3=y
CONFIG_PACKAGE_nano-plus=y
CONFIG_PACKAGE_openssh-sftp-server=y
CONFIG_PACKAGE_vim-fuller=y
CONFIG_PACKAGE_zsh=y
CONFIG_PACKAGE_kmod-shortcut-fe=y
CONFIG_PACKAGE_kmod-shortcut-fe-cm=y
CONFIG_PACKAGE_ipv6helper=y
CONFIG_PACKAGE_ip6tables=y
CONFIG_PACKAGE_ip6tables-extra=y
CONFIG_PACKAGE_ip6tables-mod-nat=y
```

wjz304, 2023-05-26:
> > 我这边修复了一下那个hostapd-common 依赖问题，编译没问题，你试试 [image]
> 
> hostapd-common这个路径是什么，我并没有搜到这个文件

package/network/services/hostapd/Makefile   但是我改了不好使

WYC-2020, 2023-05-26:
> > > 我这边修复了一下那个hostapd-common 依赖问题，编译没问题，你试试 [image]
> > 
> > 
> > hostapd-common这个路径是什么，我并没有搜到这个文件
> 
> package/network/services/hostapd/Makefile 但是我改了不好使

删除tmp 重新install 然后make menuconfig  在看下cfg80211 是否选中

wjz304, 2023-05-26:
暂时改了下 TARGET_CFLAGS 和 TARGET_LDFLAGS 解了
```
sed -i '/TARGET_LDFLAGS += -lubox -lubus/i\TARGET_CFLAGS += -ffunction-sections -fdata-sections -flto' openwrt/package/network/services/hostapd/Makefile
sed -i '/TARGET_LDFLAGS += -lubox -lubus/i\TARGET_LDFLAGS += -Wl,--gc-sections -flto=jobserver -fuse-linker-plugin' openwrt/package/network/services/hostapd/Makefile
```

boyhaibo, 2023-05-27:
> 暂时改了下 TARGET_CFLAGS 和 TARGET_LDFLAGS 解了
> 
> ```
> sed -i '/TARGET_LDFLAGS += -lubox -lubus/i\TARGET_CFLAGS += -ffunction-sections -fdata-sections -flto' package/network/services/hostapd/Makefile
> sed -i '/TARGET_LDFLAGS += -lubox -lubus/i\TARGET_LDFLAGS += -Wl,--gc-sections -flto=jobserver -fuse-linker-plugin' package/network/services/hostapd/Makefile
> ```

非常感谢，编译通过没有报错了

closed 2023-05-27 by boyhaibo

WYC-2020, 2023-05-28:
刚试了一下，我这边不打算这样修，我是这样修的，这个是链接错误，你通过增加链接属性 优化点部分代码通过，不知道功能咋样 
[image]

wjz304, 2023-05-28:
我并不是很了解lede的源码，只是抽时间看了一眼 临时解下，以你们开发人员为准，改好了，我就把我加的两条补丁删了

wjz304, 2023-05-28:
另外， openwrt 原版也有这个问题的，

mahakalaya, 2023-06-02:
> 非常感谢，用的方法编译可以通过了

wjz304, 2023-06-17:
https://github.com/coolsnowwolf/lede/issues/11193

asterwyx, 2024-05-25:
请问这个patch合上去了吗？现在我在编译wpad-basic-wolfssl的时候还有类似的链接错误。