coolsnowwolf/lede #11202: Transmission web 界面404

eearphon, 2023-05-12:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：
勾选transmission开启开关，其他都默认，用户名 root，用户组 transmission，访问9091端口web界面，显示
`404: Not Found
Couldn't find Transmission's web interface files!

Users: to tell Transmission where to look, set the TRANSMISSION_WEB_HOME environment variable to the folder where the web interface's index.html is located.

Package Builders: to set a custom default at compile time, #define PACKAGE_DATA_DIR in libtransmission/platform.c or tweak tr_getClutchDir () by hand.`

### (2) 路由器型号和固件版本

A：x86-64 r5959-52223ba9d，只勾选“luci-app-transmission”，其他默认编译

### (3) 详细日志

A：
`root@OpenWrt:~# logread | tail -n 30
Fri May 12 19:11:45 2023 authpriv.notice dropbear[2634]: Password auth succeeded for 'root' from 192.168.1.4:57175
Fri May 12 19:11:49 2023 user.notice firewall: Reloading firewall due to ifup of lan (br-lan)
Fri May 12 19:11:49 2023 user.notice nlbwmon: Reloading nlbwmon due to ifup of lan (br-lan)
Fri May 12 19:12:01 2023 kern.err kernel: [   41.351731] squashfs: Unknown parameter 'discard'
Fri May 12 19:12:01 2023 kern.info kernel: [   41.355369] F2FS-fs (sda128): Magic Mismatch, valid(0xf2f52010) - read(0x89e45589)
Fri May 12 19:12:01 2023 kern.err kernel: [   41.356226] F2FS-fs (sda128): Can't find valid F2FS filesystem in 1th superblock
Fri May 12 19:12:01 2023 kern.info kernel: [   41.357308] F2FS-fs (sda128): Magic Mismatch, valid(0xf2f52010) - read(0x2e48937b)
Fri May 12 19:12:01 2023 kern.err kernel: [   41.358130] F2FS-fs (sda128): Can't find valid F2FS filesystem in 2th superblock
Fri May 12 19:12:01 2023 kern.warn kernel: [   41.359128] exFAT-fs (sda128): mounting with "discard" option, but the device does not support discard
Fri May 12 19:12:01 2023 kern.err kernel: [   41.360326] exFAT-fs (sda128): invalid boot record signature
Fri May 12 19:12:01 2023 kern.err kernel: [   41.361117] exFAT-fs (sda128): failed to read boot sector
Fri May 12 19:12:01 2023 kern.err kernel: [   41.361898] exFAT-fs (sda128): failed to recognize exfat type
Fri May 12 19:12:05 2023 user.notice nlbwmon: Reloading nlbwmon due to ifup of loopback (lo)
Fri May 12 19:12:20 2023 user.notice firewall: Reloading firewall due to ifup of wan6 (eth1)
Fri May 12 19:12:20 2023 user.notice nlbwmon: Reloading nlbwmon due to ifup of wan6 (eth1)
Fri May 12 19:12:31 2023 kern.err kernel: [   71.432786] squashfs: Unknown parameter 'discard'
Fri May 12 19:12:31 2023 kern.err kernel: [   71.434172] /dev/sda1: Can't open blockdev
Fri May 12 19:12:35 2023 user.notice firewall: Reloading firewall due to ifup of wan (eth1)
Fri May 12 19:12:35 2023 user.notice nlbwmon: Reloading nlbwmon due to ifup of wan (eth1)
Fri May 12 19:12:49 2023 user.notice transmission: Starting with 505430000 virt mem
Fri May 12 19:12:50 2023 daemon.err transmission-daemon[6689]: [2023-05-12 11:12:50.901] Couldn't save file "/opt/transmission/settings.json": Resource busy (variant.c:1271)
Fri May 12 19:13:01 2023 daemon.info samba4-server: io_uring module found, enabling VFS io_uring. (also needs Kernel 5.4+ Support)
Fri May 12 19:13:01 2023 daemon.err nmbd[3676]: [2023/05/12 19:13:01.836396,  0] ../../source3/nmbd/nmbd.c:60(terminate)
Fri May 12 19:13:01 2023 daemon.err nmbd[3676]:   Got SIGTERM: going down...
Fri May 12 19:13:01 2023 daemon.err nmbd[6885]: [2023/05/12 19:13:01.908092,  0] ../../lib/util/become_daemon.c:135(daemon_ready)
Fri May 12 19:13:01 2023 daemon.err nmbd[6885]:   daemon_ready: daemon 'nmbd' finished starting up and ready to serve connections
Fri May 12 19:13:01 2023 daemon.err nmbd[6885]: [2023/05/12 19:13:01.911603,  0] ../../source3/nmbd/nmbd_namequery.c:109(query_name_response)
Fri May 12 19:13:01 2023 daemon.err nmbd[6885]:   query_name_response: Multiple (2) responses received for a query on subnet 192.168.1.254 for name WORKGROUP<1d>.
Fri May 12 19:13:01 2023 daemon.err nmbd[6885]:   This response was from IP 192.168.1.1, reporting an IP address of 192.168.1.1.
Fri May 12 19:13:01 2023 daemon.err smbd[6884]: [2023/05/12 19:13:01.948876,  0] ../../lib/util/become_daemon.c:135(daemon_ready)
Fri May 12 19:13:01 2023 daemon.err smbd[6884]:   daemon_ready: daemon 'smbd' finished starting up and ready to serve connections

root@OpenWrt:~# /etc/init.d/transmission restart
root@OpenWrt:~# logread | tail -n 10
Fri May 12 19:18:15 2023 daemon.err nmbd[6885]: [2023/05/12 19:18:15.142660,  0] ../../source3/nmbd/nmbd_namequery.c:109(query_name_response)
Fri May 12 19:18:15 2023 daemon.err nmbd[6885]:   query_name_response: Multiple (2) responses received for a query on subnet 192.168.1.254 for name WORKGROUP<1d>.
Fri May 12 19:18:15 2023 daemon.err nmbd[6885]:   This response was from IP 192.168.1.1, reporting an IP address of 192.168.1.1.
Fri May 12 19:23:17 2023 daemon.err nmbd[6885]: [2023/05/12 19:23:17.383146,  0] ../../source3/nmbd/nmbd_namequery.c:109(query_name_response)
Fri May 12 19:23:17 2023 daemon.err nmbd[6885]:   query_name_response: Multiple (2) responses received for a query on subnet 192.168.1.254 for name WORKGROUP<1d>.
Fri May 12 19:23:17 2023 daemon.err nmbd[6885]:   This response was from IP 192.168.1.1, reporting an IP address of 192.168.1.1.
Fri May 12 19:23:25 2023 authpriv.info dropbear[7949]: Exit (root) from <192.168.1.4:57409>: Exited normally
Fri May 12 19:23:57 2023 user.notice transmission: Starting with 505430000 virt mem
Fri May 12 19:23:58 2023 daemon.err transmission-daemon[6689]: [2023-05-12 11:23:58.095] Couldn't save file "/opt/transmission/settings.json": Resource busy (variant.c:1271)
Fri May 12 19:23:59 2023 daemon.err transmission-daemon[9014]: [2023-05-12 11:23:59.353] Couldn't save file "/opt/transmission/settings.json": Resource busy (variant.c:1271)
`

eearphon, 2023-05-12:
手工删除 /opt/transmission/settings.json ，然后重启 transmission ，这个文件还是能生成

root@OpenWrt:~# ll /opt /opt/transmission/
/opt:
drwxr-xr-x    3 root     root          3488 May 12 19:12 ./
drwxr-xr-x    1 root     root          3488 May 12 19:12 ../
drwxr-xr-x    2 root     transmis      3488 May 12 19:12 transmission/

/opt/transmission/:
drwxr-xr-x    2 root     transmis      3488 May 12 19:12 ./
drwxr-xr-x    3 root     root          3488 May 12 19:12 ../
-rw-r--r--    1 root     transmis      1406 May 12 19:23 settings.json

eearphon, 2023-05-12:
自己以前编译的当前在用的 r5281-6c4e5d7b4 则一切正常，这次想更新到最新版本，才发现这个问题

WYC-2020, 2023-05-15:
[image]
 很早以前修复的

Kisoul, 2023-05-23:
> [image] 很早以前修复的

136删掉了后面又加上了是为啥？另外L大的仓库里还不是你图中这样的。

WYC-2020, 2023-05-24:
> > [image] 很早以前修复的
> 
> 136删掉了后面又加上了是为啥？另外L大的仓库里还不是你图中这样的。

那不清楚，反正我这边用我这边的仓库是正常的 
[image]
[image]
[image]

以前我有遇到你这种情况就是这样修的

Kisoul, 2023-05-24:
> > > [image] 很早以前修复的
> > 
> > 
> > 136删掉了后面又加上了是为啥？另外L大的仓库里还不是你图中这样的。
> 
> 那不清楚，反正我这边用我这边的仓库是正常的 [image] [image] [image]
> 
> 以前我有遇到你这种情况就是这样修的

恩，看来你的仓库的代码不是L大的。你的仓库是哪fork的？可否借鉴学习一下，谢谢

WYC-2020, 2023-05-24:
> > > > [image] 很早以前修复的
> > > 
> > > 
> > > 136删掉了后面又加上了是为啥？另外L大的仓库里还不是你图中这样的。
> > 
> > 
> > 那不清楚，反正我这边用我这边的仓库是正常的 [image] [image] [image]
> > 以前我有遇到你这种情况就是这样修的
> 
> 恩，看来你的仓库的代码不是L大的。你的仓库是哪fork的？可否借鉴学习一下，谢谢

额我也是fork来的啊，我遇到这问题就是这样修复的呢，你看下你替换对了没呢，或者试下直接修改你刷好的op，/etc/init.d/下面

eearphon, 2023-05-25:
> [image] 很早以前修复的

谢谢，patch后WEB好了，但是下载任务存不住，重启就丢失，用root用户运行的

eearphon, 2023-05-25:
> > [image] 很早以前修复的
> 
> 136删掉了后面又加上了是为啥？

相当于 136 137 两行上下换了个位

WYC-2020, 2023-05-25:
> > [image] 很早以前修复的
> 
> 谢谢，patch后WEB好了，但是下载任务存不住，重启就丢失，用root用户运行的

自己排查吧，我这边没有这个问题，自己看所存储的位置 又没问题，我这边重启软路由 和重启服务 都正常

nuo12138, 2023-05-25:
我的也会丢失，还是等大雕修复吧

nuo12138, 2023-05-25:
> > > [image] 很早以前修复的
> > 
> > 
> > 谢谢，patch后WEB好了，但是下载任务存不住，重启就丢失，用root用户运行的
> 
> 自己排查吧，我这边没有这个问题，自己看所存储的位置 又没问题，我这边重启软路由 和重启服务 都正常

我的也会丢失，还是等大雕修复吧

WYC-2020, 2023-06-09:
> > > [image] 很早以前修复的
> > 
> > 
> > 谢谢，patch后WEB好了，但是下载任务存不住，重启就丢失，用root用户运行的
> 
> 这样改重启后不会丢失下载任务 #procd_add_jail transmission log procd_add_jail_mount "$web_home"

自己不看清我不是加了的么。

WYC-2020, 2023-06-09:
> > > > > [image] 很早以前修复的
> > > > 
> > > > 
> > > > 谢谢，patch后WEB好了，但是下载任务存不住，重启就丢失，用root用户运行的
> > > 
> > > 
> > > 这样改重启后不会丢失下载任务 #procd_add_jail transmission log procd_add_jail_mount "$web_home"
> > 
> > 
> > 自己不看清我不是加了的么。
> 
> 我试过你的代码，会丢失下载任务的，重启或者关闭transmission都会丢失，这个不会

嗦嘎是嘛，那你那样改吧，可能我们版本不一样，我有的库自己升级了的

eearphon, 2023-06-15:
r5959-52223ba9d，即 Oct 21, 2022 还正常的，但看了 Transmission 是 2021 年更新过的

nuo12138, 2023-06-18:
> r5959-52223ba9d，即 Oct 21, 2022 还正常的，但看了 Transmission 是 2021 年更新过的

对的，去年的源码编译是可以用的。恩山论坛有人说是openwrt官方把一些东西改了导致transmission出错，得等上游更新修复

closed 2026-09-05 by coolsnowwolf