coolsnowwolf/lede #11203: K3更新源码，编译后无线不能用了

jtzgc, 2023-05-12:
挺长时间没更新，刚拉取新源码，云编译也一切正常，刷入K3后发现无线开启不了了，之前编译一直都没问题，请教下各位大佬该怎么处理
[image]

zfdx123, 2023-05-12:
网卡型号是什么

coolsnowwolf, 2023-05-13:
切换到 5.10 内核

barton0426, 2023-05-14:
> 

SHELL= flock /home/runner/work/lede/lede/tmp/.root-copy.flock -c 'cp -fpR /home/runner/work/lede/lede/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/packages/.pkgdir/kernel/. /home/runner/work/lede/lede/staging_dir/target-arm_cortex-a9_musl_eabi/root-bcm53xx/'
touch /home/runner/work/lede/lede/staging_dir/target-arm_cortex-a9_musl_eabi/root-bcm53xx/stamp/.kernel_installed
ERROR: module '/home/runner/work/lede/lede/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/linux-5.10.176/net/netfilter/nf_flow_table_hw.ko' is missing.
make[3]: Leaving directory '/home/runner/work/lede/lede/package/kernel/linux'
make[3]: *** [modules/netfilter.mk:189: /home/runner/work/lede/lede/bin/targets/bcm53xx/generic/packages/kmod-nf-flow_5.10.176-1_arm_cortex-a9.ipk] Error 1
time: package/kernel/linux/compile#2.64#0.20#2.84
    ERROR: package/kernel/linux failed to build.
make[2]: *** [package/Makefile:116: package/kernel/linux/compile] Error 1
make[2]: Leaving directory '/home/runner/work/lede/lede'
make[1]: *** [package/Makefile:110: /home/runner/work/lede/lede/staging_dir/target-arm_cortex-a9_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/runner/work/lede/lede'
make: *** [/home/runner/work/lede/lede/include/toplevel.mk:231: world] Error 2
Error: Process completed with exit code 2.
切换新内核action报错了。。

fengqi, 2023-05-14:
RT-ACRH17 编译后无线也不能用了，完全从零编译也行，暂时退回到3月份的版本了。

nanqing0723, 2023-05-16:
请问k3现在用5.10的内核编译有wifi了吗？以后是不是都不用5.4的内核了呢？

hcym, 2023-05-16:
昨天编译的510，没刷，天气热了，冬天再说

barton0426, 2023-05-20:
> 切换到 5.10 内核

大雕，最新源码重新fork，编译的k3固件，wan的pppoe没有数据交换了，无法拨号，换回老固件正常排除了家里宽带问题

WongKwongYu, 2023-05-21:
> > 切换到 5.10 内核
> 
> 大雕，最新源码重新fork，编译的k3固件，wan的pppoe没有数据交换了，无法拨号，换回老固件正常排除了家里宽带问题

+1

sunyink, 2023-05-21:
R7800,+1,编译了无数次原来是这个原因.....

leyuanwl, 2023-07-06:
+1 树莓派3最新版还是有这个问题并且设置的ap 但是一重启就变成图上这样并且模式显示client不是master。总之不能重启一重启就有问题了

closed 2026-09-05 by coolsnowwolf