coolsnowwolf/lede #11204: SFTP和SSH不能使用

wuzhuang123, 2023-05-13:
在openssh更新到9.2后就有这个问题，winSCP的SFTP和putty的SSH都无法连接，但是FTP是可以的，网页的ttyd也可以登录。以前降级使用openssh的9.1版本还能解决，现在不行了。把LAN口改到192.168.0.1就有这个问题。
[image]
Sat May 13 10:49:45 2023 auth.info sshd[11029]: Failed password for root from 192.168.0.100 port 56331 ssh2
Sat May 13 10:49:47 2023 auth.info sshd[11029]: Connection closed by authenticating user root 192.168.0.100 port 56331 [preauth]
Sat May 13 10:49:57 2023 auth.info sshd[11062]: Failed password for root from 192.168.0.100 port 52806 ssh2
Sat May 13 10:50:01 2023 auth.info sshd[11062]: Failed password for root from 192.168.0.100 port 52806 ssh2
Sat May 13 10:50:03 2023 auth.info sshd[11062]: Connection closed by authenticating user root 192.168.0.100 port 52806 [preauth]
Sat May 13 10:50:07 2023 auth.info sshd[11090]: Failed password for root from 192.168.0.100 port 52810 ssh2
Sat May 13 10:50:21 2023 auth.err sshd[11090]: error: Received disconnect from 192.168.0.100 port 52810:13: Unable to authenticate [preauth]
Sat May 13 10:50:21 2023 auth.info sshd[11090]: Disconnected from authenticating user root 192.168.0.100 port 52810 [preauth]
Sat May 13 10:50:30 2023 daemon.err uhttpd[1913]: cat: can't open '/etc/bench.log': No such file or directory
Sat May 13 10:50:30 2023 daemon.err uhttpd[1913]: cat: can't open '/etc/bench.log': No such file or directory
Sat May 13 10:50:57 2023 daemon.notice ttyd[4824]: [2023/05/13 10:50:57:1749] N: __lws_lc_tag:  ++ [wsisrv|0|adopted] (1)
Sat May 13 10:50:57 2023 daemon.notice ttyd[4824]: [2023/05/13 10:50:57:1774] N: HTTP / - 192.168.0.100
Sat May 13 10:50:57 2023 daemon.notice ttyd[4824]: [2023/05/13 10:50:57:3030] N: HTTP /token - 192.168.0.100
Sat May 13 10:50:57 2023 daemon.notice ttyd[4824]: [2023/05/13 10:50:57:4007] N: __lws_lc_tag:  ++ [wsisrv|1|adopted] (2)
Sat May 13 10:50:57 2023 daemon.notice ttyd[4824]: [2023/05/13 10:50:57:4029] N: WS   /ws - 192.168.0.100, clients: 1
Sat May 13 10:50:57 2023 daemon.notice ttyd[4824]: [2023/05/13 10:50:57:4088] N: started process, pid: 11184
Sat May 13 10:51:02 2023 daemon.notice ttyd[4824]: [2023/05/13 10:51:02:3057] N: __lws_lc_untag:  -- [wsisrv|0|adopted] (1) 5.130s
Sat May 13 10:51:06 2023 auth.info login[11185]: root login on 'pts/0'

wuzhuang123, 2023-05-13:
型号是小米mini，MT7620

binge8, 2023-05-14:
[image]
今天刚编译的，ssh密匙登陆也是报错

WYC-2020, 2023-05-15:
这个早就不行了，很早以前就不行的 
[430-e_devcrypto-make-the-dev-crypto-engine-dynamic.patch](https://github.com/coolsnowwolf/lede/files/11473500/430-e_devcrypto-make-the-dev-crypto-engine-dynamic.patch)
这个替换lede\package\libs\openssl\patches  相同文件，应该就可以了

binge8, 2023-05-15:
> 这个早就不行了，很早以前就不行的 [430-e_devcrypto-make-the-dev-crypto-engine-dynamic.patch](https://github.com/coolsnowwolf/lede/files/11473500/430-e_devcrypto-make-the-dev-crypto-engine-dynamic.patch) 这个替换lede\package\libs\openssl\patches 相同文件，应该就可以了

也不算很久吧，大概10天前编译的密匙登陆还一直好使，之前也一直好使

WYC-2020, 2023-05-15:
> > 这个早就不行了，很早以前就不行的 [430-e_devcrypto-make-the-dev-crypto-engine-dynamic.patch](https://github.com/coolsnowwolf/lede/files/11473500/430-e_devcrypto-make-the-dev-crypto-engine-dynamic.patch) 这个替换lede\package\libs\openssl\patches 相同文件，应该就可以了
> 
> 也不算很久吧，大概10天前编译的密匙登陆还一直好使，之前也一直好使

我这个是修复密码登录的，以前会出现只有调试模式下面才能进行登录，用这个直接覆盖了不用调试模式也可以，我这边一直这么用的，sftp和ssh 都可以，至于密钥登录还没验证，应该也没问题，你们可以试试，这个密码不能登录已经很久 了openssh，我试过用原来的patch 一样 还是不行

binge8, 2023-05-15:
> 我这个是修复密码登录的，以前会出现只有调试模式下面才能进行登录，用这个直接覆盖了不用调试模式也可以，我这边一直这么用的，sftp和ssh 都可以，至于密钥登录还没验证，应该也没问题，你们可以试试，这个密码不能登录已经很久 了openssh，我试过用原来的patch 一样 还是不行

[image]
用了你的patch报错

binge8, 2023-05-16:
我是跟着库更新的，你的意思是现在库里的openssl要回退上一个版本然后再用你这个pacth？

WYC-2020, 2023-05-16:
> 我是跟着库更新的，你的意思是现在库里的openssl要回退上一个版本然后再用你这个pacth？

不清楚，我这边这个库也是跟到的没出现打补丁失败呢，算了我吧opnessl 整个目录给你
[openssl.zip](https://github.com/coolsnowwolf/lede/files/11483837/openssl.zip)

binge8, 2023-05-21:
> > 我是跟着库更新的，你的意思是现在库里的openssl要回退上一个版本然后再用你这个pacth？
> 
> 不清楚，我这边这个库也是跟到的没出现打补丁失败呢，算了我吧opnessl 整个目录给你 [openssl.zip](https://github.com/coolsnowwolf/lede/files/11483837/openssl.zip)

你这个对比了一下，和现在库里的好像完全相同啊

WYC-2020, 2023-05-25:
> > > 我是跟着库更新的，你的意思是现在库里的openssl要回退上一个版本然后再用你这个pacth？
> > 
> > 
> > 不清楚，我这边这个库也是跟到的没出现打补丁失败呢，算了我吧opnessl 整个目录给你 [openssl.zip](https://github.com/coolsnowwolf/lede/files/11483837/openssl.zip)
> 
> 你这个对比了一下，和现在库里的好像完全相同啊

你对比 之对比名字 肯定一样了 内容是由修改的 有一个patch

binge8, 2023-05-25:
> > > > 我是跟着库更新的，你的意思是现在库里的openssl要回退上一个版本然后再用你这个pacth？
> > > 
> > > 
> > > 不清楚，我这边这个库也是跟到的没出现打补丁失败呢，算了我吧opnessl 整个目录给你 [openssl.zip](https://github.com/coolsnowwolf/lede/files/11483837/openssl.zip)
> > 
> > 
> > 你这个对比了一下，和现在库里的好像完全相同啊
> 
> 你对比 之对比名字 肯定一样了 内容是由修改的 有一个patch

的确好使，密匙也没问题。之前是我搞错了。另外你为什么不提交修复呢

binge8, 2023-05-25:
> > > > > > 我是跟着库更新的，你的意思是现在库里的openssl要回退上一个版本然后再用你这个pacth？
> > > > > 
> > > > > 
> > > > > 不清楚，我这边这个库也是跟到的没出现打补丁失败呢，算了我吧opnessl 整个目录给你 [openssl.zip](https://github.com/coolsnowwolf/lede/files/11483837/openssl.zip)
> > > > 
> > > > 
> > > > 你这个对比了一下，和现在库里的好像完全相同啊
> > > 
> > > 
> > > 你对比 之对比名字 肯定一样了 内容是由修改的 有一个patch
> > 
> > 
> > 的确好使，密匙也没问题。之前是我搞错了。另外你为什么不提交修复呢
> 
> 不会

你真会开玩笑

wuzhuang123, 2023-05-26:
不管用，放弃了，作为主路由打游戏频繁断线。转官方了。。。。

binge8, 2023-06-03:
> > 我是跟着库更新的，你的意思是现在库里的openssl要回退上一个版本然后再用你这个pacth？
> 
> 不清楚，我这边这个库也是跟到的没出现打补丁失败呢，算了我吧opnessl 整个目录给你 [openssl.zip](https://github.com/coolsnowwolf/lede/files/11483837/openssl.zip)

大哥，你这个还好使吗？之前编译的用了一个多礼拜没毛病，昨天编译的好像又不行了，是不是更新了

WYC-2020, 2023-06-05:
> > > 我是跟着库更新的，你的意思是现在库里的openssl要回退上一个版本然后再用你这个pacth？
> > 
> > 
> > 不清楚，我这边这个库也是跟到的没出现打补丁失败呢，算了我吧opnessl 整个目录给你 [openssl.zip](https://github.com/coolsnowwolf/lede/files/11483837/openssl.zip)
> 
> 大哥，你这个还好使吗？之前编译的用了一个多礼拜没毛病，昨天编译的好像又不行了，是不是更新了

从来就没不好使过

binge8, 2023-06-05:
> > > > 我是跟着库更新的，你的意思是现在库里的openssl要回退上一个版本然后再用你这个pacth？
> > > 
> > > 
> > > 不清楚，我这边这个库也是跟到的没出现打补丁失败呢，算了我吧opnessl 整个目录给你 [openssl.zip](https://github.com/coolsnowwolf/lede/files/11483837/openssl.zip)
> > 
> > 
> > 大哥，你这个还好使吗？之前编译的用了一个多礼拜没毛病，昨天编译的好像又不行了，是不是更新了
> 
> 从来就没不好使过

我又搞错了，确实没问题

wuzhuang123, 2023-06-06:
> > > > > 我是跟着库更新的，你的意思是现在库里的openssl要回退上一个版本然后再用你这个pacth？
> > > > 
> > > > 
> > > > 不清楚，我这边这个库也是跟到的没出现打补丁失败呢，算了我吧opnessl 整个目录给你 [openssl.zip](https://github.com/coolsnowwolf/lede/files/11483837/openssl.zip)
> > > 
> > > 
> > > 大哥，你这个还好使吗？之前编译的用了一个多礼拜没毛病，昨天编译的好像又不行了，是不是更新了
> > 
> > 
> > 从来就没不好使过
> 
> 我又搞错了，确实没问题

好烦。这几天尝试5.4全新升级都是没wifi，ssh可以用；切回5.10后ssh不稳定，我两台HC5962相同固件一台可以ssh一台连不上.

WYC-2020, 2023-06-06:
> 好烦。这几天尝试5.4全新升级都是没wifi，ssh可以用；切回5.10后ssh不稳定，我两台HC5962相同固件一台可以ssh一台连不上.

 自己设置问题，反正我可以保证ssh 随便什么内核都可以，别问为什么，问就是亲测

WYC-2020, 2023-06-06:
> > > > > > > > 我是跟着库更新的，你的意思是现在库里的openssl要回退上一个版本然后再用你这个pacth？
> > > > > > > 
> > > > > > > 
> > > > > > > 不清楚，我这边这个库也是跟到的没出现打补丁失败呢，算了我吧opnessl 整个目录给你 [openssl.zip](https://github.com/coolsnowwolf/lede/files/11483837/openssl.zip)
> > > > > > 
> > > > > > 
> > > > > > 大哥，你这个还好使吗？之前编译的用了一个多礼拜没毛病，昨天编译的好像又不行了，是不是更新了
> > > > > 
> > > > > 
> > > > > 从来就没不好使过
> > > > 
> > > > 
> > > > 我又搞错了，确实没问题
> > > 
> > > 
> > > 好烦。这几天尝试5.4全新升级都是没wifi，ssh可以用；切回5.10后ssh不稳定，我两台HC5962相同固件一台可以ssh一台连不上.
> > 
> > 
> > 自己设置问题，反正我可以保证ssh 随便什么内核都可以，别问为什么，问就是亲测
> 
> 不保留配置升级，开机修改主机名和密码后可以连接，刚开始一会都可以，一重启就又不行了。5.4没wifi现在用的5.10应该不是内核的原因。心累。

怕不是没有开机启动吧，反正我这边一切正常，可怜的孩子

binge8, 2023-06-08:
> 怕不是没有开机启动吧，反正我这边一切正常，可怜的孩子

好像是有他说的这个问题，我用所谓的files大法把修改后的密码直接编译进去，就是不行

WYC-2020, 2023-06-08:
> 好像是有他说的这个问题，我用所谓的files大法把修改后的密码直接编译进去，就是不行

什么files大法，没懂，我这边是重启也能连接呢

wuzhuang123, 2023-06-09:
> 什么files大法，没懂，我这边是重启也能连接呢

不做修改直接编译，不修改LAN的ip可以使用SCP连接成功并且重启后仍然可用；但是修改LAN口到192.168.0.1就不能连接，这对于只有单口的设备就是致命的。如果用你的压缩包，每次重启WinSCP都会提示秘钥更改，然后出现开头的错误。

wuzhuang123, 2023-06-09:
> 什么files大法，没懂，我这边是重启也能连接呢

还有，确实每次编译版本号不会改变，即使全新拉取也不行，git都是16773和d4b3

WYC-2020, 2023-06-09:
> 还有，确实每次编译版本号不会改变，即使全新拉取也不行，git都是16773和d4b3

你这个问题和我这个又有什么关系，你那个是openssh脚本有问题，没检测到ip变化重新加载吧，有时间我看下，因为我是luci有配置界面的，自己加了一个，没你这样操作过

WYC-2020, 2023-06-09:
> 不做修改直接编译，不修改LAN的ip可以使用SCP连接成功并且重启后仍然可用；但是修改LAN口到192.168.0.1就不能连接，这对于只有单口的设备就是致命的。如果用你的压缩包，每次重启WinSCP都会提示秘钥更改，然后出现开头的错误。

确定了 看了下openssh的脚本 就是没有检测Lan变化后 进行reload操作，你加一下 应该就好了，我这边 是没有直接使用ssh 我吧openssh 直接和dropbear 集成的，这个脚本有检测网络变化重新加载的，所以我没遇到你说的这个问题，我的是这样的
[image]
[image]

wuzhuang123, 2023-06-10:
> > > > > > > > > > > > > 我是跟着库更新的，你的意思是现在库里的openssl要回退上一个版本然后再用你这个pacth？
> > > > > > > > > > > > 
> > > > > > > > > > > > 
> > > > > > > > > > > > 不清楚，我这边这个库也是跟到的没出现打补丁失败呢，算了我吧opnessl 整个目录给你 [openssl.zip](https://github.com/coolsnowwolf/lede/files/11483837/openssl.zip)
> > > > > > > > > > > 
> > > > > > > > > > > 
> > > > > > > > > > > 大哥，你这个还好使吗？之前编译的用了一个多礼拜没毛病，昨天编译的好像又不行了，是不是更新了
> > > > > > > > > > 
> > > > > > > > > > 
> > > > > > > > > > 从来就没不好使过
> > > > > > > > > 
> > > > > > > > > 
> > > > > > > > > 我又搞错了，确实没问题
> > > > > > > > 
> > > > > > > > 
> > > > > > > > 好烦。这几天尝试5.4全新升级都是没wifi，ssh可以用；切回5.10后ssh不稳定，我两台HC5962相同固件一台可以ssh一台连不上.
> > > > > > > 
> > > > > > > 
> > > > > > > 自己设置问题，反正我可以保证ssh 随便什么内核都可以，别问为什么，问就是亲测
> > > > > > 
> > > > > > 
> > > > > > 不保留配置升级，开机修改主机名和密码后可以连接，刚开始一会都可以，一重启就又不行了。5.4没wifi现在用的5.10应该不是内核的原因。心累。
> > > > > 
> > > > > 
> > > > > 怕不是没有开机启动吧，反正我这边一切正常，可怜的孩子
> > > > 
> > > > 
> > > > 好像是有他说的这个问题，我用所谓的files大法把修改后的密码直接编译进去，就是不行
> > > 
> > > 
> > > 什么files大法，没懂，我这边是重启也能连接呢
> > 
> > 
> > 不做修改直接编译，不修改LAN的ip可以使用SCP连接成功并且重启后仍然可用；但是修改LAN口到192.168.0.1就不能连接，这对于只有单口的设备就是致命的。如果用你的压缩包，每次重启WinSCP都会提示秘钥更改，然后出现开头的错误。
> 
> 确定了 看了下openssh的脚本 就是没有检测Lan变化后 进行reload操作，你加一下 应该就好了，我这边 是没有直接使用ssh 我吧openssh 直接和dropbear 集成的，这个脚本有检测网络变化重新加载的，所以我没遇到你说的这个问题，我的是这样的 [image] [image]
> 
> 还有你说的什么每次git后版本没变化 我没懂你说的什么意思
#11282 近计次编译每次页面都显示git-23.141.16773-28dd4b3，再加上不停改配置，都快分不清要刷哪个文件了

binge8, 2023-09-15:
> 这个早就不行了，很早以前就不行的 [430-e_devcrypto-make-the-dev-crypto-engine-dynamic.patch](https://github.com/coolsnowwolf/lede/files/11473500/430-e_devcrypto-make-the-dev-crypto-engine-dynamic.patch) 这个替换lede\package\libs\openssl\patches 相同文件，应该就可以了

大哥，我看昨天openssl更新了，你这个patch好像不行了啊

binge8, 2023-09-15:
> > 这个早就不行了，很早以前就不行的 [430-e_devcrypto-make-the-dev-crypto-engine-dynamic.patch](https://github.com/coolsnowwolf/lede/files/11473500/430-e_devcrypto-make-the-dev-crypto-engine-dynamic.patch) 这个替换lede\package\libs\openssl\patches 相同文件，应该就可以了
> 
> 大哥，我看昨天openssl更新了，你这个patch好像不行了啊

亲测新的openssl把ssh问题已经修复了

closed 2026-09-05 by coolsnowwolf