coolsnowwolf/lede #11205: [求助] 关于升级到5.10内核后修改CPU主频的hack

asiaon123, 2023-05-13:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：之前5.4的内核之前有个hack 7621的CPU的主频到1000hz的patch，后来被移除了，但是觉得还挺好用的，但是升级到5.10内核后不能用了，求大神帮忙适配一份5.10的patch，不是到源码，我自己编译的时候应用patch，感谢！

### (2) 路由器型号和固件版本

A：HC5962（5.10）

### (3) 详细日志

A：无

asiaon123, 2023-05-13:
有人能帮助么

zfdx123, 2023-05-13:
可以对比一下内核代码，手动打一下patch，然后diff出来patch，放进OpenWRT

WYC-2020, 2023-05-14:
322-mt7621-fix-cpu-clk-add-clkdev.patch 
5.4 5.10 都有这个了
target\linux\ramips\patches-5.10\322-mt7621-fix-cpu-clk-add-clkdev.patch
至于是不是你要的频率位置，自己去看

asiaon123, 2023-05-15:
> 322-mt7621-fix-cpu-clk-add-clkdev.patch 5.4 5.10 都有这个了 target\linux\ramips\patches-5.10\322-mt7621-fix-cpu-clk-add-clkdev.patch 至于是不是你要的频率位置，自己去看

我看了下这个是5.10的patch
但是我对c不是太了解 不知道怎么改，想改到1000hz，所以想求助一下大神

WYC-2020, 2023-05-15:
> 1000hz

5.4都被l撤销了，你确定能超到1000,按照l之前的给你改了一个 你看下行不行吧，不保证可以哈
5.10 
[322-mt7621-fix-cpu-clk-add-clkdev.patch](https://github.com/coolsnowwolf/lede/files/11473883/322-mt7621-fix-cpu-clk-add-clkdev.patch)

asiaon123, 2023-05-15:
> > 1000hz
> 
> 5.4都被l撤销了，你确定能超到1000,按照l之前的给你改了一个 你看下行不行吧，不保证可以哈 5.10 [322-mt7621-fix-cpu-clk-add-clkdev.patch](https://github.com/coolsnowwolf/lede/files/11473883/322-mt7621-fix-cpu-clk-add-clkdev.patch)

我之前用的在我的机器上1000hz还能正常😄，感谢大佬 我试一下 谢谢

asiaon123, 2023-05-15:
> > 1000hz
> 
> 5.4都被l撤销了，你确定能超到1000,按照l之前的给你改了一个 你看下行不行吧，不保证可以哈 5.10 [322-mt7621-fix-cpu-clk-add-clkdev.patch](https://github.com/coolsnowwolf/lede/files/11473883/322-mt7621-fix-cpu-clk-add-clkdev.patch)

有效的 感谢

closed 2023-05-15 by asiaon123

WYC-2020, 2023-05-15:
> > > 1000hz
> > 
> > 
> > 5.4都被l撤销了，你确定能超到1000,按照l之前的给你改了一个 你看下行不行吧，不保证可以哈 5.10 [322-mt7621-fix-cpu-clk-add-clkdev.patch](https://github.com/coolsnowwolf/lede/files/11473883/322-mt7621-fix-cpu-clk-add-clkdev.patch)
> 
> 有效的 感谢

那就好好珍藏哈哈

asiaon123, 2023-05-15:
> > > > 1000hz
> > > 
> > > 
> > > 5.4都被l撤销了，你确定能超到1000,按照l之前的给你改了一个 你看下行不行吧，不保证可以哈 5.10 [322-mt7621-fix-cpu-clk-add-clkdev.patch](https://github.com/coolsnowwolf/lede/files/11473883/322-mt7621-fix-cpu-clk-add-clkdev.patch)
> > 
> > 
> > 有效的 感谢
> 
> 那就好好珍藏哈哈

那必须的😄

asiaon123, 2023-07-01:
@WYC-2020 麻烦能帮忙再改一个5.15内核的修改CPU频率的patch么？  🙏

082530, 2024-12-23:
> > > 1000hz
> > 
> > 
> > 5.4都被l撤销了，你确定能超到1000,按照l之前的给你改了一个 你看下行不行吧，不保证可以哈 5.10 [322-mt7621-fix-cpu-clk-add-clkdev.patch](https://github.com/coolsnowwolf/lede/files/11473883/322-mt7621-fix-cpu-clk-add-clkdev.patch)
> 
> 有效的 感谢

我照着改了编译之后刷入路由器系统启动不了