coolsnowwolf/lede #11206: [X86-64] ip6tables v1.8.7 (legacy): can't initialize ip6tables table `mangle'

CkovMk, 2023-05-13:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：X86-64固件存在与与 #10209 类似的问题。

### (2) 路由器型号和固件版本

A：PVE虚拟机，使用tag:20220401 （其实是20230401）编译。

### (3) 详细日志

A：

```sh
root@vm-rt:~# /etc/init.d/qos  restart
ip6tables v1.8.7 (legacy): can't initialize ip6tables table `mangle': Table does not exist (do you need to insmod?)
Perhaps ip6tables or your kernel needs to be upgraded.
ip6tables v1.8.7 (legacy): can't initialize ip6tables table `mangle': Table does not exist (do you need to insmod?)
Perhaps ip6tables or your kernel needs to be upgraded.
Cannot find device "ifb0"
Cannot find device "ifb0"
Cannot find device "ifb0"
bad action parsing
parse_action: bad value (5:mirred)!
Illegal "action"
Cannot find device "ifb0"
Cannot find device "ifb0"
Cannot find device "ifb0"
Cannot find device "ifb0"
Cannot find device "ifb0"
Cannot find device "ifb0"
Cannot find device "ifb0"
Cannot find device "ifb0"
ip6tables v1.8.7 (legacy): can't initialize ip6tables table `mangle': Table does not exist (do you need to insmod?)
Perhaps ip6tables or your kernel needs to be upgraded.
ip6tables v1.8.7 (legacy): can't initialize ip6tables table `mangle': Table does not exist (do you need to insmod?)
Perhaps ip6tables or your kernel needs to be upgraded.
ip6tables v1.8.7 (legacy): can't initialize ip6tables table `mangle': Table does not exist (do you need to insmod?)
Perhaps ip6tables or your kernel needs to be upgraded.
ip6tables v1.8.7 (legacy): can't initialize ip6tables table `mangle': Table does not exist (do you need to insmod?)
Perhaps ip6tables or your kernel needs to be upgraded.
ip6tables v1.8.7 (legacy): host/network `10.25.5.5' not found
Try `ip6tables -h' or 'ip6tables --help' for more information.
ip6tables v1.8.7 (legacy): host/network `10.25.5.5' not found
Try `ip6tables -h' or 'ip6tables --help' for more information.
```

WYC-2020, 2023-05-15:
这无能为力了，我这边一切正常呢

CkovMk, 2023-05-23:
另外，

Cannot find device "ifb0"

这个问题，openwrt主线早就fix掉了，我试了一下确实不再报这个error了。建议把这个commit merge一下。

https://github.com/openwrt/openwrt/pull/4092/commits/551199c05a3ddfeff1b2c7d77ee2082ce165f140

WYC-2020, 2023-05-24:
可以提一个pr