coolsnowwolf/lede #11208: threaded network backlog processing is not merged and implemented on kernel 6.1.x

GreatMichaelLee, 2023-05-14:
这个问题没有解决啊，在6.1上不work啊，现在把x86 主版本切到6.1了，麻烦在6.1上实现一下。

_Originally posted by @GreatMichaelLee in https://github.com/coolsnowwolf/lede/issues/11161#issuecomment-1546698157_

WYC-2020, 2023-05-14:
你是有多喜欢6.1  关于这个的issue 三四个了 关一个你开一个，哎
[760-net-core-add-optional-threading-for-backlog-processi.patch](https://github.com/coolsnowwolf/lede/files/11471388/760-net-core-add-optional-threading-for-backlog-processi.patch)
小白鼠拿去测试吧

GreatMichaelLee, 2023-05-14:
> 你是有多喜欢6.1 关于这个的issue 三四个了 关一个你开一个，哎 [760-net-core-add-optional-threading-for-backlog-processi.patch](https://github.com/coolsnowwolf/lede/files/11471388/760-net-core-add-optional-threading-for-backlog-processi.patch) 小白鼠拿去测试吧

老板，不是我喜欢6.1，是大雕把PATCH_VER切到6.1了啊... 我喜欢的是这个feature啊，它能解决我的一些performance issue.

WYC-2020, 2023-05-14:
> 老板，不是我喜欢6.1，是大雕把PATCH_VER切到6.1了啊... 我喜欢的是这个feature啊，它能解决我的一些performance issue.

别狡辩，感觉大雕不切你就会老老实实的不提一样哈哈，大雕说我不背锅

GreatMichaelLee, 2023-05-14:
> 别狡辩，感觉大雕不切你就会老老实实的不提一样哈哈，大雕说我不背锅

嘘，🤫我可是老实巴交的良民…

GreatMichaelLee, 2023-05-14:
> 你是有多喜欢6.1 关于这个的issue 三四个了 关一个你开一个，哎 [760-net-core-add-optional-threading-for-backlog-processi.patch](https://github.com/coolsnowwolf/lede/files/11471388/760-net-core-add-optional-threading-for-backlog-processi.patch) 小白鼠拿去测试吧

小白鼠当完了，貌似没问题，哈哈。

WYC-2020, 2023-05-15:
> > 你是有多喜欢6.1 关于这个的issue 三四个了 关一个你开一个，哎 [760-net-core-add-optional-threading-for-backlog-processi.patch](https://github.com/coolsnowwolf/lede/files/11471388/760-net-core-add-optional-threading-for-backlog-processi.patch) 小白鼠拿去测试吧
> 
> 小白鼠当完了，貌似没问题，求merge 哈哈。

多测试一段时间吧 看有什么其他问题没，这玩意 我也是看linux 内核提交那边搞的，有可能有其他功能问题，具体不清楚

GreatMichaelLee, 2023-05-15:
> 多测试一段时间吧 看有什么其他问题没，这玩意 我也是看linux 内核提交那边搞的，有可能有其他功能问题，具体不清楚

ok,我跑几天看看有没有crash和其他问题

GreatMichaelLee, 2023-06-05:
> ok,我跑几天看看有没有crash和其他问题

测试至今，没有发现问题，在6.1.31上也跑了几天了。建议合入 @WYC-2020 @coolsnowwolf

WYC-2020, 2023-06-07:
> 测试至今，没有发现问题，在6.1.31上也跑了几天了。建议合入 @WYC-2020 @coolsnowwolf

你确定性能上面有提升，这玩意我都没有加入哈哈

commit 398eece1 (2023-06-25): generic: sync 6.1 patches from official source

Closed: #11208
closed 2023-06-25 by aiamadeus