coolsnowwolf/lede #11209: R4S/R2S编译报uboot-rockchip错误

fxxccp, 2023-05-14:
## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：默认配置参数、默认插件编译报错

### (2) 路由器型号和固件版本

A：R4S/R2S

### (3) 详细日志

A：binman: Node '/binman/simple-bin/fit': subnode 'images/@atf-SEQ': Failed to read ELF file: Python: No module named 'elftools'
Wrote map file './simple-bin.map' to show errors
make[4]: *** [Makefile:1108: .binman_stamp] Error 1
make[3]: *** [Makefile:325: /workdir/openwrt/build_dir/target-aarch64_generic_musl/u-boot-nanopi-r4s-rk3399/u-boot-2023.04/.built] Error 2
make[4]: Leaving directory '/workdir/openwrt/build_dir/target-aarch64_generic_musl/u-boot-nanopi-r4s-rk3399/u-boot-2023.04'
make[3]: Leaving directory '/workdir/openwrt/package/boot/uboot-rockchip'
time: package/boot/uboot-rockchip/nanopi-r4s-rk3399/compile#2.86#1.00#3.81
    ERROR: package/boot/uboot-rockchip failed to build (build variant: nanopi-r4s-rk3399).
make[2]: *** [package/Makefile:116: package/boot/uboot-rockchip/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 2
Error: Process completed with exit code 2.

WYC-2020, 2023-05-14:
sudo apt install python3-pyelftools

rianjskis, 2023-05-14:
> sudo apt install python3-pyelftools

大佬请问我这个错误是缺少什么呀，也是R2S,,使用云编译
Patch failed!  Please fix /workdir/openwrt/target/linux/rockchip/patches-6.1/113-ethernet-stmicro-stmmac-Add-SGMII-QSGMII-support.patch!
make[4]: Leaving directory '/workdir/openwrt/target/linux/rockchip'
make[4]: *** [Makefile:34: /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-6.1.28/.prepared_422b22f19fac98e5037bad7b82b0ee81] Error 1
make[3]: Leaving directory '/workdir/openwrt/target/linux'
make[3]: *** [Makefile:11: compile] Error 2
time: target/linux/compile#13.42#7.75#17.28
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:30: target/linux/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [target/Makefile:23: /workdir/openwrt/staging_dir/target-aarch64_generic_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 2
Error: Process completed with exit code 2.

WYC-2020, 2023-05-14:
113-ethernet-stmicro-stmmac-Add-SGMII-QSGMII-support.patch 这个patches 有问题 需要修复

WYC-2020, 2023-05-14:
建议别盲目跟6.1  很多型号都没测试完毕 ，很多patches 一堆问题，当然有能力自己修复的可以搞

fxxccp, 2023-05-14:
> 建议别盲目跟6.1 很多型号都没测试完毕 ，很多patches 一堆问题，当然有能力自己修复的可以搞

编译5.15报错的，目前r4s默认5.15。

rianjskis, 2023-05-14:
> 113-ethernet-stmicro-stmmac-Add-SGMII-QSGMII-support.patch 这个patches 有问题 需要修复

上个月开始用起来了6.1 R2S型号，前几天还正常，昨天晚上开始编译就报错了。

WYC-2020, 2023-05-14:
> > 建议别盲目跟6.1 很多型号都没测试完毕 ，很多patches 一堆问题，当然有能力自己修复的可以搞
> 
> 编译5.15报错的，目前r4s默认5.15。

这个我是答复的二楼 你的方法是安装一下基础环境，sudo apt install python3-pyelftools  最简单的是 吧l的redeme 里面的那个基础环境在安装一下 这个有更新

WYC-2020, 2023-05-14:
> > 113-ethernet-stmicro-stmmac-Add-SGMII-QSGMII-support.patch 这个patches 有问题 需要修复
> 
> 上个月开始用起来了6.1 R2S型号，前几天还正常，昨天晚上开始编译就报错了。

最近不是更新了6.1 么 他们没测试到呗

rianjskis, 2023-05-14:
> > > 113-ethernet-stmicro-stmmac-Add-SGMII-QSGMII-support.patch 这个patches 有问题 需要修复
> > 
> > 上个月开始用起来了6.1 R2S型号，前几天还正常，昨天晚上开始编译就报错了。
> 
> 最近不是更新了6.1 么 他们没测试到呗

哈哈哈。好吧，还是得靠大佬们维护，辛苦大佬们了。

WYC-2020, 2023-05-16:
[113-ethernet-stmicro-stmmac-Add-SGMII-QSGMII-support.patch](https://github.com/coolsnowwolf/lede/files/11483855/113-ethernet-stmicro-stmmac-Add-SGMII-QSGMII-support.patch)
来吧看有敢死队没的，功能测试一下，如果怕 可以吧113删除了编译，也就少了一个这个功能而已，不怕的 就替换这个 编译看下功能哈哈

commit 88c14312 (2023-05-16): kernel: fixes 5.15/6.1 build error

Sorry for the delay, I was scolded all night
and I could only fix it until now.

Fixes: #11209, #11213, #11214
closed 2023-05-16 by aiamadeus