coolsnowwolf/lede #11210: x86编译6.1.X内核报错。

jlzsss, 2023-05-14:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A： x86最新版编译报错

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

Packaged contents of /media/jlz/FIRMWARE/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-nft-bridge into /media/jlz/FIRMWARE/lede/bin/targets/x86/64/packages/kmod-nft-bridge_6.1.28-1_x86_64.ipk
rstrip.sh: /media/jlz/FIRMWARE/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-nft-nat/lib/modules/6.1.28/nft_nat.ko: relocatable
rstrip.sh: /media/jlz/FIRMWARE/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-nft-nat/lib/modules/6.1.28/nft_masq.ko: relocatable
rstrip.sh: /media/jlz/FIRMWARE/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-nft-nat/lib/modules/6.1.28/nft_chain_nat.ko: relocatable
Packaged contents of /media/jlz/FIRMWARE/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-nft-nat into /media/jlz/FIRMWARE/lede/bin/targets/x86/64/packages/kmod-nft-nat_6.1.28-1_x86_64.ipk
ERROR: module '/media/jlz/FIRMWARE/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.28/net/ipv4/netfilter/nf_flow_table_ipv4.ko' is missing.
make[2]: *** [modules/netfilter.mk:1165: /media/jlz/FIRMWARE/lede/bin/targets/x86/64/packages/kmod-nft-offload_6.1.28-1_x86_64.ipk] Error 1
make[2]: Leaving directory '/media/jlz/FIRMWARE/lede/package/kernel/linux'
time: package/kernel/linux/compile#10.53#2.31#12.63
    ERROR: package/kernel/linux failed to build.
make[1]: *** [package/Makefile:116: package/kernel/linux/compile] Error 1
make[1]: Leaving directory '/media/jlz/FIRMWARE/lede'
make: *** [/media/jlz/FIRMWARE/lede/include/toplevel.mk:231：package/linux/compile] 错误 2

xiangfeidexiaohuo, 2023-05-14:
2023-05-14T08:51:02.0547591Z Applying /home/runner/work/OpenWrt_Build/OpenWrt_Build/openwrt/target/linux/rockchip/patches-6.1/100-rockchip-use-system-LED-for-OpenWrt.patch using plaintext: 
2023-05-14T08:51:02.0562735Z patching file arch/arm64/boot/dts/rockchip/rk3328-nanopi-r2s.dts
2023-05-14T08:51:02.0566497Z patching file arch/arm64/boot/dts/rockchip/rk3399-nanopi-r4s.dts
2023-05-14T08:51:02.0570479Z 
2023-05-14T08:51:02.0571377Z Applying /home/runner/work/OpenWrt_Build/OpenWrt_Build/openwrt/target/linux/rockchip/patches-6.1/101-net-realtek-r8169-add-LED-configuration-from-OF.patch using plaintext: 
2023-05-14T08:51:02.0587736Z patching file drivers/net/ethernet/realtek/r8169_main.c
2023-05-14T08:51:02.0598127Z 
2023-05-14T08:51:02.0599882Z Applying /home/runner/work/OpenWrt_Build/OpenWrt_Build/openwrt/target/linux/rockchip/patches-6.1/103-arm64-rockchip-add-OF-node-for-USB-eth-on-NanoPi-R2S.patch using plaintext: 
2023-05-14T08:51:02.0652488Z patching file arch/arm64/boot/dts/rockchip/rk3328-nanopi-r2s.dts
2023-05-14T08:51:02.0660222Z 
2023-05-14T08:51:02.0661238Z Applying /home/runner/work/OpenWrt_Build/OpenWrt_Build/openwrt/target/linux/rockchip/patches-6.1/105-nanopi-r4s-sd-signalling.patch using plaintext: 
2023-05-14T08:51:02.0684631Z patching file arch/arm64/boot/dts/rockchip/rk3399-nanopi-r4s.dts
2023-05-14T08:51:02.0688327Z 
2023-05-14T08:51:02.0689627Z Applying /home/runner/work/OpenWrt_Build/OpenWrt_Build/openwrt/target/linux/rockchip/patches-6.1/106-rockchip-rock-pi-4.patch using plaintext: 
2023-05-14T08:51:02.0713350Z patching file arch/arm64/boot/dts/rockchip/Makefile
2023-05-14T08:51:02.0718589Z patching file arch/arm64/boot/dts/rockchip/rk3399-rock-pi-4.dts
2023-05-14T08:51:02.0722007Z 
2023-05-14T08:51:02.0722824Z Applying /home/runner/work/OpenWrt_Build/OpenWrt_Build/openwrt/target/linux/rockchip/patches-6.1/107-mmc-core-set-initial-signal-voltage-on-power-off.patch using plaintext: 
2023-05-14T08:51:02.0738688Z patching file drivers/mmc/core/core.c
2023-05-14T08:51:02.0744187Z 
2023-05-14T08:51:02.0745544Z Applying /home/runner/work/OpenWrt_Build/OpenWrt_Build/openwrt/target/linux/rockchip/patches-6.1/109-arm64-dts-rockchip-rk356x-Fix-PCIe-register-map.patch using plaintext: 
2023-05-14T08:51:02.0770746Z patching file arch/arm64/boot/dts/rockchip/rk356x.dtsi
2023-05-14T08:51:02.0780084Z 
2023-05-14T08:51:02.0782537Z Applying /home/runner/work/OpenWrt_Build/OpenWrt_Build/openwrt/target/linux/rockchip/patches-6.1/110-arm64-rk3568-update-gicv3-its-and-pci-msi-map.patch using plaintext: 
2023-05-14T08:51:02.0803315Z patching file arch/arm64/Kconfig
2023-05-14T08:51:02.0807669Z patching file arch/arm64/boot/dts/rockchip/rk3568.dtsi
2023-05-14T08:51:02.0810792Z patching file arch/arm64/boot/dts/rockchip/rk356x.dtsi
2023-05-14T08:51:02.0817230Z 
2023-05-14T08:51:02.0820120Z Applying /home/runner/work/OpenWrt_Build/OpenWrt_Build/openwrt/target/linux/rockchip/patches-6.1/111-irqchip-gic-v3-add-hackaround-for-rk3568-its.patch using plaintext: 
2023-05-14T08:51:02.0842706Z patching file drivers/irqchip/irq-gic-v3-its.c
2023-05-14T08:51:02.0853681Z 
2023-05-14T08:51:02.0854928Z Applying /home/runner/work/OpenWrt_Build/OpenWrt_Build/openwrt/target/linux/rockchip/patches-6.1/112-arm64-dts-rockchip-rk3568-Add-xpcs-support.patch using plaintext: 
2023-05-14T08:51:02.0876067Z patching file arch/arm64/boot/dts/rockchip/rk3568.dtsi
2023-05-14T08:51:02.0883491Z patching file arch/arm64/boot/dts/rockchip/rk356x.dtsi
2023-05-14T08:51:02.0892152Z 
2023-05-14T08:51:02.0893790Z Applying /home/runner/work/OpenWrt_Build/OpenWrt_Build/openwrt/target/linux/rockchip/patches-6.1/113-ethernet-stmicro-stmmac-Add-SGMII-QSGMII-support.patch using plaintext: 
2023-05-14T08:51:02.0916894Z patching file drivers/net/ethernet/stmicro/stmmac/dwmac-rk.c
2023-05-14T08:51:02.0922711Z Hunk #3 FAILED at 63.
2023-05-14T08:51:02.0923517Z Hunk #4 succeeded at 84 (offset 12 lines).
2023-05-14T08:51:02.0927839Z Hunk #5 succeeded at 97 (offset 12 lines).
2023-05-14T08:51:02.0931442Z Hunk #6 succeeded at 1147 (offset 13 lines).
2023-05-14T08:51:02.0934437Z Hunk #7 succeeded at 1163 (offset 13 lines).
2023-05-14T08:51:02.0937467Z Hunk #8 succeeded at 1275 (offset 14 lines).
2023-05-14T08:51:02.0940962Z Hunk #9 succeeded at 1744 with fuzz 2 (offset 58 lines).
2023-05-14T08:51:02.0944093Z Hunk #10 FAILED at 1747.
2023-05-14T08:51:02.0947101Z Hunk #11 FAILED at 1780.
2023-05-14T08:51:02.0947620Z Hunk #12 succeeded at 1755 (offset -43 lines).
2023-05-14T08:51:02.0953121Z Hunk #13 succeeded at 1854 (offset -49 lines).
2023-05-14T08:51:02.0958284Z make[4]: *** [Makefile:34: /home/runner/work/OpenWrt_Build/OpenWrt_Build/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-6.1.28/.prepared_fca42afbcd1e7f7fa7ef03693fa996e7] Error 1
2023-05-14T08:51:02.0959183Z make[3]: *** [Makefile:11: compile] Error 2
2023-05-14T08:51:02.0961191Z Hunk #14 succeeded at 1943 (offset -49 lines).
2023-05-14T08:51:02.0961789Z Hunk #15 succeeded at 1976 (offset -49 lines).
2023-05-14T08:51:02.0962276Z Hunk #16 succeeded at 1997 (offset -49 lines).
2023-05-14T08:51:02.0963086Z 3 out of 16 hunks FAILED -- saving rejects to file drivers/net/ethernet/stmicro/stmmac/dwmac-rk.c.rej
2023-05-14T08:51:02.0964088Z Patch failed!  Please fix /home/runner/work/OpenWrt_Build/OpenWrt_Build/openwrt/target/linux/rockchip/patches-6.1/113-ethernet-stmicro-stmmac-Add-SGMII-QSGMII-support.patch!
[... cut ...]
2023-05-14T08:51:02.1004915Z make[1]: *** [target/Makefile:23: /home/runner/work/OpenWrt_Build/OpenWrt_Build/openwrt/staging_dir/target-aarch64_generic_musl/stamp/.target_compile] Error 2
2023-05-14T08:51:02.1005655Z make[1]: Leaving directory '/home/runner/work/OpenWrt_Build/OpenWrt_Build/openwrt'
2023-05-14T08:51:02.1012813Z make: *** [/home/runner/work/OpenWrt_Build/OpenWrt_Build/openwrt/include/toplevel.mk:231: world] Error 2
2023-05-14T08:51:02.1100857Z ##[error]Process completed with exit code 2.

rockchip同样报错，应该是6.1.28内核

WYC-2020, 2023-05-14:
> 反馈bug/问题模板，提建议请删除
>
> ## 1.关于你要提交的问题
> Q：是否搜索了issue (使用 "x" 选择)
>
> * [ ]  没有类似的issue
>
> ## 2. 详细叙述
> ### (1) 具体问题
> A： x86最新版编译报错
>
> ### (2) 路由器型号和固件版本
> A：
>
> ### (3) 详细日志
> Packaged contents of [...]

你这个是 linux 6.1  移除了NF_FLOW_TABLE_IPV4

WYC-2020, 2023-05-14:
> 2023-05-14T08:51:02.0547591Z Applying /home/runner/work/OpenWrt_Build/OpenWrt_Build/openwrt/target/linux/rockchip/patches-6.1/100-rockchip-use-system-LED-for-OpenWrt.patch using plaintext: [...]

你i这个是 113-ethernet-stmicro-stmmac-Add-SGMII-QSGMII-support.patch 这个patch要修复 在另外一个帖子我也说过

WYC-2020, 2023-05-14:
> 你这个是 linux 6.1 移除了NF_FLOW_TABLE_IPV4

6.1 合并到这个模块了 CONFIG_NF_FLOW_TABLE_INET  ipv4和ipv6,修改一下就可以了

WYC-2020, 2023-05-14:
如此修复
[image]

commit d52eb58f (2023-05-24): kernel: modules: adapt for kernel 6.1

Adapt kernel modules to support kernel 6.1.

Fixes: #11210