coolsnowwolf/lede #11211: 为何编译内核会出现如下警告提示？！

zxlhhyccc, 2023-05-14:
[image]

WYC-2020, 2023-05-14:
没使用到这个函数而已，有一个宏把这个函数包含其中即可

zxlhhyccc, 2023-05-14:
> 没使用到这个函数而已，有一个宏把这个函数包含其中即可

看了一下，没看到哪个宏包含了这个函数！

WYC-2020, 2023-05-14:
> > 没使用到这个函数而已，有一个宏把这个函数包含其中即可
> 
> 看了一下，没看到哪个宏包含了这个函数！

明天给你看，因为这个我修过

zxlhhyccc, 2023-05-14:
> > > 没使用到这个函数而已，有一个宏把这个函数包含其中即可
> > 
> > 
> > 看了一下，没看到哪个宏包含了这个函数！
> 
> 明天给你看，因为这个我修过

请问大佬能发我你修改的么？谢谢了！

WYC-2020, 2023-05-15:
> > > > 没使用到这个函数而已，有一个宏把这个函数包含其中即可
> > > 
> > > 
> > > 看了一下，没看到哪个宏包含了这个函数！
> > 
> > 
> > 明天给你看，因为这个我修过
> 
> 请问大佬能发我你修改的么？谢谢了！

CONFIG_NF_CONNTRACK_CHAIN_EVENTS  这个宏，我这个可能 你不一定能直接用，因为可能位置不一样，我有自己的一些patch 可能位置与你有偏移 你拿去试下吧，不行就按 这个自己修
5.15 
[952-add-net-conntrack-events-support-multiple-registrant.patch](https://github.com/coolsnowwolf/lede/files/11473486/952-add-net-conntrack-events-support-multiple-registrant.patch)
6.1
[952-add-net-conntrack-events-support-multiple-registrant.patch](https://github.com/coolsnowwolf/lede/files/11473487/952-add-net-conntrack-events-support-multiple-registrant.patch)

zxlhhyccc, 2023-05-15:
谢谢大佬，请问这个补丁在编译内核的时候会有我截图的那些没有用到的函数的警告吗？

WYC-2020, 2023-05-15:
> 谢谢大佬，请问这个补丁在编译内核的时候会有我截图的那些没有用到的函数的警告吗？

不会

zxlhhyccc, 2023-05-15:
> > 谢谢大佬，请问这个补丁在编译内核的时候会有我截图的那些没有用到的函数的警告吗？
> 
> 不会

请问大佬，有5.10内核的952补丁么？

WYC-2020, 2023-05-16:
> > > 谢谢大佬，请问这个补丁在编译内核的时候会有我截图的那些没有用到的函数的警告吗？
> > 
> > 
> > 不会
> 
> 请问大佬，有5.10内核的952补丁么？烦请大佬发一个。谢谢！

5.10 不会有这个报警吧，这个l有啊

zxlhhyccc, 2023-05-16:
> > > > 谢谢大佬，请问这个补丁在编译内核的时候会有我截图的那些没有用到的函数的警告吗？
> > > 
> > > 
> > > 不会
> > 
> > 
> > 请问大佬，有5.10内核的952补丁么？烦请大佬发一个。谢谢！
> 
> 5.10 不会有这个报警吧，这个l有啊

不记得有没有这个报警，我是想准备着。。。

ZqinKing, 2023-05-18:
不用理会，或者在gcc编译的时候增加-Wnounused-function即可。

zxlhhyccc, 2023-05-22:
> 不用理会，或者在gcc编译的时候增加-Wnounused-function即可。

这个是编译内核，应该在哪里增加？不过我认为`@WYC-2020`修改的补丁效果更好。

WYC-2020, 2023-06-07:
> CONFIG_WERROR

打开这个 那有你玩的了，内核编译只要有警告统一当错误处理，有能力的可以打开 慢慢修吧哈哈

KaneGreen, 2023-08-31:
粗略看了一下，Linux内核从5.15开始默认启用`CONFIG_WERROR=y`（似乎后来又放弃了）；然后5.18再次对x86启用该项。
OpenWrt官方从23.05（目前还是rc版本）开始启用`CONFIG_WERROR=y`。
~~个人感觉以后启用`CONFIG_WERROR=y`会是大势所趋。~~

目前我看到的只有两处Warning：
`net/netfilter/nf_conntrack_ecache.c`文件第136行左右的`__nf_conntrack_eventmask_report`函数和
`net/netfilter/nf_conntrack_netlink.c`文件第3123行左右的`ctnetlink_expect_event`函数。
类型均是`defined but not used`（`unused-function`）

应该是和这个patch导致某些这些函数在新的`#ifdef`条件编译中没有被使用有关：https://github.com/coolsnowwolf/lede/blob/master/target/linux/generic/hack-5.15/952-add-net-conntrack-events-support-multiple-registrant.patch

如果能修复，`CONFIG_WERROR=y`就能开启了。

coolsnowwolf, 2026-09-05:
日志中的 `-Wunused-function` 只是 GCC 编译警告，表示在当前内核配置下某些可选 netfilter 回调未被引用，不是编译失败，也不影响生成的内核功能。该类警告由上游内核配置/编译器产生，当前 LEDE 无需修改代码。

closed 2026-09-05 by coolsnowwolf