coolsnowwolf/lede #11213: R5S编译错误

gswangmd-gssi, 2023-05-16:
Applying /home/gswangmd/lede/target/linux/generic/backport-5.15/407-v5.17-mtd-parsers-qcom-Don-t-print-error-message-on-EPROBE.patch using plaintext: 
patching file drivers/mtd/parsers/qcomsmempart.c

Applying /home/gswangmd/lede/target/linux/generic/backport-5.15/410-mtd-next-mtd-parsers-trx-allow-to-use-on-MediaTek-MIPS-SoCs.patch using plaintext: 
patching file drivers/mtd/parsers/Kconfig

Applying /home/gswangmd/lede/target/linux/generic/backport-5.15/430-v6.3-ubi-Fix-failure-attaching-when-vid_hdr-offset-equals.patch using plaintext: 
patching file drivers/mtd/ubi/build.c
Hunk #1 FAILED at 664.
Hunk #2 succeeded at 696 with fuzz 1 (offset 9 lines).
1 out of 2 hunks FAILED -- saving rejects to file drivers/mtd/ubi/build.c.rej
Patch failed!  Please fix /home/gswangmd/lede/target/linux/generic/backport-5.15/430-v6.3-ubi-Fix-failure-attaching-when-vid_hdr-offset-equals.patch!
make[4]: *** [Makefile:31: /home/gswangmd/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.15.110/.prepared_1208ba1a2970455bf1459aa07123b6e5] Error 1
make[4]: Leaving directory '/home/gswangmd/lede/target/linux/rockchip'
make[3]: *** [Makefile:11: compile] Error 2
make[3]: Leaving directory '/home/gswangmd/lede/target/linux'
time: target/linux/compile#8.73#6.98#12.01
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:30: target/linux/compile] Error 1
make[2]: Leaving directory '/home/gswangmd/lede'
make[1]: *** [target/Makefile:23: /home/gswangmd/lede/staging_dir/target-aarch64_generic_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/home/gswangmd/lede'
make: *** [/home/gswangmd/lede/include/toplevel.mk:231: world] Error 2
gswangmd@WMD-HOME:~/lede$

WYC-2020, 2023-05-16:
相同问题，第四个issue

zfdx123, 2023-05-16:
430-v6.3-ubi-Fix-failure-attaching-when-vid_hdr-offset-equals.patch 可以删除在kernel 5.15.111 中已经存在patch相关内容
@coolsnowwolf

zfdx123, 2023-05-16:
[patch修改一](https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/tree/drivers/mtd/ubi/build.c?h=v5.15.111#n665)
[patch修改二](https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/tree/drivers/mtd/ubi/build.c?h=v5.15.111#n691)

VIKINGYFY, 2023-05-16:
同样的问题，等修复吧

WYC-2020, 2023-05-16:
这玩意 昨天才把patch 提交进来，今天就不行了，这到底编译验证过没哦

zfdx123, 2023-05-16:
估计没验证直接copy的，2023-04-20内核就已经合并这个patch了

zfdx123, 2023-05-16:
https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit/drivers/mtd/ubi/build.c?h=v5.15.111&id=85d7a7044b759d865d10395a357632af00de5867

WYC-2020, 2023-05-16:
> 估计没验证直接copy的，2023-04-20内核就已经合并这个patch了

心塞塞 哈哈，应该是从官方op 那边拿过来的，然后没考虑两边5.15内核版本不一致，l 这边的内核更新比较快 很多补丁在那边适合在这边就不一定了，加上没有验证就翻车了

zfdx123, 2023-05-16:
> > 估计没验证直接copy的，2023-04-20内核就已经合并这个patch了
> 
> 心塞塞 哈哈，应该是从官方op 那边拿过来的，然后没考虑两边5.15内核版本不一致，l 这边的内核更新比较快 很多补丁在那边适合在这边就不一定了，加上没有验证就翻车了

看起来5.10似乎还需要，15是已经合并了，10代码稍微看了一眼应该是没合并进去

VIKINGYFY, 2023-05-16:
> 这玩意 昨天才把patch 提交进来，今天就不行了，这到底编译验证过没哦

我的云编译设定是每天早上4点自动编译，早上起来发现编译失败了，手动再起一次编译，依然失败了

WYC-2020, 2023-05-16:
> > > 估计没验证直接copy的，2023-04-20内核就已经合并这个patch了
> > 
> > 
> > 心塞塞 哈哈，应该是从官方op 那边拿过来的，然后没考虑两边5.15内核版本不一致，l 这边的内核更新比较快 很多补丁在那边适合在这边就不一定了，加上没有验证就翻车了
> 
> 看起来5.10似乎还需要，15是已经合并了，10代码稍微看了一眼应该是没合并进去

是的 只有5.10 才需要 5.4 5.15 6.1 都不需要

closed 2023-05-16 by aiamadeus