coolsnowwolf/lede #11217: gcc initial failed

zou0804, 2023-05-17:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述
在x86虚拟机上编译lede 20221001的版本，虚机资源为64核和64G内存，编译toolchain时报错
### (1) 具体问题
/home/builduser/luci/lede/staging_dir/host/lib/libzstd.a(.._.._.._lib_common_pool.c.o): In function `POOL_free.part.3':
pool.c:(.text+0x14f): undefined reference to `pthread_join'
/home/builduser/luci/lede/staging_dir/host/lib/libzstd.a(.._.._.._lib_common_pool.c.o): In function `POOL_create_advanced':
pool.c:(.text+0x2fa): undefined reference to `pthread_create'
/home/builduser/luci/lede/staging_dir/host/lib/libzstd.a(.._.._.._lib_common_pool.c.o): In function `POOL_resize':
pool.c:(.text+0x549): undefined reference to `pthread_create'
collect2: error: ld returned 1 exit status
make[5]: *** [/home/builduser/luci/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0/gcc/c/Make-lang.in:87: cc1] Error 1
rm gcc.pod
make[5]: Leaving directory '/home/builduser/luci/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0-initial/gcc'
make[4]: *** [Makefile:4428: all-gcc] Error 2
make[4]: Leaving directory '/home/builduser/luci/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0-initial'
make[3]: *** [Makefile:37: /home/builduser/luci/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0-initial/.built] Error 2
make[3]: Leaving directory '/home/builduser/luci/lede/toolchain/gcc/initial'
time: toolchain/gcc/initial/compile#1776.58#166.84#204.32
    ERROR: toolchain/gcc/initial failed to build.
make[2]: *** [toolchain/Makefile:97: toolchain/gcc/initial/compile] Error 1
make[2]: Leaving directory '/home/builduser/luci/lede'
make[1]: *** [toolchain/Makefile:93: /home/builduser/luci/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/stamp/.toolchain_compile] Error 2
make[1]: Leaving directory '/home/builduser/luci/lede'
make: *** [/home/builduser/luci/lede/include/toplevel.mk:231: world] Error 2

### (2) 路由器型号和固件版本
x86

### (3) 详细日志

见2中的描述

WYC-2020, 2023-05-17:
万能大法试下再说

zou0804, 2023-05-18:
> 万能大法试下再说

什么万能大法

WYC-2020, 2023-05-18:
> > 万能大法试下再说
> 
> 什么万能大法
  
make distclean

zou0804, 2023-05-22:
> > > 万能大法试下再说
> > 
> > 
> > 什么万能大法
> 
> make distclean

新拉的git代码，没必要执行distclean吧，我测试过，重新拉的20221001的代码，直接编译，仍然会报同样的错

WYC-2020, 2023-05-23:
拉旧的 与之配套的也拉去旧的

wanglouxiaozi, 2024-10-18:
修改lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0-initial/gcc/Makefile里的“ZSTD_LIB = -lzstd”，改为“ZSTD_LIB = -lzstd -lpthread”就可以了

coolsnowwolf, 2026-09-05:
该日志来自旧版本 GCC/toolchain 的 zstd 线程库链接问题。当前 LEDE 已在 toolchain/gcc/common.mk 中显式使用 `-pthread`（`ac_cv_search_ZSTD_compress=... -pthread`），并且当前 GCC/toolchain 代码已持续更新，原始 2022/2023 构建环境无法在现行 master 复现。请先使用当前 master、全新 staging_dir 和完整日志重试；若仍失败请提交新的可复现日志。现有 issue 按 not planned 关闭。

closed 2026-09-05 by coolsnowwolf