coolsnowwolf/lede #11223: NanoPi R5S LAN口的MAC无法固定，每次恢复出厂设置都会生成一个新的mac地址

Gabrielxzx, 2023-05-21:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：NanoPi R5S LAN口的MAC无法固定，每次恢复出厂设置都会生成一个新的mac地址，然后电脑就会提醒发现了新的网络。 这两天刷机新网络已经排到30多了。
情况如下图。
第一次刷入的mac为0D结尾，但恢复出厂后lan口的mac就变成了4E结尾。但是WAN口的mac是固定的。
R5S采用的是瑞芯微刷机工具直接刷入EMMC，没有安装TF卡。
[image]
[image]

编译采用的P3TERX的云编译方案。

### (2) 路由器型号和固件版本

A：FriendlyElec NanoPi R5S，内核版本为5.15.110，R23.5.1

### (3) 详细日志

A：详细的日志见附件

[系统日志-1.txt](https://github.com/coolsnowwolf/lede/files/11523593/-1.txt)
[内核日志-1.txt](https://github.com/coolsnowwolf/lede/files/11523594/-1.txt)
[系统日志-2.txt](https://github.com/coolsnowwolf/lede/files/11523596/-2.txt)
[内核日志-2.txt](https://github.com/coolsnowwolf/lede/files/11523597/-2.txt)

Gabrielxzx, 2023-05-21:
试了下，发现不是恢复出厂设置，而是重启就会生成新的mac

XXXstyle, 2023-05-21:
我的是愁于修改mac不生效。。。你这是不动也会变。。哈哈哈

Gabrielxzx, 2023-05-21:
> 我的是愁于修改mac不生效。。。你这是不动也会变。。哈哈哈

你的设备也是R5S吗？我这个是LAN口的mac在不停变化，其实不影响，因为后面一般都会再接路由器或者AP啥的，但就是很难受才过来提下，另外想看看有没有其他人遇到。

XXXstyle, 2023-05-21:
是R5S，你可以尝试关闭桥接接口功能，单独分配网口，看图上好像你的两个网卡绑定到了一个lan口上

coolsnowwolf, 2023-05-22:
这个是官方 OP 的bug，本来改好了，又被官方覆盖了

closed 2023-05-22 by coolsnowwolf