coolsnowwolf/lede #11224: 编译带luci-app-pcimodem出错

Siriling, 2023-05-21:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：编译带luci-app-pcimodem会报错，大概知道这个问题在quectel_MHI驱动上

### (2) 路由器型号和固件版本

A：x86，6.1

### (3) 详细日志

A：
2023-05-21T12:18:01.8365780Z make[3]: Entering directory '/LedeMASTER/openwrt/package/wwan/driver/quectel_MHI'
2023-05-21T12:18:02.1053321Z touch /LedeMASTER/openwrt/build_dir/target-x86_64_musl/linux-x86_64/pcie_mhi-3.2/.prepared_19f34592610ed73fdbe005a3bc96b975_6664517399ebbbc92a37c5bb081b5c53_check
2023-05-21T12:18:02.1079705Z mkdir -p /LedeMASTER/openwrt/build_dir/target-x86_64_musl/linux-x86_64/pcie_mhi-3.2
2023-05-21T12:18:02.1105117Z cp -fpR ./src/* /LedeMASTER/openwrt/build_dir/target-x86_64_musl/linux-x86_64/pcie_mhi-3.2/
2023-05-21T12:18:02.1287403Z touch /LedeMASTER/openwrt/build_dir/target-x86_64_musl/linux-x86_64/pcie_mhi-3.2/.prepared_19f34592610ed73fdbe005a3bc96b975_6664517399ebbbc92a37c5bb081b5c53
2023-05-21T12:18:02.1316796Z rm -f /LedeMASTER/openwrt/build_dir/target-x86_64_musl/linux-x86_64/pcie_mhi-3.2/.configured_*
2023-05-21T12:18:02.1351379Z rm -f /LedeMASTER/openwrt/staging_dir/target-x86_64_musl/stamp/.quectel_MHI_installed
2023-05-21T12:18:02.1404299Z (cd /LedeMASTER/openwrt/build_dir/target-x86_64_musl/linux-x86_64/pcie_mhi-3.2/./; if [ -x ./configure ]; then find /LedeMASTER/openwrt/build_dir/target-x86_64_musl/linux-x86_64/pcie_mhi-3.2/ -name config.guess | xargs -r chmod u+w; find /LedeMASTER/openwrt/build_dir/target-x86_64_musl/linux-x86_64/pcie_mhi-3.2/ -name config.guess | xargs -r -n1 cp --remove-destination /LedeMASTER/openwrt/scripts/config.guess; find /LedeMASTER/openwrt/build_dir/target-x86_64_musl/linux-x86_64/pcie_mhi-3.2/ -name config.sub | xargs -r chmod u+w; find /LedeMASTER/openwrt/build_dir/target-x86_64_musl/linux-x86_64/pcie_mhi-3.2/ -name config.sub | xargs -r -n1 cp --remove-destination /LedeMASTER/openwrt/scripts/config.sub; AR="x86_64-openwrt-linux-musl-gcc-ar" AS="x86_64-openwrt-linux-musl-gcc -c -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/LedeMASTER/openwrt/build_dir/target-x86_64_musl/linux-x86_64/pcie_mhi-3.2=pcie_mhi-3.2 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD=x86_64-openwrt-linux-musl-ld NM="x86_64-openwrt-linux-musl-gcc-nm" CC="x86_64-openwrt-linux-musl-gcc" GCC="x86_64-openwrt-linux-musl-gcc" CXX="x86_64-openwrt-linux-musl-g++" RANLIB="x86_64-openwrt-linux-musl-gcc-ranlib" STRIP=x86_64-openwrt-linux-musl-strip OBJCOPY=x86_64-openwrt-linux-musl-objcopy OBJDUMP=x86_64-openwrt-linux-musl-objdump SIZE=x86_64-openwrt-linux-musl-size CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/LedeMASTER/openwrt/build_dir/target-x86_64_musl/linux-x86_64/pcie_mhi-3.2=pcie_mhi-3.2 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/LedeMASTER/openwrt/build_dir/target-x86_64_musl/linux-x86_64/pcie_mhi-3.2=pcie_mhi-3.2 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CPPFLAGS="-I/LedeMASTER/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/include -I/LedeMASTER/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include/fortify -I/LedeMASTER/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include " LDFLAGS="-L/LedeMASTER/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/lib -L/LedeMASTER/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/lib -znow -zrelro "   ./configure --target=x86_64-openwrt-linux --host=x86_64-openwrt-linux --build=x86_64-pc-linux-gnu --program-prefix="" --program-suffix="" --prefix=/usr --exec-prefix=/usr --bindir=/usr/bin --sbindir=/usr/sbin --libexecdir=/usr/lib --sysconfdir=/etc --datadir=/usr/share --localstatedir=/var --mandir=/usr/man --infodir=/usr/info --disable-nls  ; fi; )
2023-05-21T12:18:02.1422835Z touch /LedeMASTER/openwrt/build_dir/target-x86_64_musl/linux-x86_64/pcie_mhi-3.2/.configured_68b329da9893e34099c7d8ad5cb9c940
2023-05-21T12:18:02.1447651Z rm -f /LedeMASTER/openwrt/build_dir/target-x86_64_musl/linux-x86_64/pcie_mhi-3.2/.built
2023-05-21T12:18:02.1477669Z touch /LedeMASTER/openwrt/build_dir/target-x86_64_musl/linux-x86_64/pcie_mhi-3.2/.built_check
2023-05-21T12:18:02.1503106Z make -C "/LedeMASTER/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.29" ARCH="x86" CROSS_COMPILE="x86_64-openwrt-linux-musl-" CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/LedeMASTER/openwrt/build_dir/target-x86_64_musl/linux-x86_64/pcie_mhi-3.2=pcie_mhi-3.2 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" M="/LedeMASTER/openwrt/build_dir/target-x86_64_musl/linux-x86_64/pcie_mhi-3.2"  modules
2023-05-21T12:18:02.1531561Z make[4]: Entering directory '/LedeMASTER/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.29'
2023-05-21T12:18:02.5112416Z   CC [M]  /LedeMASTER/openwrt/build_dir/target-x86_64_musl/linux-x86_64/pcie_mhi-3.2/core/mhi_init.o
2023-05-21T12:18:03.3306180Z   CC [M]  /LedeMASTER/openwrt/build_dir/target-x86_64_musl/linux-x86_64/pcie_mhi-3.2/core/mhi_main.o
2023-05-21T12:18:04.3893379Z   CC [M]  /LedeMASTER/openwrt/build_dir/target-x86_64_musl/linux-x86_64/pcie_mhi-3.2/core/mhi_pm.o
2023-05-21T12:18:05.1115482Z   CC [M]  /LedeMASTER/openwrt/build_dir/target-x86_64_musl/linux-x86_64/pcie_mhi-3.2/core/mhi_boot.o
2023-05-21T12:18:05.4458827Z /LedeMASTER/openwrt/build_dir/target-x86_64_musl/linux-x86_64/pcie_mhi-3.2/core/mhi_boot.c: In function '__mhi_download_rddm_in_panic':
2023-05-21T12:18:05.4460970Z /LedeMASTER/openwrt/build_dir/target-x86_64_musl/linux-x86_64/pcie_mhi-3.2/core/mhi_boot.c:203:23: error: implicit declaration of function 'prandom_u32'; did you mean 'prandom_u32_max'? [-Werror=implicit-function-declaration]
2023-05-21T12:18:05.4461891Z   203 |         sequence_id = prandom_u32() & BHIE_RXVECSTATUS_SEQNUM_BMSK;
2023-05-21T12:18:05.4462634Z       |                       ^~~~~~~~~~~
2023-05-21T12:18:05.4462951Z       |                       prandom_u32_max
2023-05-21T12:18:05.7792610Z cc1: some warnings being treated as errors
2023-05-21T12:18:05.7839825Z make[5]: *** [scripts/Makefile.build:250: /LedeMASTER/openwrt/build_dir/target-x86_64_musl/linux-x86_64/pcie_mhi-3.2/core/mhi_boot.o] Error 1
2023-05-21T12:18:05.7843257Z make[4]: *** [Makefile:2012: /LedeMASTER/openwrt/build_dir/target-x86_64_musl/linux-x86_64/pcie_mhi-3.2] Error 2
2023-05-21T12:18:05.7843853Z make[4]: Leaving directory '/LedeMASTER/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.29'
2023-05-21T12:18:05.7844587Z make[3]: *** [Makefile:50: /LedeMASTER/openwrt/build_dir/target-x86_64_musl/linux-x86_64/pcie_mhi-3.2/.built] Error 2
2023-05-21T12:18:05.7846668Z make[3]: Leaving directory '/LedeMASTER/openwrt/package/wwan/driver/quectel_MHI'
2023-05-21T12:18:05.7847499Z time: package/wwan/driver/quectel_MHI/compile#3.62#0.33#3.95
2023-05-21T12:18:05.7856161Z     ERROR: package/wwan/driver/quectel_MHI failed to build.
2023-05-21T12:18:05.7858825Z make[2]: *** [package/Makefile:116: package/wwan/driver/quectel_MHI/compile] Error 1
2023-05-21T12:18:05.7867754Z make[2]: Leaving directory '/LedeMASTER/openwrt'
2023-05-21T12:18:05.7872100Z make[1]: *** [package/Makefile:110: /LedeMASTER/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
2023-05-21T12:18:06.3030180Z make[1]: Leaving directory '/LedeMASTER/openwrt'
2023-05-21T12:18:06.3119472Z make: *** [/LedeMASTER/openwrt/include/toplevel.mk:231: world] Error 2
2023-05-21T12:18:06.3453220Z ##[error]Process completed with exit code 2.

WYC-2020, 2023-05-22:
[image]
简单视频了一下，自己测试一下功能，自己覆盖替换
[quectel_MHI.tar.gz](https://github.com/coolsnowwolf/lede/files/11526873/quectel_MHI.tar.gz)

Siriling, 2023-05-27:
> [image] 简单适合配一下，自己测试一下功能，自己外壳替换 [quectel_MHI.tar.gz](https://github.com/coolsnowwolf/lede/files/11526873/quectel_MHI.tar.gz)
感谢大佬，已解决了。经实测编译通过，功能也没有问题。

coolsnowwolf, 2026-09-05:
已核对当前源码：`package/wwan/driver/quectel_MHI/src/core/mhi_boot.c` 已在提交 `68253688d81`（`quectel_MHI: add support Linux Kernel 6.1+`）中为 Linux 6.1 及以上改用 `get_random_u32()`，三处 `prandom_u32()` 调用均有版本兼容处理。issue 作者也已确认附件方案编译通过且功能正常，故按已解决关闭。

closed 2026-09-05 by coolsnowwolf