coolsnowwolf/lede #11226: r5s 编译，报错

gswangmd-gssi, 2023-05-22:
Applying /home/gswangmd/lede/target/linux/generic/pending-5.15/790-bus-mhi-core-add-SBL-state-callback.patch using plaintext: 
can't find file to patch at input line 23
Perhaps you used the wrong -p or --strip option?
The text leading up to this was:
--------------------------
|From 5f7c5e1c0d7a79be144e5efc1f24728ddd7fc25c Mon Sep 17 00:00:00 2001
|From: Robert Marko <robimarko@gmail.com>
|Date: Sat, 5 Nov 2022 20:02:56 +0100
|Subject: [PATCH 1/2] bus: mhi: core: add SBL state callback
|
|Add support for SBL state callback in MHI core.
|
|It is required for ath11k MHI devices in order to be able to set QRTR
|instance ID in the SBL state so that QRTR instance ID-s dont conflict in
|case of multiple PCI/MHI cards or AHB + PCI/MHI card.
|Setting QRTR instance ID is only possible in SBL state and there is
|currently no way to ensure that we are in that state, so provide a
|callback that the controller can trigger off.
|
|Signed-off-by: Robert Marko <robimarko@gmail.com>
|---
| drivers/bus/mhi/core/main.c | 1 +
| include/linux/mhi.h         | 2 ++
| 2 files changed, 3 insertions(+)
|
|--- a/drivers/bus/mhi/core/main.c
|+++ b/drivers/bus/mhi/core/main.c
--------------------------
No file to patch.  Skipping patch.
1 out of 1 hunk ignored
patching file include/linux/mhi.h
Patch failed!  Please fix /home/gswangmd/lede/target/linux/generic/pending-5.15/790-bus-mhi-core-add-SBL-state-callback.patch!
make[4]: *** [Makefile:32: /home/gswangmd/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.15.112/.prepared_66c570a388c79e5d5c74d4c3e6ec2df6] Error 1
make[4]: Leaving directory '/home/gswangmd/lede/target/linux/rockchip'
make[3]: *** [Makefile:11: compile] Error 2
make[3]: Leaving directory '/home/gswangmd/lede/target/linux'
time: target/linux/compile#9.61#7.34#11.98
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:30: target/linux/compile] Error 1
make[2]: Leaving directory '/home/gswangmd/lede'
make[1]: *** [target/Makefile:23: /home/gswangmd/lede/staging_dir/target-aarch64_generic_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/home/gswangmd/lede'
make: *** [/home/gswangmd/lede/include/toplevel.mk:231: world] Error 2

WYC-2020, 2023-05-22:
patch 过时

WYC-2020, 2023-05-23:
[790-bus-mhi-core-add-SBL-state-callback.patch](https://github.com/coolsnowwolf/lede/files/11538447/790-bus-mhi-core-add-SBL-state-callback.patch)

commit a0948d91 (2023-05-24): kernel: refresh 5.15 patches (#11235)

Fixes: cc22efe ("kernel: bump 5.15 to 5.15.112")
Fixes: #11226, #11228
Closed: #11232
closed 2023-05-24 by aiamadeus