coolsnowwolf/lede #11227: MTK7620 ykl1 编译出错

mitsukileung, 2023-05-22:
Applying /workdir/openwrt/target/linux/generic/backport-5.10/430-v6.3-ubi-Fix-failure-attaching-when-vid_hdr-offset-equals.patch using plaintext: 
patching file drivers/mtd/ubi/build.c
make[3]: *** [Makefile:121: /workdir/openwrt/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/linux-5.10.180/.prepared] Error 1
Hunk #1 FAILED at 665.
Hunk #2 succeeded at 697 with fuzz 1 (offset 9 lines).
1 out of 2 hunks FAILED -- saving rejects to file drivers/mtd/ubi/build.c.rej
Patch failed!  Please fix /workdir/openwrt/target/linux/generic/backport-5.10/430-v6.3-ubi-Fix-failure-attaching-when-vid_hdr-offset-equals.patch!
make[3]: Leaving directory '/workdir/openwrt/toolchain/kernel-headers'
time: toolchain/kernel-headers/compile#9.53#4.35#11.51
    ERROR: toolchain/kernel-headers failed to build.
make[2]: *** [toolchain/Makefile:97: toolchain/kernel-headers/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [toolchain/Makefile:93: /workdir/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/stamp/.toolchain_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 2
Error: Process completed with exit code 2.

wzming, 2023-05-22:
K2P编译也报这个错误，云编译和本地编译都是报这个错误

mitsukileung, 2023-05-22:
> > 应该是内核升级导致的，改回去就可以了
> 
> 我是直接把这个patch干掉了，正在编译中

删了也是编译不过

wzming, 2023-05-22:
> > > 应该是内核升级导致的，改回去就可以了
> > 
> > 
> > 我是直接把这个patch干掉了，正在编译中
> 
> 删了也是编译不过

我这边删除patch后能正常编译，你要看下删了之后编译的报错日志了

WYC-2020, 2023-05-23:
更新5.10后不需要这个补丁了 自己删除就是了 @mitsukileung

commit 3c488e7b (2023-05-23): generic: 5.10: drop upstream patch (#11231)

Fixes: #11227
closed 2023-05-23 by aiamadeus