coolsnowwolf/lede #11232: rockchip armv8 R2S R4S R5C R66S R68S H66K H68K 编译失败

VIKINGYFY, 2023-05-22:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
今早4点自动编译失败

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：
编译失败，应该是补丁出问题了

### (2) 路由器型号和固件版本

A：
R2S R4S R5C R66S R68S H66K H68K
CONFIG_TARGET_rockchip_armv8=y

### (3) 详细日志

A：
Applying /home/runner/work/OpenWRT-CI/OpenWRT-CI/owrt/target/linux/generic/pending-5.15/790-bus-mhi-core-add-SBL-state-callback.patch using plaintext: 
can't find file to patch at input line 23
Perhaps you used the wrong -p or --strip option?
The text leading up to this was:
--------------------------
|From 5f7c5e1c0d7a79be144e5efc1f24728ddd7fc25c Mon Sep 17 00:00:00 2001
make[3]: *** [Makefile:122: /home/runner/work/OpenWRT-CI/OpenWRT-CI/owrt/build_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/linux-5.15.112/.prepared] Error 1
|From: Robert Marko <robimarko@gmail.com>
|Date: Sat, 5 Nov [2022](https://github.com/VIKINGYFY/OpenWRT-CI/actions/runs/5049956543/jobs/9060071784#step:12:2023) 20:02:56 +0100
|Subject: [PATCH 1/2] bus: mhi: core: add SBL state callback
|
|Add support for SBL state callback in MHI core.
|
|It is required for ath11k MHI devices in order to be able to set QRTR
|instance ID in the SBL state so that QRTR instance ID-s dont conflict in
|case of multiple PCI/MHI cards or AHB + PCI/MHI card.
make[2]: *** [toolchain/Makefile:97: toolchain/kernel-headers/compile] Error 1
|Setting QRTR instance ID is only possible in SBL state and there is
|currently no way to ensure that we are in that state, so provide a
|callback that the controller can trigger off.
|
|Signed-off-by: Robert Marko <robimarko@gmail.com>
make[1]: *** [toolchain/Makefile:93: /home/runner/work/OpenWRT-CI/OpenWRT-CI/owrt/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/stamp/.toolchain_compile] Error 2
|---
make: *** [/home/runner/work/OpenWRT-CI/OpenWRT-CI/owrt/include/toplevel.mk:231: world] Error 2
| drivers/bus/mhi/core/main.c | 1 +
| include/linux/mhi.h         | 2 ++
| 2 files changed, 3 insertions(+)
|
|--- a/drivers/bus/mhi/core/main.c
|+++ b/drivers/bus/mhi/core/main.c
--------------------------
No file to patch.  Skipping patch.
1 out of 1 hunk ignored
patching file include/linux/mhi.h
Patch failed!  Please fix /home/runner/work/OpenWRT-CI/OpenWRT-CI/owrt/target/linux/generic/pending-5.15/790-bus-mhi-core-add-SBL-state-callback.patch!
make[3]: Leaving directory '/home/runner/work/OpenWRT-CI/OpenWRT-CI/owrt/toolchain/kernel-headers'
time: toolchain/kernel-headers/compile#10.42#4.89#12.84
    ERROR: toolchain/kernel-headers failed to build.
make[2]: Leaving directory '/home/runner/work/OpenWRT-CI/OpenWRT-CI/owrt'
make[1]: Leaving directory '/home/runner/work/OpenWRT-CI/OpenWRT-CI/owrt'
Error: Process completed with exit code 2.

WYC-2020, 2023-05-22:
就别再发相同的了，这790patch在通用里面，所有用5.15的都不会过的，等修复吧，这个问题好几个issue了，都还没醒呢，晚点看

WYC-2020, 2023-05-23:
[790-bus-mhi-core-add-SBL-state-callback.patch](https://github.com/coolsnowwolf/lede/files/11538441/790-bus-mhi-core-add-SBL-state-callback.patch)
拿去

kxdn3, 2023-05-23:
一天了还没修复

commit a0948d91 (2023-05-24): kernel: refresh 5.15 patches (#11235)

Fixes: cc22efe ("kernel: bump 5.15 to 5.15.112")
Fixes: #11226, #11228
Closed: #11232
closed 2023-05-24 by aiamadeus