coolsnowwolf/lede #11238: ERROR: toolchain/kernel-headers failed to build.

free-diy, 2023-05-24:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：编译报错

### (2) 路由器型号和固件版本

A：x86

### (3) 详细日志

A：No file to patch.  Skipping patch.
1 out of 1 hunk ignored
patching file include/linux/mhi.h
Patch failed!  Please fix /workdir/openwrt/target/linux/generic/pending-5.15/790-bus-mhi-core-add-SBL-state-callback.patch!
make[3]: Leaving directory '/workdir/openwrt/toolchain/kernel-headers'
time: toolchain/kernel-headers/compile#10.57#5.05#13.11
make[3]: *** [Makefile:122: /workdir/openwrt/build_dir/toolchain-x86_64_gcc-11.3.0_musl/linux-5.15.112/.prepared] Error 1
    ERROR: toolchain/kernel-headers failed to build.
make[2]: *** [toolchain/Makefile:97: toolchain/kernel-headers/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [toolchain/Makefile:93: /workdir/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/stamp/.toolchain_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 2
Error: Process completed with exit code 2.

WYC-2020, 2023-05-24:
你可真是，你下面那个issue 和你这个一模一样 非要来单独提一个

free-diy, 2023-05-24:
> 你可真是，你下面那个问题和你这个一模一样 非要来单独提一个

下面那个是r4s,我这是x86，看清楚好伐

closed 2023-05-24 by coolsnowwolf