coolsnowwolf/lede #11244: 最新代码R4S还是编译失败

liuyanantj, 2023-05-26:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [X] 没有类似的issue

## 2. 详细叙述
最新的代码编译R4S固件  单线程编译  还是失败  Ubuntu 2020 lts环境

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

make[4]: Leaving directory '/home/tc/lede/build_dir/target-aarch64_generic_musl/u-boot-nanopi-r4s-rk3399/u-boot-2023.04'
/home/tc/lede/staging_dir/host/bin/sed -i -e 's/CONFIG_TOOLS_LIBCRYPTO=y/# CONFIG_TOOLS_LIBCRYPTO is not set/' /home/tc/lede/build_dir/target-aarch64_generic_musl/u-boot-nanopi-r4s-rk3399/u-boot-2023.04/.config
/home/tc/lede/staging_dir/host/bin/sed -i -e 's#CONFIG_MKIMAGE_DTC_PATH=.*#CONFIG_MKIMAGE_DTC_PATH="/home/tc/lede/build_dir/target-aarch64_generic_musl/u-boot-nanopi-r4s-rk3399/u-boot-2023.04/scripts/dtc/dtc"#g' /home/tc/lede/build_dir/target-aarch64_generic_musl/u-boot-nanopi-r4s-rk3399/u-boot-2023.04/.config
echo 'CONFIG_IDENT_STRING=" OpenWrt"' >> /home/tc/lede/build_dir/target-aarch64_generic_musl/u-boot-nanopi-r4s-rk3399/u-boot-2023.04/.config
touch /home/tc/lede/build_dir/target-aarch64_generic_musl/u-boot-nanopi-r4s-rk3399/u-boot-2023.04/.configured_68b329da9893e34099c7d8ad5cb9c940
rm -f /home/tc/lede/build_dir/target-aarch64_generic_musl/u-boot-nanopi-r4s-rk3399/u-boot-2023.04/.built
touch /home/tc/lede/build_dir/target-aarch64_generic_musl/u-boot-nanopi-r4s-rk3399/u-boot-2023.04/.built_check
make  -C /home/tc/lede/build_dir/target-aarch64_generic_musl/u-boot-nanopi-r4s-rk3399/u-boot-2023.04 CROSS_COMPILE=aarch64-openwrt-linux-musl- DTC="/home/tc/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.15.112/scripts/dtc/dtc" HOSTCC="gcc" HOSTCFLAGS="-O2 -I/home/tc/lede/staging_dir/host/include -I/home/tc/lede/staging_dir/hostpkg/include -I/home/tc/lede/staging_dir/target-aarch64_generic_musl/host/include -I/home/tc/lede/staging_dir/host/include -I/home/tc/lede/staging_dir/hostpkg/include -I/home/tc/lede/staging_dir/target-aarch64_generic_musl/host/include -std=gnu11" HOSTLDFLAGS="-L/home/tc/lede/staging_dir/host/lib -L/home/tc/lede/staging_dir/hostpkg/lib -L/home/tc/lede/staging_dir/target-aarch64_generic_musl/host/lib" LOCALVERSION="-OpenWrt-r5992-aab8a0568" STAGING_PREFIX="/home/tc/lede/staging_dir/host" PKG_CONFIG_PATH="/home/tc/lede/staging_dir/host/lib/pkgconfig" PKG_CONFIG_LIBDIR="/home/tc/lede/staging_dir/host/lib/pkgconfig" PKG_CONFIG_EXTRAARGS="--static" V='' PATH=/home/tc/lede/staging_dir/host/bin:/home/tc/lede/staging_dir/target-aarch64_generic_musl/host/bin:/home/tc/lede/staging_dir/hostpkg/bin:/home/tc/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/bin:/home/tc/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/bin:/home/tc/lede/staging_dir/host/bin:/home/tc/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/bin:/home/tc/lede/staging_dir/host/bin:/home/tc/lede/staging_dir/host/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin BL31=/home/tc/lede/staging_dir/target-aarch64_generic_musl/image/rk3399_bl31_v1.35.elf
make[4]: Entering directory '/home/tc/lede/build_dir/target-aarch64_generic_musl/u-boot-nanopi-r4s-rk3399/u-boot-2023.04'
  HOSTCC  scripts/basic/fixdep
  HOSTCC  scripts/kconfig/conf.o
  HOSTCC  scripts/kconfig/zconf.tab.o
  HOSTLD  scripts/kconfig/conf
scripts/kconfig/conf  --syncconfig Kconfig
.config:2296:warning: override: reassigning to symbol IDENT_STRING
  UPD     include/config.h
  CFG     u-boot.cfg
  GEN     include/autoconf.mk
  GEN     include/autoconf.mk.dep
  CFG     spl/u-boot.cfg
  GEN     spl/include/autoconf.mk
  CFG     tpl/u-boot.cfg
  GEN     tpl/include/autoconf.mk
  UPD     include/config/uboot.release
  UPD     include/generated/version_autogenerated.h
  UPD     include/generated/timestamp_autogenerated.h
  UPD     include/generated/dt.h
  ENVC    include/generated/env.txt
  ENVP    include/generated/env.in
  ENVT    include/generated/environment.h
  CC      lib/asm-offsets.s
  UPD     include/generated/generic-asm-offsets.h
  CC      arch/arm/lib/asm-offsets.s
  UPD     include/generated/asm-offsets.h
make -f ./scripts/Makefile.build obj=scripts/dtc
  SHIPPED scripts/dtc/pylibfdt/libfdt.i
  PYMOD   rebuild
Traceback (most recent call last):
  File "scripts/dtc/pylibfdt/setup.py", line 23, in <module>
    from setuptools import setup, Extension
ModuleNotFoundError: No module named 'setuptools'
make[6]: *** [scripts/dtc/pylibfdt/Makefile:33: rebuild] Error 1
make[5]: *** [scripts/Makefile.build:397: scripts/dtc/pylibfdt] Error 2
make[4]: *** [Makefile:2003: scripts_dtc] Error 2
make[4]: Leaving directory '/home/tc/lede/build_dir/target-aarch64_generic_musl/u-boot-nanopi-r4s-rk3399/u-boot-2023.04'
make[3]: *** [Makefile:325: /home/tc/lede/build_dir/target-aarch64_generic_musl/u-boot-nanopi-r4s-rk3399/u-boot-2023.04/.built] Error 2
make[3]: Leaving directory '/home/tc/lede/package/boot/uboot-rockchip'
time: package/boot/uboot-rockchip/nanopi-r4s-rk3399/compile#6.27#2.55#8.29
    ERROR: package/boot/uboot-rockchip failed to build (build variant: nanopi-r4s-rk3399).
make[2]: *** [package/Makefile:116: package/boot/uboot-rockchip/compile] Error 1
make[2]: Leaving directory '/home/tc/lede'
make[1]: *** [package/Makefile:110: /home/tc/lede/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/tc/lede'
make: *** [/home/tc/lede/include/toplevel.mk:231：world] 错误 2

救救我吧

WYC-2020, 2023-05-26:
懒得给你找是哪一个了 

sudo apt-get update

sudo apt-get -y install build-essential asciidoc binutils bzip2 gawk gettext git libncurses5-dev libz-dev patch python3 python2.7 unzip zlib1g-dev lib32gcc1 libc6-dev-i386 subversion flex uglifyjs git-core gcc-multilib p7zip p7zip-full msmtp libssl-dev texinfo libglib2.0-dev xmlto qemu-utils upx libelf-dev autoconf automake libtool autopoint device-tree-compiler g++-multilib antlr3 gperf wget curl swig rsync aria2 ca-certificates python3-pyelftools python3-setuptools yasm libpython3-dev
 

拿去直接运行

WYC-2020, 2023-05-26:
你应该只需要这个 python3-setuptools 但是怕是其他还有什么依赖直接给你我这边准备的

liuyanantj, 2023-05-26:
> 你应该只需要这个 python3-setuptools 但是怕是其他还有什么依赖直接给你我这边准备的

感谢大佬  弄完了就好使了！

closed 2023-05-27 by aiamadeus