coolsnowwolf/lede #11246: R4S开启gpu驱动编译失败

GreenTeodoro839, 2023-05-26:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：R4S编译失败

### (2) 路由器型号和固件版本

A：R4S

### (3) 详细日志

A：骷颅头的这个配置有gpu，https://github.com/DHDAXCW/NanoPi-R4S-R4SE/commit/7e2b5b41c81cf8a83d14f1dbac2cf1c6ada8cc96，结果就是这个action https://github.com/DHDAXCW/NanoPi-R4S-R4SE/actions/runs/4829329071/jobs/8604234061，编译失败。我本地编译也是卡在这里，不过去年相同配置正常，不知道改了什么。

nuo12138, 2023-05-27:
lede里面也有r4s的gpu驱动啊，我这边是编译成功的

WYC-2020, 2023-05-27:
日志

GreenTeodoro839, 2023-05-27:
> 日志

make[2] -C package/system/ubox compile
 make[2] -C package/utils/ucode compile
 make[2] -C package/lean/vsftpd-alt compile
 make[2] -C package/network/services/odhcpd compile
 make[2] -C package/community/OpenAppFilter/open-app-filter compile
 make[2] -C package/utils/busybox compile
 make[2] -C customfeeds/packages/libs/libgd compile
 make[2] -C customfeeds/packages/libs/sqlite3 compile
 make[2] -C customfeeds/packages/utils/bash compile
 make[2] -C customfeeds/packages/utils/bc compile
 make[2] -C customfeeds/packages/libs/avahi compile
 make[2] -C customfeeds/packages/utils/cgroupfs-mount compile
 make[2] -C package/utils/e2fsprogs compile
 make[2] -C customfeeds/packages/utils/parted compile
 make[2] -C customfeeds/packages/admin/netdata compile
 make[2] -C package/utils/f2fs-tools compile
 make[2] -C customfeeds/packages/libs/libmbim compile
 make[2] -C customfeeds/packages/utils/irqbalance compile
 make[2] -C customfeeds/packages/libs/libqrtr-glib compile
 make[2] -C customfeeds/luci/applications/luci-app-wifischedule compile
 make[2] -C customfeeds/luci/applications/luci-app-vsftpd compile
 make[2] -C package/system/rpcd compile
 make[2] -C package/lean/autocore compile
 make[2] -C package/system/urandom-seed compile
 make[2] -C customfeeds/packages/libs/avahi compile
 make[2] -C package/utils/f2fs-tools compile
 make[2] -C package/community/luci-app-irqbalance compile
 make[2] -C customfeeds/packages/libs/libqmi compile
 make[2] -C package/kernel/linux compile
 make[2] -C customfeeds/luci/libs/rpcd-mod-rrdns compile
 make[2] -C customfeeds/luci/applications/luci-app-netdata compile
 make[2] -C customfeeds/packages/libs/avahi compile
 make[2] -C package/utils/f2fs-tools compile
 make[2] -C package/utils/f2fs-tools compile
 make[2] -C package/community/luci-app-aliyundrive-webdav compile
 make[2] -C customfeeds/packages/net/samba4 host-compile
    ERROR: package/kernel/linux failed to build.
make -r package/compile: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
make: *** [/home/d/actions/r4s/_work/NanoPi-R4S-R4SE/NanoPi-R4S-R4SE/openwrt/include/toplevel.mk:231: package/compile] Error 1
Error: Process completed with exit code 2.

完整记录在上面那个action输出

closed 2023-05-28 by GreenTeodoro839