coolsnowwolf/lede #11247: R7800最新源码仍未能通过编译

pk49800, 2023-05-27:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：使用最新源码编译固件报错，具体表现为与昨天提交的错误相似，缺少某个文件

### (2) 路由器型号和固件版本

A：R7800

### (3) 详细日志

```bash
make[3]: Entering directory '/home/pk49800/lede/feeds/luci/libs/rpcd-mod-rrdns'
 make[4] scripts/config/conf
mkdir -p /home/pk49800/lede/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/stamp
SHELL= flock /home/pk49800/lede/tmp/.root-copy.flock -c 'cp -fpR /home/pk49800/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/urandom-seed/.pkgdir/urandom-seed/. /home/pk49800/lede/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/'
touch /home/pk49800/lede/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/stamp/.urandom-seed_installed
echo "urandom-seed" >> /home/pk49800/lede/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/pkginfo/urandom-seed.default.install
make[3]: Leaving directory '/home/pk49800/lede/package/system/urandom-seed'
time: package/system/urandom-seed/compile#0.10#0.00#0.10
mkdir -p /home/pk49800/lede/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/stamp
SHELL= flock /home/pk49800/lede/tmp/.root-copy.flock -c 'cp -fpR /home/pk49800/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/rpcd-mod-rrdns-20170710/.pkgdir/rpcd-mod-rrdns/. /home/pk49800/lede/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/'
touch /home/pk49800/lede/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/stamp/.rpcd-mod-rrdns_installed
echo "rpcd-mod-rrdns" >> /home/pk49800/lede/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/pkginfo/rpcd-mod-rrdns.default.install
make[3]: Leaving directory '/home/pk49800/lede/feeds/luci/libs/rpcd-mod-rrdns'
time: package/feeds/luci/rpcd-mod-rrdns/compile#0.08#0.00#0.08
mkdir -p /home/pk49800/lede/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/stamp
SHELL= flock /home/pk49800/lede/tmp/.root-copy.flock -c 'cp -fpR /home/pk49800/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/luci-app-wol/.pkgdir/luci-app-wol/. /home/pk49800/lede/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/'
touch /home/pk49800/lede/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/stamp/.luci-app-wol_installed
echo "luci-app-wol" >> /home/pk49800/lede/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/pkginfo/luci-app-wol.default.install
mkdir -p /home/pk49800/lede/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/stamp
SHELL= flock /home/pk49800/lede/tmp/.root-copy.flock -c 'cp -fpR /home/pk49800/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/luci-app-wol/.pkgdir/luci-i18n-wol-zh-cn/. /home/pk49800/lede/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/'
touch /home/pk49800/lede/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/stamp/.luci-i18n-wol-zh-cn_installed
echo "luci-i18n-wol-zh-cn" >> /home/pk49800/lede/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/pkginfo/luci-app-wol.default.install
make[3]: Leaving directory '/home/pk49800/lede/feeds/luci/applications/luci-app-wol'
time: package/feeds/luci/luci-app-wol/compile#0.66#0.05#0.72
mkdir -p /home/pk49800/lede/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/stamp

SHELL= flock /home/pk49800/lede/tmp/.root-copy.flock -c 'cp -fpR /home/pk49800/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/packages/.pkgdir/kernel/. /home/pk49800/lede/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/'
touch /home/pk49800/lede/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/stamp/.kernel_installed
ERROR: module '/home/pk49800/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/linux-5.4.243/arch/arm/crypto/sha1-arm.ko' is missing.
make[3]: *** [modules/crypto.mk:917: /home/pk49800/lede/bin/targets/ipq806x/generic/packages/kmod-crypto-sha1_5.4.243-1_arm_cortex-a15_neon-vfpv4.ipk] Error 1
make[3]: Leaving directory '/home/pk49800/lede/package/kernel/linux'
time: package/kernel/linux/compile#1.85#0.08#1.93
    ERROR: package/kernel/linux failed to build.
make[2]: *** [package/Makefile:116: package/kernel/linux/compile] Error 1
make[2]: Leaving directory '/home/pk49800/lede'
make[1]: *** [package/Makefile:110: /home/pk49800/lede/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/stamp/.package_compile] Error 2
make[1]: *** Waiting for unfinished jobs....
make[2]: Leaving directory '/home/pk49800/lede'
make[1]: Leaving directory '/home/pk49800/lede'
make: *** [/home/pk49800/lede/include/toplevel.mk:231: world] Error 2
```

WYC-2020, 2023-05-27:
[image]

pk49800, 2023-05-27:
> [image]

清理了tmp文件夹还是无法编译成功，还是报一样的错，请问有什么解决方案，或许我有些地方清理的不够彻底？

WYC-2020, 2023-05-27:
> > [image]
> 
> 清理了tmp文件夹还是无法编译成功，还是报一样的错，请问有什么解决方案，或许我有些地方清理的不够彻底？

改了后 重新编译一下内核，在编译你的模块

WYC-2020, 2023-05-27:
> > [image]
> 
> 清理了tmp文件夹还是无法编译成功，还是报一样的错，请问有什么解决方案，或许我有些地方清理的不够彻底？

先暂时别用 我细看了一下 虽然编译没问题，但是这是一个空包
[image]
我在研究研究

pk49800, 2023-05-27:
> > > [image]
> > 
> > 
> > 清理了tmp文件夹还是无法编译成功，还是报一样的错，请问有什么解决方案，或许我有些地方清理的不够彻底？
> 
> 先暂时别用 我细看了一下 虽然编译没问题，但是这是一个空包 [image] 我在研究研究

好的，我修改完文件但不会单独编译内核和模块就直接用命令跑编译固件没报错，既然大佬说有问题，那就暂时不更新了

WYC-2020, 2023-05-27:
> > > > [image]
> > > 
> > > 
> > > 清理了tmp文件夹还是无法编译成功，还是报一样的错，请问有什么解决方案，或许我有些地方清理的不够彻底？
> > 
> > 
> > 先暂时别用 我细看了一下 虽然编译没问题，但是这是一个空包 [image] 我在研究研究
> 
> 好的，我修改完文件但不会单独编译内核和模块就直接用命令跑编译固件没报错，既然大佬说有问题，那就暂时不更新了
[image]
  如此修复，刚那个修改撤回，就只改这里就行了

commit 0d1a56f7 (2023-05-27): generic: refresh 5.15 kernel config

Enable CONFIG_ARM_CRYPTO on these targets.

Fixes: #11247
closed 2023-05-27 by aiamadeus