coolsnowwolf/lede #11248: 拉取最新源码后Action编译x86固件提示package中内置的smartdns编译失败

bi4tms, 2023-05-27:
6.1/5.15/5.4内核均报错
日志如下：
[image]

WYC-2020, 2023-05-27:
j1

bi4tms, 2023-05-27:
> j1

仔细看了下日志，即使设置成j1，在Download package中也是提升smartdns下载错误

mzl0408, 2023-05-27:
下载的时候就是失败的

WYC-2020, 2023-05-27:
> 下载的时候就是失败的

等会给你们看下，因为我没用这个仓库下的sartdns 我自己用的我的修改版

bi4tms, 2023-05-27:
上一个版本可以正常编译，能否回滚至上个版本
@aiamadeus

WYC-2020, 2023-05-27:
[替换他的makefile.zip](https://github.com/coolsnowwolf/lede/files/11581834/makefile.zip)

bi4tms, 2023-05-27:
> [替换他的makefile.zip](https://github.com/coolsnowwolf/lede/files/11581834/makefile.zip)

在线用Action编译应该怎么搞

WYC-2020, 2023-05-27:
> > [替换他的makefile.zip](https://github.com/coolsnowwolf/lede/files/11581834/makefile.zip)
> 
> 在线用Action编译应该怎么搞

等l那边修复 或者自己修改编译脚本 加入 cp 覆盖就行

closed 2023-05-28 by bi4tms