coolsnowwolf/lede #11249: smartdns怎么也下载不成功，编译失败也是因为这个

mzl0408, 2023-05-27:
make[2] -C feeds/packages/net/phtunnel download
 make[2] -C feeds/packages/net/pptpd download
 make[2] -C feeds/packages/utils/pv download
 make[2] -C feeds/packages/net/qBittorrent-static download
 make[2] -C feeds/packages/net/rclone download
 make[2] -C feeds/packages/net/rclone-ng download
 make[2] -C feeds/packages/net/rclone-webui-react download
 make[2] -C feeds/packages/net/rpcbind download
 make[2] -C feeds/packages/utils/rpcd-mod-lxc download
 make[2] -C feeds/packages/lang/ruby download
 make[2] -C feeds/packages/utils/runc download
 make[2] -C feeds/packages/net/shadowsocks-libev download
 make[2] -C feeds/packages/sound/shairport-sync download
 make[2] -C feeds/packages/net/smartdns download
    ERROR: package/feeds/packages/smartdns failed to build.
make package/download: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
 make[1] target/download
 make[2] -C target/linux download

WYC-2020, 2023-05-27:
j1

bi4tms, 2023-05-27:
> j1

改成j1亦无法成功下载

WYC-2020, 2023-05-27:
[替换他的makefile.zip](https://github.com/coolsnowwolf/lede/files/11581841/makefile.zip)

mzl0408, 2023-05-27:
> [替换他的makefile.zip](https://github.com/coolsnowwolf/lede/files/11581841/makefile.zip)

替换后需要再make download吗

mzl0408, 2023-05-27:
> [替换他的makefile.zip](https://github.com/coolsnowwolf/lede/files/11581841/makefile.zip)

终于搞定了，谢谢

closed 2023-05-28 by aiamadeus