coolsnowwolf/lede #11251: x64编译过程中提示缺少：libphy.ko

ricksuzade-maker, 2023-05-28:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：x64编译过程中提示缺少：libphy.ko

### (2) 路由器型号和固件版本

x86-64

### (3) 详细日志

A：
make[3]: Leaving directory '/home/zhiliang/scripts/lede/package/firmware/linux-firmware'
time: package/firmware/linux-firmware/compile#19.69#5.19#22.96
make[3]: Entering directory '/home/zhiliang/scripts/lede/package/kernel/linux'
mkdir -p /home/zhiliang/scripts/lede/staging_dir/target-x86_64_musl/root-x86/stamp

SHELL= flock /home/zhiliang/scripts/lede/tmp/.root-copy.flock -c 'cp -fpR /home/zhiliang/scripts/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/.pkgdir/kernel/. /home/zhiliang/scripts/lede/staging_dir/target-x86_64_musl/root-x86/'
touch /home/zhiliang/scripts/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.kernel_installed
Package kmod-phylink is missing dependencies for the following libraries:
libphy.ko
make[3]: *** [modules/netdevices.mk:123: /home/zhiliang/scripts/lede/bin/targets/x86/64/packages/kmod-phylink_6.1.30-1_x86_64.ipk] Error 1
make[3]: Leaving directory '/home/zhiliang/scripts/lede/package/kernel/linux'
time: package/kernel/linux/compile#2.48#0.24#2.66
    ERROR: package/kernel/linux failed to build.
make[2]: *** [package/Makefile:116: package/kernel/linux/compile] Error 1
make[2]: Leaving directory '/home/zhiliang/scripts/lede'
make[1]: *** [package/Makefile:110: /home/zhiliang/scripts/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/zhiliang/scripts/lede'
make: *** [/home/zhiliang/scripts/lede/include/toplevel.mk:231: world] Error 2

WYC-2020, 2023-05-28:
最近模块更新导致的 
[image]
自己加回来

ricksuzade-maker, 2023-05-28:
谢谢，我去提pr

closed 2023-05-29 by coolsnowwolf