coolsnowwolf/lede #11255: phicomm K3编译报错

barton0426, 2023-05-29:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：编译中报错，sha1-arm.ko' is missing.

### (2) 路由器型号和固件版本

A：phicomm k3 最新的commit

### (3) 详细日志

A：
install -m0644 /home/runner/work/lede/lede/build_dir/target-arm_cortex-a9_musl_eabi/linux-firmware-20230515/brcm/brcmfmac43143.bin /home/runner/work/lede/lede/build_dir/target-arm_cortex-a9_musl_eabi/linux-firmware-20230515/.pkgdir/brcmfmac-firmware-usb/lib/firmware/brcm/
touch /home/runner/work/lede/lede/build_dir/target-arm_cortex-a9_musl_eabi/linux-firmware-20230515/.pkgdir/brcmfmac-firmware-usb.installed
mkdir -p /home/runner/work/lede/lede/staging_dir/target-arm_cortex-a9_musl_eabi/root-bcm53xx/stamp
SHELL= flock /home/runner/work/lede/lede/tmp/.root-copy.flock -c 'cp -fpR /home/runner/work/lede/lede/build_dir/target-arm_cortex-a9_musl_eabi/linux-firmware-20230515/.pkgdir/brcmfmac-firmware-usb/. /home/runner/work/lede/lede/staging_dir/target-arm_cortex-a9_musl_eabi/root-bcm53xx/'
touch /home/runner/work/lede/lede/staging_dir/target-arm_cortex-a9_musl_eabi/root-bcm53xx/stamp/.brcmfmac-firmware-usb_installed
echo "brcmfmac-firmware-usb" >> /home/runner/work/lede/lede/staging_dir/target-arm_cortex-a9_musl_eabi/pkginfo/linux-firmware.default.install
make[3]: Leaving directory '/home/runner/work/lede/lede/package/firmware/linux-firmware'
time: package/firmware/linux-firmware/compile#16.32#2.73#17.41
make[3]: Entering directory '/home/runner/work/lede/lede/package/kernel/linux'
mkdir -p /home/runner/work/lede/lede/staging_dir/target-arm_cortex-a9_musl_eabi/root-bcm53xx/stamp

SHELL= flock /home/runner/work/lede/lede/tmp/.root-copy.flock -c 'cp -fpR /home/runner/work/lede/lede/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/packages/.pkgdir/kernel/. /home/runner/work/lede/lede/staging_dir/target-arm_cortex-a9_musl_eabi/root-bcm53xx/'
touch /home/runner/work/lede/lede/staging_dir/target-arm_cortex-a9_musl_eabi/root-bcm53xx/stamp/.kernel_installed
ERROR: module '/home/runner/work/lede/lede/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/linux-5.4.243/arch/arm/crypto/sha1-arm.ko' is missing.
make[3]: *** [modules/crypto.mk:917: /home/runner/work/lede/lede/bin/targets/bcm53xx/generic/packages/kmod-crypto-sha1_5.4.243-1_arm_cortex-a9.ipk] Error 1
make[3]: Leaving directory '/home/runner/work/lede/lede/package/kernel/linux'
time: package/kernel/linux/compile#3.32#0.34#3.66
    ERROR: package/kernel/linux failed to build.
make[2]: *** [package/Makefile:116: package/kernel/linux/compile] Error 1
make[2]: Leaving directory '/home/runner/work/lede/lede'
make[1]: Leaving directory '/home/runner/work/lede/lede'
make[1]: *** [package/Makefile:110: /home/runner/work/lede/lede/staging_dir/target-arm_cortex-a9_musl_eabi/stamp/.package_compile] Error 2
make: *** [/home/runner/work/lede/lede/include/toplevel.mk:231: world] Error 2
Error: Process completed with exit code 2.

WYC-2020, 2023-05-29:
[image]
 吧这个加到 config-5.4  5.10 你也可以加

zwillhill, 2023-06-04:
> [image] 吧这个加到 config-5.4 5.10 你也可以加

需要全加？，加了后CONFIG_ARM_CRYPTO=y后云编译测试还是提示crypto/sha1-arm.ko' is missing

WYC-2020, 2023-06-04:
> > [image] 吧这个加到 config-5.4 5.10 你也可以加
> 
> 需要全加？，加了后CONFIG_ARM_CRYPTO=y后云编译测试还是提示crypto/sha1-arm.ko' is missing

服，你改了你都不menuconfig更新下config能不出问题

barton0426, 2023-06-05:
> [image] 吧这个加到 config-5.4 5.10 你也可以加

请问大佬，k3用5.15内核稳定吗 之前用5.10的wan口失效了 我看大雕回退到5.4了

closed 2026-09-05 by coolsnowwolf