coolsnowwolf/lede #11259: 根据系统日志可以查出断网原因吗？

XinSSS, 2023-05-29:
最近两天断了两次，截取日志如下

```
Sun May 28 11:18:27 2023 daemon.info pppd[9071]: LCP terminated by peer
Sun May 28 11:18:27 2023 daemon.info pppd[9071]: Connect time 5760.1 minutes.
Sun May 28 11:18:27 2023 daemon.info pppd[9071]: Sent 3586131155 bytes, received 2021644557 bytes.
Sun May 28 11:18:27 2023 daemon.notice ttyd[6071]: [2023/05/28 11:18:27:1248] N: rops_handle_POLLIN_netlink: DELADDR
Sun May 28 11:18:27 2023 daemon.err miniupnpd[6418]: ioctl(s, SIOCGIFADDR, ...): Address not available
Sun May 28 11:18:27 2023 daemon.err miniupnpd[6418]: Failed to get IP for interface pppoe-wan
Sun May 28 11:18:27 2023 daemon.warn miniupnpd[6418]: SendNATPMPPublicAddressChangeNotification: cannot get public IP address, stopping
Sun May 28 11:18:27 2023 daemon.notice ttyd[6071]: [2023/05/28 11:18:27:1272] N: rops_handle_POLLIN_netlink: DELADDR
Sun May 28 11:18:27 2023 daemon.notice netifd: Network device 'pppoe-wan' link is down
Sun May 28 11:18:27 2023 daemon.notice netifd: Interface 'wan' has lost the connection
Sun May 28 11:18:27 2023 daemon.notice pppd[9071]: Modem hangup
Sun May 28 11:18:27 2023 daemon.notice pppd[9071]: Connection terminated.
Sun May 28 11:18:27 2023 daemon.info pppd[9071]: Connect time 5760.1 minutes.
Sun May 28 11:18:27 2023 daemon.info pppd[9071]: Sent 3586131155 bytes, received 2021644597 bytes.
Sun May 28 11:18:27 2023 daemon.info pppd[9071]: Exit.
Sun May 28 11:18:27 2023 daemon.notice netifd: Interface 'wan' is now down
Sun May 28 11:18:27 2023 daemon.notice netifd: Interface 'wan' is disabled
Sun May 28 11:18:27 2023 daemon.notice netifd: Interface 'Modem' is disabled
Sun May 28 11:18:27 2023 daemon.notice netifd: Interface 'wan' is enabled
Sun May 28 11:18:27 2023 daemon.notice netifd: Interface 'wan' is setting up now
Sun May 28 11:18:27 2023 daemon.notice netifd: Interface 'Modem' is enabled
Sun May 28 11:18:27 2023 daemon.notice netifd: Network device 'eth0' link is down
Sun May 28 11:18:27 2023 daemon.notice netifd: Interface 'wan' has link connectivity loss
Sun May 28 11:18:27 2023 daemon.notice netifd: Interface 'Modem' has link connectivity loss
Sun May 28 11:18:27 2023 kern.info kernel: [4200240.106397] 8021q: adding VLAN 0 to HW filter on device eth0
Sun May 28 11:18:27 2023 daemon.notice netifd: wan (23467): uci: Entry not found
Sun May 28 11:18:27 2023 daemon.notice netifd: wan (23467): sh: out of range
```

wuzhuang123, 2023-05-29:
没啥用。一会启用一会禁用的。

XinSSS, 2023-05-29:
好吧

WYC-2020, 2023-05-30:
你这玩意确实断的很频繁，应该是有什么问题，我这边也会断，但是是运营商那边强制断的。因为间隔是固定的两天

guolongchu, 2023-06-05:
有可能是网卡驱动的故障！

closed 2026-09-05 by coolsnowwolf