coolsnowwolf/lede #11260: 编译mt76出错

xiangfeidexiaohuo, 2023-05-30:
Revert "mt76: drop upstream patches" 这个提交开始 居然x86  5.4编译mt76出错

2023-05-30T00:14:48.9080126Z make[3]: Entering directory '/home/runner/work/OpenWrt_Build/OpenWrt_Build/openwrt/package/kernel/mt76'
2023-05-30T00:14:48.9113790Z [26/45] Building C object CMakeFiles/init.dir/sysupgrade.c.o
2023-05-30T00:14:48.9573963Z [27/45] Building C object CMakeFiles/init.dir/watchdog.c.o
2023-05-30T00:14:49.0022810Z [28/45] Building C object CMakeFiles/init.dir/utils/utils.c.o
2023-05-30T00:14:49.1789952Z touch /home/runner/work/OpenWrt_Build/OpenWrt_Build/openwrt/build_dir/target-x86_64_musl/linux-x86_64/mt76-2022-12-22-5b509e80/.prepared_aab6f640434b82ab64d41378f7d1e147_6664517399ebbbc92a37c5bb081b5c53_check
2023-05-30T00:14:49.1813949Z . /home/runner/work/OpenWrt_Build/OpenWrt_Build/openwrt/include/shell.sh; xzcat /home/runner/work/OpenWrt_Build/OpenWrt_Build/openwrt/dl/mt76-2022-12-22-5b509e80.tar.xz | tar -C /home/runner/work/OpenWrt_Build/OpenWrt_Build/openwrt/build_dir/target-x86_64_musl/linux-x86_64/mt76-2022-12-22-5b509e80/.. -xf -
2023-05-30T00:14:49.1850045Z xzcat: /home/runner/work/OpenWrt_Build/OpenWrt_Build/openwrt/dl/mt76-2022-12-22-5b509e80.tar.xz: No such file or directory
2023-05-30T00:14:49.1851046Z tar: This does not look like a tar archive
2023-05-30T00:14:49.1892076Z make[3]: Leaving directory '/home/runner/work/OpenWrt_Build/OpenWrt_Build/openwrt/package/kernel/mt76'
2023-05-30T00:14:49.1892533Z tar: Exiting with failure status due to previous errors
2023-05-30T00:14:49.1892856Z time: package/kernel/mt76/compile#0.20#0.06#0.28
2023-05-30T00:14:49.1893152Z     ERROR: package/kernel/mt76 failed to build.
2023-05-30T00:14:49.1927516Z [29/45] Linking C executable init
2023-05-30T00:14:49.1973184Z make[3]: *** [Makefile:564: /home/runner/work/OpenWrt_Build/OpenWrt_Build/openwrt/build_dir/target-x86_64_musl/linux-x86_64/mt76-2022-12-22-5b509e80/.prepared_aab6f640434b82ab64d41378f7d1e147_6664517399ebbbc92a37c5bb081b5c53] Error 2
2023-05-30T00:14:49.1975970Z make[2]: *** [package/Makefile:116: package/kernel/mt76/compile] Error 1

WYC-2020, 2023-05-30:
你这个连mt包都下载失败了，和这个提交没关系吧，自己重新下载mt的包

xiangfeidexiaohuo, 2023-05-30:
> 你这个连mt包都下载失败了，和这个提交没关系吧，自己重新下载mt的包

https://github.com/coolsnowwolf/lede/pull/11262/commits/7fadb02903a22e1e895488a2cd7c969b226abd31

closed 2023-05-30 by xiangfeidexiaohuo