coolsnowwolf/lede #11266: R5S 线刷官方系统进emmc后在网页端升级到lede固件，运行一段时间后mmc会挂掉

FantasyGmm, 2023-05-31:
Nano Pi R5S 2204硬件版本

当日最新commit进行编译产生的系统，取消了部分默认选择的插件，手动添加了luci-smartdns，后直接编译的

emmc cache开始报错后，然后就无法进行读写了，接着WebUI无法使用了，但是并不影响联网和插件的运行（smartdns与qb均能使用）

同一个镜像直接刷入tf卡可以正常运行，并不会死掉，emmc会在开机后过一段时间后死掉，每次都能复现

使用badblocks 对emmc进行过全盘读写测试，没有坏块，但是有插入nvme硬盘，emmc在nvme下面，不知道是否是散热问题导致的掉盘

内核报错日志
```
[ 1309.948958] mmc1: cache flush error -110
[ 1309.989234] mmc1: tried to HW reset card, got error -110
[ 1309.990019] mmcblk1: recovery failed!
[ 1309.990466] blk_update_request: I/O error, dev mmcblk1, sector 14143376 op 0x1:(WRITE) flags 0x800 phys_seg 8 prio class 0
[ 1309.991571] Aborting journal on device mmcblk1p3-8.
[ 1309.994116] blk_update_request: I/O error, dev mmcblk1, sector 14098456 op 0x1:(WRITE) flags 0x20800 phys_seg 1 prio class 0
[ 1309.995322] Buffer I/O error on dev mmcblk1p3, logical block 1606659, lost sync page write
[ 1309.996090] JBD2: Error -5 detected when updating journal superblock for mmcblk1p3-8.
[ 1320.191030] EXT4-fs error (device mmcblk1p3): ext4_journal_check_start:83: comm luci: Detected aborted journal
[ 1320.193677] blk_update_request: I/O error, dev mmcblk1, sector 1245184 op 0x1:(WRITE) flags 0x23800 phys_seg 1 prio class 0
[ 1320.194659] Buffer I/O error on dev mmcblk1p3, logical block 0, lost sync page write
[ 1320.195358] EXT4-fs (mmcblk1p3): I/O error while writing superblock
[ 1320.195935] EXT4-fs (mmcblk1p3): Remounting filesystem read-only
[ 2922.454375] blk_update_request: I/O error, dev mmcblk1, sector 478526 op 0x0:(READ) flags 0x800 phys_seg 1 prio class 0
[ 2922.455365] SQUASHFS error: Failed to read block 0x89a7c7e: -5
[ 2922.455877] SQUASHFS error: Unable to read metadata cache entry [89a7c7e]
[ 2922.456471] SQUASHFS error: Unable to read directory block [89a6ca8:1d71]
[ 2922.458707] blk_update_request: I/O error, dev mmcblk1, sector 478600 op 0x0:(READ) flags 0x800 phys_seg 1 prio class 0
[ 2922.459661] SQUASHFS error: Failed to read block 0x89b1096: -5
[ 2922.460173] SQUASHFS error: Unable to read metadata cache entry [89b1096]
[ 2922.460765] SQUASHFS error: Unable to read directory block [89b1096:ea0]
[ 2922.462962] blk_update_request: I/O error, dev mmcblk1, sector 478494 op 0x0:(READ) flags 0x800 phys_seg 1 prio class 0
[ 2922.464006] SQUASHFS error: Failed to read block 0x89a3c46: -5
[ 2922.464540] SQUASHFS error: Unable to read metadata cache entry [89a3c46]
[ 2922.465170] SQUASHFS error: Unable to read directory block [89a3c46:0]
[ 2922.465803] SQUASHFS error: Unable to read metadata cache entry [89a7c7e]
[ 2922.466419] SQUASHFS error: Unable to read directory block [89a7c7e:298]
[ 2922.483441] SQUASHFS error: Unable to read metadata cache entry [89a3c46]
[ 2922.484062] SQUASHFS error: Unable to read directory block [89a3c46:486]
[ 2922.484651] SQUASHFS error: Unable to read metadata cache entry [89a3c46]
[ 2922.485280] SQUASHFS error: Unable to read directory block [89a3c46:486]
[ 2922.521011] blk_update_request: I/O error, dev mmcblk1, sector 478634 op 0x0:(READ) flags 0x800 phys_seg 1 prio class 0
[ 2922.522025] SQUASHFS error: Failed to read block 0x89b57d2: -5
[ 2922.522545] SQUASHFS error: Unable to read metadata cache entry [89b57d2]
[ 2922.523141] SQUASHFS error: Unable to read directory block [89b57d2:19e3]
[ 2922.523779] SQUASHFS error: Unable to read metadata cache entry [89b57d2]
[ 2922.524380] SQUASHFS error: Unable to read directory block [89b57d2:19e3]
[ 2922.524993] SQUASHFS error: Unable to read metadata cache entry [89b57d2]
[ 2922.525619] SQUASHFS error: Unable to read directory block [89b57d2:19e3]
[ 2922.526233] SQUASHFS error: Unable to read metadata cache entry [89b57d2]
[ 2922.526828] SQUASHFS error: Unable to read directory block [89b57d2:19e3]
[ 2922.613348] SQUASHFS error: Unable to read metadata cache entry [89b57d2]
[ 2922.613967] SQUASHFS error: Unable to read directory block [89b57d2:19e3]
[ 2922.614595] SQUASHFS error: Unable to read metadata cache entry [89b57d2]
[ 2922.615326] SQUASHFS error: Unable to read directory block [89b57d2:19e3]
[ 2922.616116] SQUASHFS error: Unable to read metadata cache entry [89b57d2]
[ 2922.616730] SQUASHFS error: Unable to read directory block [89b57d2:19e3]
[ 2922.618498] SQUASHFS error: Unable to read metadata cache entry [89b57d2]
[ 2922.619565] SQUASHFS error: Unable to read directory block [89b57d2:19e3]
[ 2922.677689] SQUASHFS error: Unable to read metadata cache entry [89b57d2]
[ 2922.678444] SQUASHFS error: Unable to read directory block [89b57d2:19e3]
[ 2922.679165] SQUASHFS error: Unable to read metadata cache entry [89b57d2]
[ 2922.679803] SQUASHFS error: Unable to read directory block [89b57d2:19e3]
[ 2922.680425] blk_update_request: I/O error, dev mmcblk1, sector 478606 op 0x0:(READ) flags 0x800 phys_seg 1 prio class 0
[ 2922.682329] SQUASHFS error: Unable to read metadata cache entry [89b57d2]
[ 2922.682939] SQUASHFS error: Unable to read directory block [89b57d2:19e3]
[ 2922.683573] SQUASHFS error: Unable to read metadata cache entry [89b57d2]
[ 2922.684176] SQUASHFS error: Unable to read directory block [89b57d2:19e3]
[... 235 lines cut ...]
[ 2924.145416] SQUASHFS error: Unable to read metadata cache entry [89b57d2]
[ 2924.146022] SQUASHFS error: Unable to read directory block [89b57d2:19e3]
[ 2924.200545] SQUASHFS error: Unable to read metadata cache entry [89b57d2]
[ 2924.201206] SQUASHFS error: Unable to read directory block [89b57d2:19e3]
[ 2924.201845] SQUASHFS error: Unable to read metadata cache entry [89b57d2]
[ 2924.202449] SQUASHFS error: Unable to read directory block [89b57d2:19e3]
[ 2924.203069] SQUASHFS error: Unable to read metadata cache entry [89b57d2]
[ 2924.203669] SQUASHFS error: Unable to read directory block [89b57d2:19e3]
[ 2924.204288] SQUASHFS error: Unable to read metadata cache entry [89b57d2]
[ 2924.204890] SQUASHFS error: Unable to read directory block [89b57d2:19e3]
[ 2924.260718] SQUASHFS error: Unable to read metadata cache entry [89b57d2]
[ 2924.261383] SQUASHFS error: Unable to read directory block [89b57d2:19e3]
[ 2924.262024] SQUASHFS error: Unable to read metadata cache entry [89b57d2]
[ 2924.262626] SQUASHFS error: Unable to read directory block [89b57d2:19e3]
[ 2924.263249] SQUASHFS error: Unable to read metadata cache entry [89b57d2]
[ 2924.263941] SQUASHFS error: Unable to read directory block [89b57d2:19e3]
[ 2924.264574] SQUASHFS error: Unable to read metadata cache entry [89b57d2]
[ 2924.265231] SQUASHFS error: Unable to read directory block [89b57d2:19e3]
[ 2924.321305] SQUASHFS error: Unable to read metadata cache entry [89b57d2]
[ 2924.321926] SQUASHFS error: Unable to read directory block [89b57d2:19e3]
[ 2924.322559] SQUASHFS error: Unable to read metadata cache entry [89b57d2]
[ 2924.323161] SQUASHFS error: Unable to read directory block [89b57d2:19e3]
[ 2924.323781] SQUASHFS error: Unable to read metadata cache entry [89b57d2]
[ 2924.324381] SQUASHFS error: Unable to read directory block [89b57d2:19e3]
```

jcoseql, 2023-05-31:
友善官方推荐用SD版本固件自带的EMMC刷机工具刷第三方固件

Wilson-coder, 2023-06-01:
和软件的关系可能并不大，大概率是你这个路由器自身的EMMC芯片质量出问题了。我路由里面一个4T的固态硬盘，彻底挂掉之前也是从I/O错误开始的。

FantasyGmm, 2023-06-01:
> 和软件的关系可能并不大，大概率是你这个路由器自身的EMMC芯片质量出问题了。我路由里面一个4T的固态硬盘，彻底挂掉之前也是从I/O错误开始的。

之前全盘扫过，读写测试都进行了也没出现问题

coolsnowwolf, 2023-06-01:
先用 usb emmc 的 image，用 rk 的usb 线刷工具选择清零下

rk 的 emmc 有时候写入前没有完全擦除导致的

或者 dd 的时候加入 conv=fdatasync 参数即可避免

FantasyGmm, 2023-06-01:
> 先用 usb emmc 的 image，用 rk 的usb 线刷工具选择清零下
> 
> rk 的 emmc 有时候写入前没有完全擦除导致的
> 
> 或者 dd 的时候加入 conv=fsync 参数即可避免

友善提供的usb emmc包内的rkdev没办法执行擦除emmc，目前刷入usb emmc内的系统后接了个u盘 把镜像dd进去了 并带了上面的参数 剩余容量新建了分区并格式化ext4了

FantasyGmm, 2023-06-03:
换成其他的OP进行了测试，友善或者imm的就不会出现上面的问题，imm稳定运行了一个小时了，lede则是会随机崩溃，有时候二十多分钟就会崩，有时候四十多分钟会崩

WYC-2020, 2023-06-05:
> 换成其他的OP进行了测试，友善或者imm的就不会出现上面的问题，imm稳定运行了一个小时了，lede则是会随机崩溃，有时候二十多分钟就会崩，有时候四十多分钟会崩

哈哈可怜的孩子 应该是最近某次提交导致，你可以试试 回滚到一个月 或者半个月 再试试

FantasyGmm, 2023-06-05:
> > 换成其他的OP进行了测试，友善或者imm的就不会出现上面的问题，imm稳定运行了一个小时了，lede则是会随机崩溃，有时候二十多分钟就会崩，有时候四十多分钟会崩
> 
> 哈哈可怜的孩子 应该是最近某次提交导致，你可以试试 回滚到一个月 或者半个月 再试试

大概一个月前也有试过刷进emmc，然后当时也是出现一样的情况，后来就临时用tf卡用了，现在tf卡需要换到其他sbc上用了，所以得用回emmc了

WYC-2020, 2023-06-05:
> blk_update_request

那看来是l这边有什么bug没有修复，或者少了某个patch

FantasyGmm, 2023-06-05:
> 

imm那边用的6.1的内核了我也没办法对比，至于友善那边我甚至连设备树在哪里都没找到

jcoseql, 2023-06-06:
> > 先用 usb emmc 的 image，用 rk 的usb 线刷工具选择清零下
> > rk 的 emmc 有时候写入前没有完全擦除导致的
> > 或者 dd 的时候加入 conv=fsync 参数即可避免
> 
> 友善提供的usb emmc包内的rkdev没办法执行擦除emmc，目前刷入usb emmc内的系统后接了个u盘，把镜像dd到emmc并带上conv=fsync参数，reboot重启，成功刷入lede了, 剩余容量新建了分区并格式化ext4了
> 
> 启动后关于mmc输出了这些
> 
> ```
> root@OpenWrt:~# dmesg |grep mmc
> [    1.353564] dwmmc_rockchip fe2b0000.mmc: IDMAC supports 32-bit address mode.
> [    1.354211] dwmmc_rockchip fe2b0000.mmc: Using internal DMA controller.
> [    1.354800] dwmmc_rockchip fe2b0000.mmc: Version ID is 270a
> [    1.355318] dwmmc_rockchip fe2b0000.mmc: DW MMC controller at irq 30,32 bit host data width,256 deep fifo
> [    1.356615] dwmmc_rockchip fe2b0000.mmc: Got CD GPIO
> [    1.369596] mmc_host mmc0: Bus speed (slot 0) = 375000Hz (slot req 400000Hz, actual 375000HZ div = 0)
> [    1.382992] mmc1: SDHCI controller on fe310000.mmc [fe310000.mmc] using ADMA
> [    1.453366] mmc1: new HS200 MMC card at address 0001
> [    1.454394] mmcblk1: mmc1:0001 AJTD4R 14.6 GiB
> [    1.455970]  mmcblk1: p1 p2
> [    1.456652] mmcblk1boot0: mmc1:0001 AJTD4R 4.00 MiB
> [    1.457913] mmcblk1boot1: mmc1:0001 AJTD4R 4.00 MiB
> [    1.458923] mmcblk1rpmb: mmc1:0001 AJTD4R 4.00 MiB, chardev (248:0)
> [   18.089871] EXT4-fs (mmcblk1p1): warning: mounting unchecked fs, running e2fsck is recommended
> [   18.091308] EXT4-fs (mmcblk1p1): mounted filesystem without journal. Opts: (null). Quota mode: none.
> [   22.650144] EXT4-fs (mmcblk1p1): mounted filesystem without journal. Opts: . Quota mode: none.
> ```
> 
> 还是崩溃了
> 
> ```
> [ 4877.265507] mmc1: cache flush error -110
> [ 4877.308495] mmc1: tried to HW reset card, got error -110
> [ 4877.308979] mmcblk1: recovery failed!
> [ 4877.309412] blk_update_request: I/O error, dev mmcblk1, sector 418046 op 0x0:(READ) flags 0x800 phys_seg 70 prio class 0
> [ 4877.310463] SQUASHFS error: Failed to read block 0x6c1ff46: -5
> [ 4877.311015] SQUASHFS error: Unable to read fragment cache entry [6c1ff46]
> [ 4877.311618] SQUASHFS error: Unable to read page, block 6c1ff46, size 44728
> [ 4877.312225] SQUASHFS error: Unable to read fragment cache entry [6c1ff46]
> [ 4877.312817] SQUASHFS error: Unable to read page, block 6c1ff46, size 44728
> [ 4877.313419] SQUASHFS error: Unable to read fragment cache entry [6c1ff46]
> [ 4877.314010] SQUASHFS error: Unable to read page, block 6c1ff46, size 44728
> [ 4877.314612] SQUASHFS error: Unable to read fragment cache entry [6c1ff46]
> [ 4877.315216] SQUASHFS error: Unable to read page, block 6c1ff46, size 44728
> [ 4877.315824] SQUASHFS error: Unable to read fragment cache entry [6c1ff46]
> [ 4877.316416] SQUASHFS error: Unable to read page, block 6c1ff46, size 44728
> [ 4877.317038] SQUASHFS error: Unable to read fragment cache entry [6c1ff46]
> [ 4877.317632] SQUASHFS error: Unable to read page, block 6c1ff46, size 44728
> [ 4877.318241] SQUASHFS error: Unable to read fragment cache entry [6c1ff46]
> [ 4877.318833] SQUASHFS error: Unable to read page, block 6c1ff46, size 44728
> [ 4879.215383] SQUASHFS error: Unable to read fragment cache entry [6c1ff46]
> [ 4879.216001] SQUASHFS error: Unable to read page, block 6c1ff46, size 44728
> [ 4879.216616] SQUASHFS error: Unable to read fragment cache entry [6c1ff46]
> [ 4879.217210] SQUASHFS error: Unable to read page, block 6c1ff46, size 44728
> [ 4879.217816] SQUASHFS error: Unable to read fragment cache entry [6c1ff46]
> [ 4879.218410] SQUASHFS error: Unable to read page, block 6c1ff46, size 44728
> [ 4879.888999] SQUASHFS error: Unable to read fragment cache entry [6c1ff46]
> [ 4879.889617] SQUASHFS error: Unable to read page, block 6c1ff46, size 44728
> [ 4879.890231] SQUASHFS error: Unable to read fragment cache entry [6c1ff46]
> [ 4879.890825] SQUASHFS error: Unable to read page, block 6c1ff46, size 44728
> [ 4879.891487] SQUASHFS error: Unable to read fragment cache entry [6c1ff46]
> [ 4879.892082] SQUASHFS error: Unable to read page, block 6c1ff46, size 44728
> [ 4880.642407] SQUASHFS error: Unable to read fragment cache entry [6c1ff46]
> [ 4880.643027] SQUASHFS error: Unable to read page, block 6c1ff46, size 44728
> [ 4880.643702] SQUASHFS error: Unable to read fragment cache entry [6c1ff46]
> [ 4880.644301] SQUASHFS error: Unable to read page, block 6c1ff46, size 44728
> [ 4880.644908] SQUASHFS error: Unable to read fragment cache entry [6c1ff46]
> [ 4880.645502] SQUASHFS error: Unable to read page, block 6c1ff46, size 44728
> [ 4881.290990] SQUASHFS error: Unable to read fragment cache entry [6c1ff46]
> [ 4881.291646] SQUASHFS error: Unable to read page, block 6c1ff46, size 44728
> [ 4881.292260] SQUASHFS error: Unable to read fragment cache entry [6c1ff46]
> [ 4881.292854] SQUASHFS error: Unable to read page, block 6c1ff46, size 44728
> [ 4881.293461] SQUASHFS error: Unable to read fragment cache entry [6c1ff46]
> [ 4881.294053] SQUASHFS error: Unable to read page, block 6c1ff46, size 44728
> ```

如果你还没擦除过emmc的话，我还是建议你试试线刷包配合rkdev擦除emmc。线刷包可以在releases里找R22.11.13，我每次折腾前都用它。然后用友善SD版本固件自带的EMMC写入工具

coolsnowwolf, 2023-06-08:
上游代码的bug，已经从 6.4 内核种 backport 移植过去解决

closed 2023-06-27 by FantasyGmm