coolsnowwolf/lede #11269: 均衡负载（mwan3）无法使用

fxxccp, 2023-06-01:
## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

均衡负载（mwan3）已经好几个月不能使用了，我不是单线多拨。我有两部宽带叠加带宽，分别是wan1、wan2，表现为只能使用wan1。去年年底编译的正常使用，目前已完全废了。

### (2) 路由器型号和固件版本

A：x86，5.15.112

### (3) 详细日志

Thu Jun  1 09:34:51 2023 user.info mwan3track[4808]: Check (ping) success for target "114.114.114.114" on interface wan1 (pppoe-wan1)
Thu Jun  1 09:34:51 2023 user.info mwan3track[4808]: Lost 162 ping(s) on interface wan1 (pppoe-wan1)
Thu Jun  1 09:47:42 2023 user.info mwan3track[4808]: Check (ping) success for target "114.114.114.114" on interface wan1 (pppoe-wan1)
Thu Jun  1 09:47:42 2023 user.info mwan3track[4808]: Lost 109 ping(s) on interface wan1 (pppoe-wan1)
Thu Jun  1 09:51:53 2023 user.info mwan3track[4808]: Check (ping) success for target "114.114.114.114" on interface wan1 (pppoe-wan1)
Thu Jun  1 09:51:53 2023 user.info mwan3track[4808]: Lost 35 ping(s) on interface wan1 (pppoe-wan1)
Thu Jun  1 10:00:24 2023 user.info mwan3track[4808]: Check (ping) success for target "114.114.114.114" on interface wan1 (pppoe-wan1)
Thu Jun  1 10:00:24 2023 user.info mwan3track[4808]: Lost 72 ping(s) on interface wan1 (pppoe-wan1)
Thu Jun  1 10:04:35 2023 user.info mwan3track[4808]: Check (ping) success for target "114.114.114.114" on interface wan1 (pppoe-wan1)
Thu Jun  1 10:04:35 2023 user.info mwan3track[4808]: Lost 35 ping(s) on interface wan1 (pppoe-wan1)
Thu Jun  1 10:08:46 2023 user.info mwan3track[4808]: Check (ping) success for target "114.114.114.114" on interface wan1 (pppoe-wan1)
Thu Jun  1 10:08:46 2023 user.info mwan3track[4808]: Lost 35 ping(s) on interface wan1 (pppoe-wan1)
Thu Jun  1 10:17:17 2023 user.info mwan3track[4808]: Check (ping) success for target "114.114.114.114" on interface wan1 (pppoe-wan1)
Thu Jun  1 10:17:17 2023 user.info mwan3track[4808]: Lost 72 ping(s) on interface wan1 (pppoe-wan1)
Thu Jun  1 10:45:41 2023 daemon.err uhttpd[1887]: cat: can't open '/etc/bench.log': No such file or directory
Thu Jun  1 10:45:41 2023 daemon.err uhttpd[1887]: cat: can't open '/etc/bench.log': No such file or directory
Thu Jun  1 10:47:35 2023 user.info mwan3track[4808]: Check (ping) success for target "114.114.114.114" on interface wan1 (pppoe-wan1)
Thu Jun  1 10:47:35 2023 user.info mwan3track[4808]: Lost 258 ping(s) on interface wan1 (pppoe-wan1)

WYC-2020, 2023-06-01:
这个没测试条件，这个问题好像以前也有这个issue

alexliu269, 2023-06-10:
跃点都设置成1 设置完以后 不要再碰设置了 直接重启 查看防火墙是否加载成功 负载均衡状态是否能看到负载百分比

closed 2026-09-05 by coolsnowwolf