coolsnowwolf/lede #11271: pppoe无法拨号，麻烦大佬看下

kernel-my, 2023-06-02:
各位大佬帮忙看一下，我用pve跑openwrt-x64-R23.4.1这个版本的openwrt，pppoe拨号无法拨号，换成DHCP就可以呢，帮忙看一下是不是哪里设置的有问题，日志如下：

Thu Jun  1 23:53:43 2023 daemon.notice procd: /etc/rc.d/S13openssl: Enabling engine devcrypto
Thu Jun  1 23:53:43 2023 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Thu Jun  1 23:53:43 2023 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Thu Jun  1 23:53:44 2023 daemon.notice procd: /etc/rc.d/S50autoreboot: Command failed: ubus call service delete { "name": "cron" } (Not found)
Thu Jun  1 23:53:44 2023 user.notice : Added device handler type: 8021ad
Thu Jun  1 23:53:44 2023 user.notice : Added device handler type: 8021q
Thu Jun  1 23:53:44 2023 user.notice : Added device handler type: macvlan
Thu Jun  1 23:53:44 2023 user.notice : Added device handler type: veth
Thu Jun  1 23:53:44 2023 user.notice : Added device handler type: bridge
Thu Jun  1 23:53:44 2023 user.notice : Added device handler type: Network device
Thu Jun  1 23:53:44 2023 user.notice : Added device handler type: tunnel
Thu Jun  1 23:53:44 2023 daemon.notice procd: /etc/rc.d/S50autoreboot: Auto REBOOT has started.
Thu Jun  1 23:53:44 2023 cron.err crond[1687]: crond (busybox 1.36.0) started, log level 5
Thu Jun  1 23:53:44 2023 daemon.notice netifd: Interface 'lan' is enabled
Thu Jun  1 23:53:44 2023 daemon.notice netifd: Interface 'lan' is setting up now
Thu Jun  1 23:53:44 2023 daemon.notice netifd: Interface 'lan' is now up
Thu Jun  1 23:53:44 2023 kern.info kernel: [   15.700288] 8021q: adding VLAN 0 to HW filter on device eth0
Thu Jun  1 23:53:44 2023 kern.info kernel: [   15.700805] br-lan: port 1(eth0) entered blocking state
Thu Jun  1 23:53:44 2023 kern.info kernel: [   15.701150] br-lan: port 1(eth0) entered disabled state
Thu Jun  1 23:53:44 2023 kern.info kernel: [   15.701474] device eth0 entered promiscuous mode
Thu Jun  1 23:53:44 2023 kern.info kernel: [   15.701922] br-lan: port 1(eth0) entered blocking state
Thu Jun  1 23:53:44 2023 kern.info kernel: [   15.702215] br-lan: port 1(eth0) entered forwarding state
Thu Jun  1 23:53:44 2023 daemon.notice netifd: bridge 'br-lan' link is up
Thu Jun  1 23:53:44 2023 daemon.notice netifd: Interface 'lan' has link connectivity
Thu Jun  1 23:53:44 2023 daemon.notice netifd: Interface 'loopback' is enabled
Thu Jun  1 23:53:44 2023 daemon.notice netifd: Interface 'loopback' is setting up now
Thu Jun  1 23:53:44 2023 daemon.notice netifd: Interface 'loopback' is now up
Thu Jun  1 23:53:44 2023 daemon.notice netifd: Interface 'wan' is enabled
Thu Jun  1 23:53:44 2023 daemon.notice netifd: Network device 'eth0' link is up
Thu Jun  1 23:53:44 2023 daemon.notice netifd: Network device 'eth2' link is up
Thu Jun  1 23:53:44 2023 daemon.notice netifd: Network device 'eth3' link is up
Thu Jun  1 23:53:44 2023 daemon.notice netifd: Network device 'lo' link is up
Thu Jun  1 23:53:44 2023 daemon.notice netifd: Interface 'loopback' has link connectivity
Thu Jun  1 23:53:44 2023 daemon.notice netifd: Network device 'eth1' link is up
Thu Jun  1 23:53:44 2023 daemon.notice netifd: Interface 'wan' has link connectivity
Thu Jun  1 23:53:44 2023 daemon.notice netifd: Interface 'wan' is setting up now
Thu Jun  1 23:53:44 2023 kern.info kernel: [   15.705077] 8021q: adding VLAN 0 to HW filter on device eth2
Thu Jun  1 23:53:44 2023 kern.info kernel: [   15.705931] br-lan: port 2(eth2) entered blocking state
Thu Jun  1 23:53:44 2023 kern.info kernel: [   15.706226] br-lan: port 2(eth2) entered disabled state
Thu Jun  1 23:53:44 2023 kern.info kernel: [   15.706542] device eth2 entered promiscuous mode
Thu Jun  1 23:53:44 2023 kern.info kernel: [   15.706821] br-lan: port 2(eth2) entered blocking state
Thu Jun  1 23:53:44 2023 kern.info kernel: [   15.707101] br-lan: port 2(eth2) entered forwarding state
Thu Jun  1 23:53:44 2023 kern.info kernel: [   15.708922] 8021q: adding VLAN 0 to HW filter on device eth3
Thu Jun  1 23:53:44 2023 kern.info kernel: [   15.709476] br-lan: port 3(eth3) entered blocking state
Thu Jun  1 23:53:44 2023 kern.info kernel: [   15.709765] br-lan: port 3(eth3) entered disabled state
Thu Jun  1 23:53:44 2023 kern.info kernel: [   15.710076] device eth3 entered promiscuous mode
Thu Jun  1 23:53:44 2023 kern.info kernel: [   15.710363] br-lan: port 3(eth3) entered blocking state
Thu Jun  1 23:53:44 2023 kern.info kernel: [   15.710640] br-lan: port 3(eth3) entered forwarding state
Thu Jun  1 23:53:44 2023 kern.info kernel: [   15.712534] 8021q: adding VLAN 0 to HW filter on device eth1
Thu Jun  1 23:53:44 2023 daemon.notice netifd: wan (1741): uci: Entry not found
Thu Jun  1 23:53:44 2023 daemon.notice netifd: wan (1741): sh: out of range
Thu Jun  1 23:53:44 2023 daemon.info pppd[1763]: Plugin pppoe.so loaded.
Thu Jun  1 23:53:44 2023 daemon.info pppd[1763]: PPPoE plugin from pppd 2.4.9
Thu Jun  1 23:53:44 2023 daemon.notice pppd[1763]: pppd 2.4.9 started by root, uid 0
Thu Jun  1 23:53:44 2023 daemon.info pppd[1763]: PPP session is 8592
Thu Jun  1 23:53:44 2023 daemon.warn pppd[1763]: Connected to 40:ee:dd:da:68:f0 via interface eth1
Thu Jun  1 23:53:44 2023 kern.info kernel: [   15.914554] pppoe-wan: renamed from ppp0
Thu Jun  1 23:53:44 2023 daemon.info pppd[1763]: Renamed interface ppp0 to pppoe-wan
Thu Jun  1 23:53:44 2023 daemon.info pppd[1763]: Using interface pppoe-wan
Thu Jun  1 23:53:44 2023 daemon.notice pppd[1763]: Connect: pppoe-wan <--> eth1
Thu Jun  1 23:53:44 2023 daemon.info pppd[1763]: syncppp not active
Thu Jun  1 23:53:44 2023 daemon.info avahi-daemon[2048]: Found user 'nobody' (UID 65534) and group 'nogroup' (GID 65534).
Thu Jun  1 23:53:44 2023 daemon.info avahi-daemon[2048]: Successfully dropped root privileges.
Thu Jun  1 23:53:44 2023 daemon.info avahi-daemon[2048]: avahi-daemon 0.8 starting up.
Thu Jun  1 23:53:44 2023 daemon.warn avahi-daemon[2048]: WARNING: No NSS support for mDNS detected, consider installing nss-mdns!
Thu Jun  1 23:53:44 2023 daemon.info avahi-daemon[2048]: No service file found in /etc/avahi/services.
Thu Jun  1 23:53:44 2023 user.notice ucitrack: Setting up /etc/config/network reload dependency on /etc/config/dhcp
Thu Jun  1 23:53:44 2023 daemon.info avahi-daemon[2048]: Joining mDNS multicast group on interface br-lan.IPv6 with address fd36:a087:f11a::1.
[... 307 lines cut ...]
Thu Jun  1 23:54:36 2023 daemon.info avahi-daemon[2048]: Interface eth1.IPv6 no longer relevant for mDNS.
Thu Jun  1 23:54:36 2023 daemon.info avahi-daemon[2048]: Leaving mDNS multicast group on interface eth1.IPv6 with address fe80::7c9c:72ff:fec4:642f.
Thu Jun  1 23:54:36 2023 daemon.notice netifd: Interface 'wan' is disabled
Thu Jun  1 23:54:36 2023 daemon.notice netifd: Interface 'wan' is enabled
Thu Jun  1 23:54:36 2023 daemon.notice netifd: Interface 'wan' is setting up now
Thu Jun  1 23:54:36 2023 daemon.info avahi-daemon[2048]: Withdrawing address record for fe80::7c9c:72ff:fec4:642f on eth1.
Thu Jun  1 23:54:36 2023 kern.info kernel: [   67.839761] 8021q: adding VLAN 0 to HW filter on device eth1
Thu Jun  1 23:54:36 2023 daemon.notice netifd: wan (6750): uci: Entry not found
Thu Jun  1 23:54:36 2023 daemon.notice netifd: wan (6750): sh: out of range
Thu Jun  1 23:54:36 2023 daemon.info pppd[6764]: Plugin pppoe.so loaded.
Thu Jun  1 23:54:36 2023 daemon.info pppd[6764]: PPPoE plugin from pppd 2.4.9
Thu Jun  1 23:54:36 2023 daemon.notice pppd[6764]: pppd 2.4.9 started by root, uid 0
Thu Jun  1 23:54:38 2023 daemon.info avahi-daemon[2048]: Joining mDNS multicast group on interface eth1.IPv6 with address fe80::7c9c:72ff:fec4:642f.
Thu Jun  1 23:54:38 2023 daemon.info avahi-daemon[2048]: New relevant interface eth1.IPv6 for mDNS.
Thu Jun  1 23:54:38 2023 daemon.info avahi-daemon[2048]: Registering new address record for fe80::7c9c:72ff:fec4:642f on eth1.*.
Thu Jun  1 23:54:41 2023 kern.err kernel: [   73.238580] squashfs: Unknown parameter 'discard'
Thu Jun  1 23:54:41 2023 kern.info kernel: [   73.239883] F2FS-fs (sda128): Magic Mismatch, valid(0xf2f52010) - read(0x89e45589)
Thu Jun  1 23:54:41 2023 kern.err kernel: [   73.240315] F2FS-fs (sda128): Can't find valid F2FS filesystem in 1th superblock
Thu Jun  1 23:54:41 2023 kern.info kernel: [   73.241118] F2FS-fs (sda128): Magic Mismatch, valid(0xf2f52010) - read(0x2e48937b)
Thu Jun  1 23:54:41 2023 kern.err kernel: [   73.241546] F2FS-fs (sda128): Can't find valid F2FS filesystem in 2th superblock
Thu Jun  1 23:54:41 2023 kern.err kernel: [   73.242347] exFAT-fs (sda128): invalid boot record signature
Thu Jun  1 23:54:41 2023 kern.err kernel: [   73.242686] exFAT-fs (sda128): failed to read boot sector
Thu Jun  1 23:54:41 2023 kern.err kernel: [   73.243021] exFAT-fs (sda128): failed to recognize exfat type
Thu Jun  1 23:54:51 2023 daemon.warn pppd[6764]: Timeout waiting for PADO packets
Thu Jun  1 23:54:51 2023 daemon.err pppd[6764]: Unable to complete PPPoE Discovery

kernel-my, 2023-06-02:
这是后面一半的日志：
Thu Jun  1 23:54:51 2023 daemon.info pppd[6764]: Exit.
Thu Jun  1 23:54:51 2023 daemon.notice netifd: Interface 'wan' is now down
Thu Jun  1 23:54:51 2023 daemon.notice ttyd[5160]: [2023/06/01 23:54:51:3255] N: rops_handle_POLLIN_netlink: DELADDR
Thu Jun  1 23:54:51 2023 daemon.info avahi-daemon[2048]: Interface eth1.IPv6 no longer relevant for mDNS.
Thu Jun  1 23:54:51 2023 daemon.info avahi-daemon[2048]: Leaving mDNS multicast group on interface eth1.IPv6 with address fe80::7c9c:72ff:fec4:642f.
Thu Jun  1 23:54:51 2023 daemon.notice netifd: Interface 'wan' is disabled
Thu Jun  1 23:54:51 2023 daemon.notice netifd: Interface 'wan' is enabled
Thu Jun  1 23:54:51 2023 daemon.notice netifd: Interface 'wan' is setting up now
Thu Jun  1 23:54:51 2023 daemon.info avahi-daemon[2048]: Withdrawing address record for fe80::7c9c:72ff:fec4:642f on eth1.
Thu Jun  1 23:54:51 2023 kern.info kernel: [   82.929776] 8021q: adding VLAN 0 to HW filter on device eth1
Thu Jun  1 23:54:51 2023 daemon.notice netifd: wan (7133): uci: Entry not found
Thu Jun  1 23:54:51 2023 daemon.notice netifd: wan (7133): sh: out of range
Thu Jun  1 23:54:51 2023 daemon.info pppd[7147]: Plugin pppoe.so loaded.
Thu Jun  1 23:54:51 2023 daemon.info pppd[7147]: PPPoE plugin from pppd 2.4.9
Thu Jun  1 23:54:51 2023 daemon.notice pppd[7147]: pppd 2.4.9 started by root, uid 0
Thu Jun  1 23:54:52 2023 daemon.info avahi-daemon[2048]: Joining mDNS multicast group on interface eth1.IPv6 with address fe80::7c9c:72ff:fec4:642f.
Thu Jun  1 23:54:52 2023 daemon.info avahi-daemon[2048]: New relevant interface eth1.IPv6 for mDNS.
Thu Jun  1 23:54:52 2023 daemon.info avahi-daemon[2048]: Registering new address record for fe80::7c9c:72ff:fec4:642f on eth1.*.
Thu Jun  1 23:55:06 2023 daemon.warn pppd[7147]: Timeout waiting for PADO packets
Thu Jun  1 23:55:06 2023 daemon.err pppd[7147]: Unable to complete PPPoE Discovery
Thu Jun  1 23:55:06 2023 daemon.info pppd[7147]: Exit.
Thu Jun  1 23:55:06 2023 daemon.notice netifd: Interface 'wan' is now down
Thu Jun  1 23:55:06 2023 daemon.notice ttyd[5160]: [2023/06/01 23:55:06:4154] N: rops_handle_POLLIN_netlink: DELADDR
Thu Jun  1 23:55:06 2023 daemon.info avahi-daemon[2048]: Interface eth1.IPv6 no longer relevant for mDNS.
Thu Jun  1 23:55:06 2023 daemon.info avahi-daemon[2048]: Leaving mDNS multicast group on interface eth1.IPv6 with address fe80::7c9c:72ff:fec4:642f.
Thu Jun  1 23:55:06 2023 daemon.notice netifd: Interface 'wan' is disabled
Thu Jun  1 23:55:06 2023 daemon.notice netifd: Interface 'wan' is enabled
Thu Jun  1 23:55:06 2023 daemon.notice netifd: Interface 'wan' is setting up now
Thu Jun  1 23:55:06 2023 daemon.info avahi-daemon[2048]: Withdrawing address record for fe80::7c9c:72ff:fec4:642f on eth1.
Thu Jun  1 23:55:06 2023 kern.info kernel: [   98.019707] 8021q: adding VLAN 0 to HW filter on device eth1
Thu Jun  1 23:55:06 2023 daemon.notice netifd: wan (7424): uci: Entry not found
Thu Jun  1 23:55:06 2023 daemon.notice netifd: wan (7424): sh: out of range
Thu Jun  1 23:55:06 2023 daemon.info pppd[7438]: Plugin pppoe.so loaded.
Thu Jun  1 23:55:06 2023 daemon.info pppd[7438]: PPPoE plugin from pppd 2.4.9
Thu Jun  1 23:55:06 2023 daemon.notice pppd[7438]: pppd 2.4.9 started by root, uid 0
Thu Jun  1 23:55:08 2023 daemon.info avahi-daemon[2048]: Joining mDNS multicast group on interface eth1.IPv6 with address fe80::7c9c:72ff:fec4:642f.
Thu Jun  1 23:55:08 2023 daemon.info avahi-daemon[2048]: New relevant interface eth1.IPv6 for mDNS.
Thu Jun  1 23:55:08 2023 daemon.info avahi-daemon[2048]: Registering new address record for fe80::7c9c:72ff:fec4:642f on eth1.*.
Thu Jun  1 23:55:11 2023 kern.err kernel: [  103.307672] /dev/sda1: Can't open blockdev
Thu Jun  1 23:55:11 2023 kern.err kernel: [  103.308720] /dev/sda1: Can't open blockdev
Thu Jun  1 23:55:11 2023 kern.err kernel: [  103.309363] /dev/sda1: Can't open blockdev
Thu Jun  1 23:55:11 2023 kern.err kernel: [  103.309853] squashfs: Unknown parameter 'discard'
Thu Jun  1 23:55:11 2023 kern.err kernel: [  103.310410] /dev/sda1: Can't open blockdev
Thu Jun  1 23:55:21 2023 daemon.warn pppd[7438]: Timeout waiting for PADO packets
Thu Jun  1 23:55:21 2023 daemon.err pppd[7438]: Unable to complete PPPoE Discovery
Thu Jun  1 23:55:21 2023 daemon.info pppd[7438]: Exit.
Thu Jun  1 23:55:21 2023 daemon.notice netifd: Interface 'wan' is now down
Thu Jun  1 23:55:21 2023 daemon.notice ttyd[5160]: [2023/06/01 23:55:21:5055] N: rops_handle_POLLIN_netlink: DELADDR
Thu Jun  1 23:55:21 2023 daemon.info avahi-daemon[2048]: Interface eth1.IPv6 no longer relevant for mDNS.
Thu Jun  1 23:55:21 2023 daemon.info avahi-daemon[2048]: Leaving mDNS multicast group on interface eth1.IPv6 with address fe80::7c9c:72ff:fec4:642f.
Thu Jun  1 23:55:21 2023 daemon.notice netifd: Interface 'wan' is disabled
Thu Jun  1 23:55:21 2023 daemon.notice netifd: Interface 'wan' is enabled
Thu Jun  1 23:55:21 2023 daemon.notice netifd: Interface 'wan' is setting up now
Thu Jun  1 23:55:21 2023 daemon.info avahi-daemon[2048]: Withdrawing address record for fe80::7c9c:72ff:fec4:642f on eth1.
Thu Jun  1 23:55:21 2023 kern.info kernel: [  113.109855] 8021q: adding VLAN 0 to HW filter on device eth1
Thu Jun  1 23:55:21 2023 daemon.notice netifd: wan (7762): uci: Entry not found
Thu Jun  1 23:55:21 2023 daemon.notice netifd: wan (7762): sh: out of range
Thu Jun  1 23:55:21 2023 daemon.info pppd[7776]: Plugin pppoe.so loaded.
[... cut ...]
Thu Jun  1 23:56:59 2023 daemon.info pppd[13817]: PPPoE plugin from pppd 2.4.9
Thu Jun  1 23:56:59 2023 daemon.notice pppd[13817]: pppd 2.4.9 started by root, uid 0
Thu Jun  1 23:57:00 2023 daemon.info avahi-daemon[2048]: Joining mDNS multicast group on interface eth1.IPv6 with address fe80::7c9c:72ff:fec4:642f.
Thu Jun  1 23:57:00 2023 daemon.info avahi-daemon[2048]: New relevant interface eth1.IPv6 for mDNS.
Thu Jun  1 23:57:00 2023 daemon.info avahi-daemon[2048]: Registering new address record for fe80::7c9c:72ff:fec4:642f on eth1.*.

kernel-my, 2023-06-02:
哥哥们，我是哪里设置有问题吗？网上各种方法都试过呢，包括添加mac地址，关掉ipv6啥的都试过呢

coolsnowwolf, 2023-06-02:
Radius User Restriction  黑号了

kernel-my, 2023-06-03:
>  Radius User Restriction  黑号了

刚玩这个，啥意思啊？我该怎么解决，移动的mac地址也解绑呢，还是播不了号，崩溃

WYC-2020, 2023-06-03:
黑号就是被运营商拉黑了，自己是不是用这个搞了多播之类的，自己打电话让运营商给你解封，有的地方运营商真对多播会拉黑的自己玩之前了解一下

kernel-my, 2023-06-03:
感谢两位老哥，你说多播那我想起来，我现在路由器已经拨号呢，后面openwrt在拨号，就播不上，应该是被占用了，我先打电话问问能不能多播

WYC-2020, 2023-06-03:
> 感谢两位老哥，你说多播那我想起来，我现在路由器已经拨号呢，后面openwrt在拨号，就播不上，应该是被占用了，我先打电话问问能不能多播

建议你是吧原来的播的断了，再用这个播，如果可以播成功就没黑号，只是不能多播而已，如果还是不行那就是被拉黑了，你才去找他们

kernel-my, 2023-06-03:
> > 感谢两位老哥，你说多播那我想起来，我现在路由器已经拨号呢，后面openwrt在拨号，就播不上，应该是被占用了，我先打电话问问能不能多播
> 
> 建议你是吧原来的播的断了，再用这个播，如果可以播成功就没黑号，只是不能多播而已，如果还是不行那就是被拉黑了，你才去找他们

ok,先试试哈，不通晚点再问

closed 2023-06-05 by kernel-my