coolsnowwolf/lede #11272: 兆能m2自编译固件刷入后无限重启

zqs2020git, 2023-06-03:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
兆能m2自编译固件刷入后无限重启

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述
原来正常的情况下系统升级刷入自编译固件，无限重启，后刷自编译bui也是无限重启，后刷入正常的ubi后刷入自编译bin还是自动重启，自编译固件无法正常试用，编译无错，求解

### (1) 具体问题

A：

### (2) 路由器型号和固件版本
ipq6018  256内存  内核5.15
`Format: Log Type - Time(microsec) - Message - Optional Info
Log Type: B - Since Boot(Power On Reset),  D - Delta,  S - Statistic
S - QC_IMAGE_VERSION_STRING=BOOT.XF.0.3-00086-IPQ60xxLZB-1
S - IMAGE_VARIANT_STRING=IPQ6018LA
S - OEM_IMAGE_VERSION_STRING=crm-ubuntu200
S - Boot Interface: NAND
S - Secure Boot: Off
S - Boot Config @ 0x000a602c = 0x000002e5
S - JTAG ID @ 0x000a607c = 0x001390e1
S - OEM ID @ 0x000a6080 = 0x00000000
S - Serial Number @ 0x000a4128 = 0x8e01d2f0
S - OEM Config Row 0 @ 0x000a4188 = 0x0000000000000000
S - OEM Config Row 1 @ 0x000a4190 = 0x0000000000000000
S - Feature Config Row 0 @ 0x000a4130 = 0x0000800018200021
S - Feature Config Row 1 @ 0x000a4138 = 0x02c3e83783000009
S - PBL Patch Ver: 1
S - I-cache: On
S - D-cache: On
B -      3413 - PBL, Start
B -       592 - bootable_media_detect_entry, Start
B -      4339 - bootable_media_detect_success, Start
B -      5207 - elf_loader_entry, Start
B -      5380 - auth_hash_seg_entry, Start
B -      7846 - auth_hash_seg_exit, Start
B -      8344 - elf_segs_hash_verify_entry, Start
B -    103770 - elf_segs_hash_verify_exit, Start
B -    107963 - auth_xbl_sec_hash_seg_entry, Start
B -    108108 - auth_xbl_sec_hash_seg_exit, Start
B -    114651 - xbl_sec_segs_hash_verify_entry, Start
B -    114652 - xbl_sec_segs_hash_verify_exit, Start
B -    115581 - PBL, End
B -     97508 - SBL1, Start
B -    237259 - GCC [RstStat:0x10, RstDbg:0x700100] WDog Stat : 0x4
B -    239699 - clock_init, Start
D -      2684 - clock_init, Delta
B -    248453 - boot_flash_init, Start
D -     30439 - boot_flash_init, Delta
B -    281972 - sbl1_ddr_set_default_params, Start
D -       274 - sbl1_ddr_set_default_params, Delta
B -    288591 - boot_config_data_table_init, Start
D -      4788 - boot_config_data_table_init, Delta - (575 Bytes)
B -    298442 - CDT Version:2,Platform ID:8,Major ID:3,Minor ID:2,Subtype:0
B -    303261 - Image Load, Start
D -      6619 - OEM_MISC Image Loaded, Delta - (0 Bytes)
B -    312594 - Image Load, Start
D -      5063 - PMIC Image Loaded, Delta - (0 Bytes)
B -    320463 - sbl1_ddr_set_params, Start
B -    325465 - CPR configuration: 0x555
B -    328668 - Pre_DDR_clock_init, Start
D -       213 - Pre_DDR_clock_init, Delta
D -         0 - sbl1_ddr_set_params, Delta
B -    361425 - Image Load, Start
D -       427 - APDP Image Loaded, Delta - (0 Bytes)
[... 351 lines cut ...]
SF: Unsupported flash IDs: manuf 00, jedec 0000, ext_jedec 0000
ipq_spi: SPI Flash not found (bus/cs/speed/mode) = (0/0/48000000/0)
128 MiB
MMC:   <NULL>: 0
PCI0 is not defined in the device tree
In:    serial@78B1000
Out:   serial@78B1000
Err:   serial@78B1000
machid: 8030200
eth5 MAC Address from ART is not valid
Net:   MAC0 addr:80:12:df:f5:d1:d0
PHY ID1: 0x4d
PHY ID2: 0xd0b1
EDMA ver 1 hw init
Num rings - TxDesc:1 (0-0) TxCmpl:1 (0-0)
RxDesc:1 (15-15) RxFill:1 (7-7)
ipq6018_edma_alloc_rings: successfull
ipq6018_edma_setup_ring_resources: successfull
ipq6018_edma_configure_rings: successfull
ipq6018_edma_hw_init: successfull
eth0
Hit any key to stop autoboot:  0 
IPQ6018# 
`
A：

davidtall, 2023-06-04:
same issue

davidtall, 2023-06-04:
git checkout 7e397cd22f7a2b175d4ef7064d79b477a58d9990  回滚到 https://github.com/davidtall/lede/commit/7e397cd22f7a2b175d4ef70  是正常的

mitsukileung, 2023-06-04:
IPQ807x ax6 也是一样不断重启

davidtall, 2023-06-05:
大概率是更新了内核 patch 后导致的，回滚到 https://github.com/coolsnowwolf/lede/commit/7e397cd22f7a2b175d4ef7064d79b477a58d9990 是正常。另外有没有大佬加一下ipq60xx 的 5.15的 cpufreq 驱动？

WYC-2020, 2023-06-05:
你这个回滚的太多了，感觉应该不是这个，这个提交就只是加入了一个tls模块而已，应该是另外某个提交，你可以在细化一下 具体到某一个提交上，但是就得你不断的撤回，然后不断的编译。不断的刷，哈哈 看你喜欢折腾不

> 大概率是更新了内核 patch 后导致的，回滚到 [7e397cd](https://github.com/coolsnowwolf/lede/commit/7e397cd22f7a2b175d4ef7064d79b477a58d9990) 是正常。另外有没有大佬加一下ipq60xx 的 5.15的 cpufreq 驱动？

Gaojianli, 2023-06-29:
> 你这个回滚的太多了，感觉应该不是这个，这个提交就只是加入了一个tls模块而已，应该是另外某个提交，你可以在细化一下 具体到某一个提交上，但是就得你不断的撤回，然后不断的编译。不断的刷，哈哈 看你喜欢折腾不
> 
> > 大概率是更新了内核 patch 后导致的，回滚到 [7e397cd](https://github.com/coolsnowwolf/lede/commit/7e397cd22f7a2b175d4ef7064d79b477a58d9990) 是正常。另外有没有大佬加一下ipq60xx 的 5.15的 cpufreq 驱动？

经过测试，在a0948d91a1d7cdd1ffa3226ae197968b1f64c9f4就存在重启的问题了

AnClark, 2023-07-20:
我在用这个仓库编译时，也存在无限重启的情况。L大的另一个仓库openwrt-gl-ax1800编译出的兆能M2固件，则不存在该问题。

panic的日志似乎与非法内存访问有关，一加载nand驱动就崩溃。不排除是DTS的问题。

```
[ 0.695141] nand: ESMT PSR1GA30DT
[ 0.701713] nand: 128 MiB, SLC, erase size: 128 KiB, page size: 2048, OOB size: 64
[ 0.705169] Unable to handle kernel read from unreadable memory at virtual address 00000000000002bc
[ 0.712401] Mem abort info:
[ 0.721343] ESR = 0x0000000096000005
[ 0.724118] EC = 0x25: DABT (current EL), IL = 32 bits
[ 0.727932] SET = 0, FnV = 0
[ 0.733395] EA = 0, S1PTW = 0
[ 0.736267] FSC = 0x05: level 1 translation fault
[ 0.739302] Data abort info:
[ 0.744165] ISV = 0, ISS = 0x00000005
[ 0.747285] CM = 0, WnR = 0
[ 0.750843] [00000000000002bc] user address but active_mm is swapper
[ 0.753984] Internal error: Oops: 96000005 [https://github.com/coolsnowwolf/lede/issues/1] SMP
[ 0.760395] Modules linked in:
[ 0.764994] CPU: 1 PID: 1 Comm: swapper/0 Not tainted 5.15.112 #0
[ 0.768124] Hardware name: ZN M2 (DT)
[ 0.774284] pstate: 60400005 (nZCv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[ 0.777934] pc : nvmem_unregister+0x18/0x70
[ 0.784699] lr : mtd_device_parse_register+0x1b0/0x2f0
[ 0.788869] sp : ffffffc008bfbb20
[ 0.794072] x29: ffffffc008bfbb20 x28: ffffffc0087b62d0 x27: ffffff800a59b7f8
[ 0.797467] x26: ffffffc0088a15e8 x25: ffffff8001ea2410 x24: 0000000000000000
[ 0.804585] x23: 0000000000000000 x22: 0000000000000000 x21: ffffffc0087b62d0
[ 0.811702] x20: 00000000fffffdfb x19: ffffff80025c4080 x18: 0000000000000068
[ 0.818819] x17: 697320424f4f202c x16: 38343032203a657a x15: ffffffc008ab7b10
[ 0.825939] x14: 0000000000000000 x13: 0000000000000001 x12: 0000000000000000
[ 0.833056] x11: 0000000000000000 x10: 0101010101010101 x9 : ffffffc008bfb790
[ 0.840175] x8 : 7f7f7f7f7f7f7f7f x7 : fefefefefefefeff x6 : 8080808080808080
[ 0.847292] x5 : 0000000000000000 x4 : 8080808080808080 x3 : 0000000000000000
[ 0.854411] x2 : 0000000000000001 x1 : 0000000000000009 x0 : 00000000000002bc
[ 0.861530] Call trace:
[ 0.868638] nvmem_unregister+0x18/0x70
[ 0.870897] mtd_device_parse_register+0x1b0/0x2f0
[ 0.874718] qcom_nandc_probe+0x49c/0x820
[ 0.879578] platform_probe+0x68/0xe0
[ 0.883657] really_probe.part.0+0x9c/0x30c
[ 0.887304] __driver_probe_device+0x98/0x170
[ 0.891298] driver_probe_device+0x44/0x120
[ 0.895811] __driver_attach+0x94/0x190
[ 0.899803] bus_for_each_dev+0x60/0xa0
[ 0.903622] driver_attach+0x24/0x30
[ 0.907442] bus_add_driver+0x108/0x1fc
[ 0.911261] driver_register+0x78/0x130
[ 0.914820] __platform_driver_register+0x28/0x34
[ 0.918642] qcom_nandc_driver_init+0x1c/0x28
[ 0.923503] do_one_initcall+0x50/0x1c0
[ 0.927842] kernel_init_freeable+0x234/0x29c
[ 0.931489] kernel_init+0x24/0x120
[ 0.936001] ret_from_fork+0x10/0x20
[ 0.939306] Code: 910af000 910003fd 52800022 f9800011 (885f7c03)
[ 0.943137] ---[ end trace cc5fb4357c618128 ]---
[ 0.949113] Kernel panic - not syncing: Oops: Fatal exception
[ 0.953801] SMP: stopping secondary CPUs
[ 0.959445] Kernel Offset: disabled
[ 0.963431] CPU features: 0x00000040,00000802
[ 0.966645] Memory Limit: none
[ 0.971157] Rebooting in 1 seconds..
```

wordscode, 2023-07-27:
难怪我编译刷入没反应进不了网页

davidtall, 2023-08-11:
> > 你这个回滚的太多了，感觉应该不是这个，这个提交就只是加入了一个tls模块而已，应该是另外某个提交，你可以在细化一下 具体到某一个提交上，但是就得你不断的撤回，然后不断的编译。不断的刷，哈哈 看你喜欢折腾不
> > > 大概率是更新了内核 patch 后导致的，回滚到 [7e397cd](https://github.com/coolsnowwolf/lede/commit/7e397cd22f7a2b175d4ef7064d79b477a58d9990) 是正常。另外有没有大佬加一下ipq60xx 的 5.15的 cpufreq 驱动？
> 
> 经过测试，在 [a0948d9](https://github.com/coolsnowwolf/lede/commit/a0948d91a1d7cdd1ffa3226ae197968b1f64c9f4) 就存在重启的问题了

@chaichunyang 大佬要不要来修复一下这个问题

chaichunyang, 2023-08-15:
> @chaichunyang 大佬要不要来修复一下这个问题

我只是refresh了一下patch，水平有限，实在不知道是哪个patch引起，希望有大佬能帮忙解决

closed 2026-09-05 by coolsnowwolf