coolsnowwolf/lede #11274: tcp_no_window_check & tcp_be_liberal

XXXstyle, 2023-06-04:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述
看到昨天的更新又撤回，这让我很疑惑，到底是tcp_no_window_check好还是tcp_be_liberal
### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：R5S

### (3) 详细日志

A：

WYC-2020, 2023-06-05:
纠结什么，跟到l走就行， 如果你自己fork了，你可以自己改回来使用tcp_no_window_check，因为我这边目前就是用的这个，因为考虑这两个是不同的功能，感觉不应该用tcp_be_liberal来控制这个功能，看个人喜好，反正无所谓的，并且5.15 6.1 这两个内核tcp_no_window_check这个值和tcp_be_liberal 默认都是0的，所以即使用tcp_be_liberal 都一样，排除认为去修改内核参数哈，但是5.4和5.10就不一样了，他们tcp_no_window_check值为1 是开启了的，tcp_be_liberal是为0的，所以如果改用tcp_be_liberal 自然就不一样了，其实我在考虑5.15 6.1 这个tcp_no_window_check 值应该取1吧，禁用窗口检查，这些都是根据你自己的网络情况来决定是否开启

XXXstyle, 2023-06-05:
> 纠结什么，走到l走就走，如果你自己fork了，你可以自己改回来使用tcp_no_window_check，因为我这边目前就是用的这个，因为考这个两个是不同的功能，感觉不应该使用tcp_be_liberal来控制这个功能，看个人喜欢，反正无所谓的，并且5.15 6.1这两个内核tcp_no_window_check这个值和tcp_be_liberal默认都是0的，所以即使用tcp_be_liberal都一样，排除认为去修改内部参数哈，但是是5.4和5.10就不是一样了，他们tcp_no_window_check值为1是启动了的，tcp_be_liberal是为0的，所以如果改用tcp_be_liberal自然就不一样了，其实我在考5.15 6.1这个tcp_ no_window_check 应该取1吧，禁止使用窗口检查，这些都是根据你自己的网络情况来决定是否开启

大佬有没有测试过这两个的性能差异，因为我看介绍这两个似乎差不多的优点，又是差不多的缺点，可惜我这边没有测试的环境

XXXstyle, 2023-06-05:
想在6.1试一下开启tcp_no_window_check，大佬能否给个开启后的修改文件，谢谢！

WYC-2020, 2023-06-05:
[image]
[image]
这里改成1就行了，sfe这玩意在软路由上面基本就是没啥多大提升，因为现在软路由都是性能过剩，在硬路由上面作用到才大一点

closed 2023-06-05 by XXXstyle

XXXstyle, 2023-06-05:
> [image] [image] 这里改成1就行了，sfe这玩意在软路由上面基本就是没啥多大提升，因为现在软路由都是性能过剩，在硬路由上面作用到才大一点

权衡利弊后在6.1下打开了tcp_be_liberal，网站ping值降低了10%以上，samba4传输速度也稳定了一些，综合好评！